pschatzmann/arduino-audio-tools #1: Stream data in the moment we choose

pablofr918, 2021-05-11:
Hello there!! First of all thank you for all your work, because this repository helped me a lot.

I wanted to know if it is possible to make the **out->read(data, len)** out of the callback function.
For example, I would make a **a2dp_source.start("Beats_Pablo")**, without calling the callback and then I think there is another function that makes the **WriteData(Data)**. So is it possible to do this, creating a data variable to use in the WritData()?
(The data variable could be written with **out->read (Channels *Data, size_t len)** but out of the callback).

pschatzmann, 2021-05-11:
I propose to switch to using Streams - this hides all the complexity around callbacks. 
Then you just read from the source A2DPStream and write to the Sink Stream - or you let the StreamCopy class do the copy…

pablofr918, 2021-05-11:
I will check it and tell you, thank you very much!

pablofr918, 2021-05-11:
I'm not sure how it works, so what I think it could be the way is:
1. Create an out as: A2DPStream **out::A2DPStream::instance();**
2. The out in the setup() will be in TX mode as: **in.begin(TX_MODE, "Beats_Pablo");**
But after this, I need to create the **in** from which I'll make the copy, that needs to be the mp3 stream. Thats the part I don't know how to do.
I don't know if it has to be an A2DPStream class too or what.

pablofr918, 2021-05-11:
Yes, I'm using the mp3-a2dp but the problem is that I would like to do a connexion to a sink without sending the data instantly, and then when I want to send it, make the mp3.begin and send. But that's what I'm stuck in.

pschatzmann, 2021-05-11:
The callback will be automatically called after A2DP has been started. You just need to feed it with data. So at the end of your mp3 file move to the next and feed it with data by calling  mp3->begin(file, out);
I think that should work....

pablofr918, 2021-05-11:
I will try it, thank you very much for the support!

pablofr918, 2021-05-20:
Hi again, I'm trying to do what you said, but doesn't work. What I'm doing is calling the start() and then when I want I send an MQTT message that makes the mp3->begin(file,out). The first time I do it it works, but if I want to do it again it crashes and does nothing. I would like to know how the get_sound_data works, so maybe I could do it using another way. For example, there is a function in BluetoothA2DPSource.h whichi is **bool writeData(SoundData *data);** and if I could convert my mp3 file to that data it would be easy to send via a2dp, do you know any way to get to this point?