pschatzmann/arduino-audio-tools #7: Error compiling base-file_mp3-a2dp

rrezaei2, 2021-07-04:
Hello,

By compiling, I get following error:
T result= 0;
Buffers.h Error: conversion from 'int' to Non-scalar type 'channels' requested.

My Target is to send and play some Sound file, e.g. as mp3 using esp32 Bluetooth and a headset.

Regards,
Reza

pschatzmann, 2021-07-04:
for which example did you get the error ?

rrezaei2, 2021-07-04:
Base-file_mp3-a2dp

pschatzmann, 2021-07-04:
Very strange, I tried to reproduce your issue but I don't get any errors:

- Can you check that you have the latest version of the libraries and example
- ->Arduino -> Preferences -> Compiler Warnings is at Default
- Checking in the Board Manager: ESP32 I am using the latest version which is 1.0.6

closed 2021-10-11 by pschatzmann