pschatzmann/arduino-audio-tools #9: :question:

akira-kun, 2021-07-15:
Hello, your tools are really interesting and useful!! 

I trying to develop a esp32 webradio bluetooth sender, so my question is: 
Is it possible to receive mp3 stream from internet and retransmit it with the a2dp lib?

I tried to change base-file_mp3-a2dp, but I got some errors with T result = 0;
I'm using the lastest esp32 version (arduino IDE).

Thanks ;)

pschatzmann, 2021-07-15:
Unfortunately you can not use Bluetooth and WIFI at the same time on a ESP32.

pierreverbakel, 2021-08-28:
That is a big shame, but ... one could use two ESP32s and transport the data via a wire I guess?

pschatzmann, 2021-08-28:
Yes, you could connect them via I2S...

podaen, 2021-11-03:
I have a project where I use bluetoothLE and wifi in to one project. I don't use them at the same time but they are in it and it works. But with wifi streams I don't know, You can't mis a thing... right? Therefore you need too delay the internet connection, and first send all available data too the spreaker. Then the buffer is empty so you could fill him again. What the wifi concerns... its like having a bad internet connection. It also works with buffers so you would have some time to do the bluetooth part.

podaen, 2022-03-13:
I tried to add a bluetooth ble with the audiotools, but it diden't work. Is there a chance to send wifi command when the music is not playing?

pschatzmann, 2023-02-13:
See https://github.com/pschatzmann/ESP32-A2DP/wiki/WIFI-and-A2DP-Coexistence

closed 2023-02-13 by pschatzmann