pschatzmann/arduino-audio-tools #10: Trying to establish a Stream from i2s to a2dp.

Electricious, 2021-07-31:
Hello,
first of all, thanks for those awesome tools!
As the title says, I'm trying to establish a stream between i2s (in) and a2dp (out).
In the folder sandbox, there already is an example that won't work, at least not for me. With help of the working example, which is the other way round, I was able to build it.
As I was trying to understand the code, what does what, and so on, I stumbled over one thing I don't understand.
In the Class I2SStream and the object that calls ".begin" is just installing the drivers and configurations for the I2S. 
The Class A2DPStream an the object that calls ".begin" is connecting to the Bluetooth device and opens a useable connection.
The copier in the loop is doing two channels out of one from the Streams I defined as input and output.
The question I got is, where happens the "read" of the actual data from the I2S. I have to read the data at some point, with something like "i2s.read()", which I could find in some classes, but I didn't get, if the code is already using the i2s.read somewhere and the problem is actually a different one.

I'll put my sketch in as a reference, so that you know what I'm talking about. :)

`#include "Arduino.h"
#include "AudioTools.h"
#include "AudioA2DP.h"

using namespace audio_tools;

// I2S<int32_t> i2s;
I2SStream i2sStream;                            // Access I2S as stream
A2DPStream a2dpStream = A2DPStream::instance(); // access A2DP as stream
// ChannelConverter<int32_t> converter(&convertFrom32To16);
// ConverterFillLeftAndRight<int32_t> bothChannels;
StreamCopy copier(a2dpStream, i2sStream); // copy i2sStream to a2dpStream

// Arduino Setup
void setup(void)
{
    Serial.begin(115200);

    // start bluetooth
    a2dpStream.begin(TX_MODE, "Mpow M30");
    Serial.println("A2DP is connected now...");

    // start i2s input with default configuration
    Serial.println("starting I2S...");
    I2SConfig config = i2sStream.defaultConfig(RX_MODE);
    config.sample_rate = 44100;
    config.channels = 2;
    config.bits_per_sample = 16;
    i2sStream.begin(config);

    Serial.println("I'm at the end of SETUP");
}

// Arduino loop - copy data
void loop()
{
    if (a2dpStream)
    {
        Serial.println("I'm in the LOOP");
        copier.copy();
    }
}`

pschatzmann, 2021-07-31:
The copy is  doing a read from I2SStream. If you check in the source, it uses a buffered read from I2S by calling i2s.readBytes(data, length);

I think your example should work: I suggest to set the logging level to debug to find where the issue is....

Electricious, 2021-08-01:
Ah yes, I see it, good to know.
I enabled the Debug logger, as you suggested, and don't quite get it.
Following I post a few rounds.

```
[I] AudioA2DP.h : 144 - Starting a2dp_source...
[I] AudioA2DP.h : 151 - a2dp_source is connected...
starting I2S...
[D] I2S_ESP32.h : 30 - begin
[I] I2S_ESP32.h : 237 - mode: RX
[I] I2S_ESP32.h : 238 - sample rate: 44100
[I] I2S_ESP32.h : 239 - bits per sample: 16
[I] I2S_ESP32.h : 240 - number of channels: 1
[I] I2S_ESP32.h : 245 - pin bck_io_num: 14
[I] I2S_ESP32.h : 246 - pin ws_io_num: 15
[I] I2S_ESP32.h : 247 - pin data_num: 32
[D] I2S_ESP32.h : 91 - begin - started
[I] AudioA2DP.h : 68 - [D] Buffers.h : 32 - writeArray: 512
a2dp_stream_source_sound_data 128 -> 0
[I] AudioA2DP.h : 200 - write 512->512
[I] AudioCopy.h : 82 - StreamCopy::copy 512 -> 512 bytes - in 1 hops
[D] Buffers.h : 32 - writeArray: 512
[I] AudioA2DP.h : 200 - write 512->512
[I] AudioCopy.h : 82 - StreamCopy::copy 512 -> 512 bytes - in 1 hops
[D] Buffers.h : 32 - writeArray: 512
[I] AudioA2DP.h : 200 - write 512->512
[I] AudioCopy.h : 82 - StreamCopy::copy 512 -> 512 bytes - in 1 hops
[I] AudioA2DP.h : 68 - [D] Buffers.h : 32 - writeArray: 512ce_sound_data 128 -> 0
riteArray: 512
[I] AudioA2DP.h : 200 - write 512->512
[I] AudioCopy.h : 82 - StreamCopy::copy 512 -> 512 bytes - in 1 hops
[D] Buffers.h : 32 - writeArray: 512
[I] AudioA2DP.h : 200 - write 512->512
[I] AudioCopy.h : 82 - [I] AudioA2DP.h : 68 - a2dp_stream_source_sound_data 128 -> 01 hops
2dp_stream_source_sound_data 128 -> 0
```
that the cycles aren't always the same or got mixed up isn't an issue, I guess, due to the speed, the print to Serial will surely do something wrong.

And just to be sure, we are talking from the perspective of the ESP, when we send data. So RX means, that the ESP receives and TX that the ESP sends, right? Not that I got that mixed up.

But in the Log, everything seems fine to met but the "a2dp_stream_source_sound_data 128 -> 0".
I don't expect there an 0, don't I?
Additionally I should say, that I use a SPH0645 MEMS microphone as data for I2S, that gives an int32, but I'm sure that the Streamer is converting this.

pschatzmann, 2021-08-01:
I will look into this...

fmvoxasoft, 2021-09-03:
> Ah yes, I see it, good to know.
> I enabled the Debug logger, as you suggested, and don't quite get it.
> Following I post a few rounds.
>
> ```
> [I] AudioA2DP.h : 144 - Starting a2dp_source...
> [I] [...]

I tried your code on my setup (DSP (i2s master) and ESP (i2s slave)) and got the same problem. 

But this code example was working, when the ESP was the master. I am using this [code example](https://www.pschatzmann.ch/home/2021/04/29/bluetooth-a2dp-streaming-from-an-digital-i2s-microphone/) where the i2s -> a2dp is over a callback function.

pschatzmann, 2021-11-03:
Recently, I have changed the logic quite a bit:
- writes and reads are synchronized with a mutex
- I am using a Ringbuffer now and the size can be configured
- writes to A2DP are blocking now

I did not test it with this scenario - but with mp3 decoding from files and it is working like a charm

pschatzmann, 2021-11-03:
Here is an [improved sketch](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/streams-i2s-a2dp/streams-i2s-a2dp.ino):
```
08:48:08.232 -> [I] AudioA2DP.h : 129 - Connecting to LEXON MINO L
08:48:08.232 -> [I] AudioA2DP.h : 136 - Starting a2dp_source...
08:48:08.746 -> [W] AudioA2DP.h : 290 - ==> state: Connecting
08:48:10.236 -> [W] AudioA2DP.h : 290 - ==> state: Connected
08:48:10.723 -> [I] AudioA2DP.h : 144 - a2dp_source is connected...
08:48:10.723 -> starting I2S...
08:48:10.723 -> [I] I2SESP32.h : 244 - rx/tx mode: RX
08:48:10.723 -> [I] I2SESP32.h : 245 - sample rate: 44100
08:48:10.723 -> [I] I2SESP32.h : 246 - bits per sample: 16
08:48:10.756 -> [I] I2SESP32.h : 247 - number of channels: 2
08:48:10.756 -> [I] I2SESP32.h : 248 - is_master: Master
08:48:10.756 -> [I] I2SESP32.h : 249 - mode: 0
08:48:10.756 -> [I] I2SESP32.h : 253 - pin bck_io_num: 14
08:48:10.756 -> [I] I2SESP32.h : 254 - pin ws_io_num: 15
08:48:10.756 -> [I] I2SESP32.h : 255 - pin data_num: 32
08:48:10.756 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.756 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.789 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.789 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.789 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.789 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.789 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.789 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.823 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.823 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.823 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.823 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.823 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.823 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.856 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.856 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.856 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.856 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.856 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.856 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.893 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.927 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.927 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.927 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.927 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.927 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.927 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.961 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.961 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.961 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.961 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.961 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.961 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.994 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.994 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.994 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.994 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.994 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:10.994 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.028 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.028 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.028 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.028 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.028 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.028 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.062 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.062 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.062 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.062 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.062 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
[... cut ...]
08:48:11.099 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.099 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.099 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.099 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:11.099 -> [W] AudioA2DP.h : 199 - is_a2dp_active -> true with 63488 bytes
08:48:18.824 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
08:48:18.824 -> [I] AudioCopy.h : 274 - StreamCopy::copy 1024 bytes - in 1 hops
```

closed 2021-11-03 by pschatzmann

Electricious, 2021-11-09:
Thank you for putting so much work into it!
Unfortunately, it still isn't working...
`rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
starting A2DP...
[I] AudioA2DP.h : 129 - Connecting to Mpow M30
[I] AudioA2DP.h : 136 - Starting a2dp_source...
[D][BluetoothA2DPSource.cpp:111] BluetoothA2DPSource(): BluetoothA2DPSource, 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060f30  A0      : 0x800df744  A1      : 0x3ffc7760  
A2      : 0x00000064  A3      : 0x00000060  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3f402a50  A12     : 0x00000000  A13     : 0x3ffbb45c  
A14     : 0x3ffbb2d1  A15     : 0x3ffbb45d  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000064  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400014fd:0x3ffc7760 0x400df741:0x3ffc7770 0x400e355a:0x3ffc7a80 0x400e3592:0x3ffc7b10 0x400dac1e:0x3ffc7b50 0x400d8729:0x3ffc7bb0 0x400d75f3:0x3ffc7bd0 0x400dae02:0x3ffc7c90 0x400901da:0x3ffc7cb0
--- exit ---

  #0  0x400014fd:0x3ffc7760 in ?? ??:0
  #1  0x400df741:0x3ffc7770 in _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529
  #2  0x400e355a:0x3ffc7a80 in _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
  #3  0x400e3592:0x3ffc7b10 in vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
  #4  0x400dac1e:0x3ffc7b50 in log_printf at /home/andreas/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:533
  #5  0x400d8729:0x3ffc7bb0 in BluetoothA2DPSource::set_volume(unsigned char) at lib/ESP32-A2DP/src/BluetoothA2DPSource.h:146
  #6  0x400d75f3:0x3ffc7bd0 in audio_tools::A2DPStream::begin(audio_tools::A2DPConfig) at lib/arduino-audio-tools/src/AudioA2DP.h:373
      (inlined by) setup() at src/main.cpp:34
  #7  0x400dae02:0x3ffc7c90 in loopTask(void*) at /home/andreas/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
  #8  0x400901da:0x3ffc7cb0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)`

A few notes here:
- I use Visual Studio Code and PlatformIO
- An ESP32-WROVER-E on a Dev-Kit

- In src, of the arduino-audio-tools, there are some errors due to typos of the files, that you want to include. In I2SStream.h = I2sConfig.h should be I2SConfig.h ; In AudioOutput.h = int24.h should be Int24.h and in PortAudioStream.h is portaudio.h missing.

Electricious, 2021-11-09:
Somehow is the logging the problem, it seems.
The code went to the point "ESP_LOGI(BT_AV_TAG, "%s set_volume: %d", volume);" in the BluetoothA2DPSource.h, where the variable "volume" is volume=100 'd'. At this point, it starts crashing. I commented it out and it went further than that. But still, when I compare your log with mine, I see only the "infos" but mine shows all the "debugs" as well. Not the LOGDs but the ESP_LOGDs.. I don't know if this creates some kind of overhead.

pschatzmann, 2021-11-09:
The Log level should be at most Info - better Warning. Debug is giving too much info and the sound generation can not catch up...

Electricious, 2021-11-09:
Do you know how to disable the logging?
I only have the copied line "AudioLogger::instance().begin(Serial, AudioLogger::Info);" in my sketch. all the other logging is in all your code. Is there a file where I can edit it?

I'm glad that I can help

pschatzmann, 2021-11-09:
I would recommend AudioLogger::instance().begin(Serial, AudioLogger::Warning) in the sketch
In AudioConfig.h you can set/change the defaults and deactivate logging at all...

Electricious, 2021-11-09:
Can you point me where?
I disabled it but the loggings like ESP_LOG are still showing up in the terminal.

pschatzmann, 2021-11-09:
This is the ESP logging: in Arduino:
-> Tools -> Core Debug Level -> e.g. Warn

Electricious, 2021-11-09:
Oh, so in Platformio it would be probably set in the .ini file.

pschatzmann, 2021-11-09:
build_flags = -DCORE_DEBUG_LEVEL=5
 
I think 5 is debug

Electricious, 2021-11-09:
So yeah, the ESP logging caused the error, I even got the stream, but my sound is super noisy and loud but also I hear my environment in the background.
Any suggestions where to tweak?

pschatzmann, 2021-11-09:
The default I2S format might not be fitting with your device
  config.i2s_format = I2S_MSB_FORMAT;
  config.i2s_format = I2S_LSB_FORMAT;
I dont remember which one is the default...

pschatzmann, 2021-11-09:
Could you indicate how you solved your problem ?
What log level are you using now and which format ...

Electricious, 2021-11-10:
Yeah sure.
Basically, I just commented the line in the platform.ini, that says "build_flags = -DCORE_DEBUG_LEVEL=5".
I used an ESP-PROG for inline debugging and saw, that the error occurs, after this line:
```
ESP_LOGD(BT_APP_TAG, "x%x - len: %d", __func__, len);
```
In my terminal snippet, you can see that it printed the function, but after that went mad.
```
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
starting A2DP...
[I] AudioA2DP.h : 129 - Connecting to Mpow M30
[I] AudioA2DP.h : 136 - Starting a2dp_source...
[D][BluetoothA2DPSource.cpp:111] BluetoothA2DPSource(): BluetoothA2DPSource, 
```
There you see the line at the bottom "[D][BluetoothA2DPSource.cpp:111] BluetoothA2DPSource(): BluetoothA2DPSource, " and that the code line prints the value, but the code didn't go further.
So for test purposes, I commented this line and it kinda worked, but I had still a lot of debug logs in the terminal and not a stable nor working stream.
After our discussion, I uncommented the line from before again and commented the DCORE_DEBUG_LEVEL completely.
Then my serial terminal didn't show any logs from the esp, only yours, which indicates the status of the code and the stream worked, except of the noise. But still, you could hear behind that noise, that it kinda works. The terminal looked similar to this then.

> Here is an [improved sketch](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/streams-i2s-a2dp/streams-i2s-a2dp.ino):
>
> ```
> 08:48:08.232 -> [I] AudioA2DP.h : 129 - Connecting to [...]

The format issue I haven't tackled yet.

Electricious, 2021-11-11:
So I tried different formats!
Default is: I2S_STD_FORMAT
However, I used I2S_MSB_FORMAT and with that, it worked!
I also used the build_flag DCORE_DEBUG_LEVEL again.
```build_flags = -DCORE_DEBUG_LEVEL=2 // stands for Warning```
Now it works perfectly, as you said, like a charm.
Thanks again for everything! :)

miroslavpetrov, 2023-05-24:
@BirdLightning Sorry for commenting on closed issue. Can you share the configuration for SPH0645, I have tried with I2S_MSB_FORMAT but i only get noise.