pschatzmann/arduino-audio-tools #17: Feature Request

Dweaver309, 2021-09-17:
Your libs are great I am using them in my new "Jukebox" music play biased on your file_mp3-a2dp.ino sketch. Is it possible to control the volume to the Bluetooth device? If it can be done please post an example. Thanks for the great work.

pschatzmann, 2021-09-18:
Have a look at the ConverterScaler class. This can be passed as argument to the copy method.
The volume can then be controlled by calling setFactor(0.0) for mute or setFactor(1.0) to set the max volume.
 Tomorrow I might be able to make an example...

Dweaver309, 2021-09-19:
Thanks, I'm looking forward to seeing the example.

pschatzmann, 2021-09-20:
Have a look at https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control

Dweaver309, 2021-09-20:
I am not sure I understand the example if you have time could you post an example using  your file_mp3-a2dp.ino sketch. Thanks again.

pschatzmann, 2021-09-20:
Oh, I see you did not use the Stream API.  But for file_mp3-a2dp.ino  it is easy as well:

You just change the sketch to add

```
float volume = 1.0

// callback used by A2DP to provide the sound data
int32_t get_sound_data(Channels* data, int32_t len) {  
  // set volume
  for (int j=0;j<len;j++){
    data[j].channel1 = data[j].channel1 * volume;
    data[j].channel2 = data[j].channel2 * volume;
  }
  return out == nullptr ? 0 : out->read(data, len);
}
```

Changing the volume is then just setting the volume variable to a value between 1.0 and 1.0

Dweaver309, 2021-09-20:
I set the volume to 0.0 and that to 1.0 using a float volume = 0.0 then float
volume = 1.0. The loudness did not change.
Do you think the two bluetooth devices I am using do not work with the call
back?

Thanks for all the help

pschatzmann, 2021-09-21:
Does it help if you use volatile  float volume = 1.0  ?
If not try to to check with Serial.print(volume) that the values are really updated

Dweaver309, 2021-09-21:
Tried volatile ... Serial.print(volume) returns the correct change?

pschatzmann, 2021-09-21:
Did you use the corrected version of get_sound_data() ?

Dweaver309, 2021-09-21:
I will update all my libs now and try. Thanks

Dweaver309, 2021-09-21:
I just updated my arduino-audio-tools lib
This is my get_sound_data:

// callback used by A2DP to provide the sound data
int32_t get_sound_data(Channels *data, int32_t len)
{
  // set volume
  for (int j = 0; j < len; j++)
  {
    data[j].channel1 = data[j].channel1 * volume;
    data[j].channel2 = data[j].channel2 * volume;
  }

  return out == nullptr ? 0 : out->read(data, len);
}

pschatzmann, 2021-09-21:
Nop that won't work.... I suggest that you use the version from 7 hours ago

Dweaver309, 2021-09-21:
The version posted 7 hours ago works great!

Thanks for all your help

closed 2021-09-21 by Dweaver309