pschatzmann/arduino-audio-tools #38: printMetaData in player-url-i2s

podaen, 2021-11-06:
I am not recieving any data out of  printMetaData in the  player-url-i2s example. Can someone confirm he does...

This is one of the url I am currently using

https://22323.live.streamtheworld.com/TOPRETRO.mp3

pschatzmann, 2021-11-06:
I confirm that there is an issue. I checked with the mp3 files where I remember having it seen working, but right now it seems to be broken.
 
I am looking into it..

pschatzmann, 2021-11-06:
I have the ID3 metadata corrected and it is working for files now. Please note that SHOUTcast or Icecast use a different mechanism which has not been implemented yet. 

So I think it is normal that it is not working in the player-url-i2s example.

Therefore I will also remove this code from the examples

podaen, 2021-11-07:
I am testing and stil I have to clone those libraries to get it work and I need to COMMENT them otherwise I get the message ipadressv6 error.

> #include <MP3DecoderHelix.h>
> #include <helix_pgm.h>
> #include <helix_log.h>
> #include <CommonHelix.h>
> #include <AACDecoderHelix.h>
> #include <WiFiUdp.h>
> #include <WiFiType.h>
> #include <WiFiSTA.h>
> #include <WiFiServer.h>
> #include <WiFiScan.h>
> #include <WiFiMulti.h>
> #include <WiFiGeneric.h>
> #include <WiFiClient.h>
> #include <WiFiAP.h>
> #include <WiFi.h>
> #include <ETH.h>
> #include <WiFiClientSecure.h>
> #include <ssl_client.h>

Anyway these are the test result...

> [I] HttpHeader.h : 155 - HttpHeader::readLine -> HTTP/1.0 200 OK
> [I] HttpHeader.h : 443 - HttpReplyHeader::parse1stLine: HTTP/1.0 200 OK
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> P3P: CP="ALL IND DSP COR ADM CONo CUR CUSo IVAo IVDo PSA PSD TAI TELo OUR SAMo CNT COM INT NAV ONL PHY PRE PUR UNI"
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'P3P' : 'CP="ALL IND DSP COR ADM CONo CUR CUSo IVAo IVDo PSA PSD TAI TELo OUR SAMo CNT COM INT NAV ONL PHY PRE PUR UNI"'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-br: 128
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-br' : '128'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-pub: 1
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-pub' : '1'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-description: Liquidsoap Radio!
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-description' : 'Liquidsoap Radio!'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-audio-info: channels=2;samplerate=44100;bitrate=128
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-audio-info' : 'channels=2;samplerate=44100;bitrate=128'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Access-Control-Allow-Origin' : '*'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-url: http://savonet.sf.net
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-url' : 'http://savonet.sf.net'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Instance-id: 1c083e4e39c22dd00dad8289fdca3cfb
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Instance-id' : '1c083e4e39c22dd00dad8289fdca3cfb'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Cache-Control: no-cache
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Cache-Control' : 'no-cache'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Server: AIS aise5.infomaniak.ch Streaming Server 8.5.6
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Server' : 'AIS aise5.infomaniak.ch Streaming Server 8.5.6'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-genre: Misc
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-genre' : 'Misc'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Expires: Mon, 26 Jul 1997 05:00:00 GMT
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Expires' : 'Mon, 26 Jul 1997 05:00:00 GMT'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-metaint: 0
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-metaint' : '0'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Access-Control-Allow-Headers: Content-Type, Icy-Metadata
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Access-Control-Allow-Headers' : 'Content-Type, Icy-Metadata'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Pragma: no-cache
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Pragma' : 'no-cache'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-name: Central Charts
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-name' : 'Central Charts'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Connection: close
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Connection' : 'close'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Content-Type: audio/mpeg
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Content-Type' : 'audio/mpeg'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Set-Cookie: AISSessionId=607a3b123d623ec0_17650988_E2DD9mBv_MTg1Ljc0LjcwLjc!_0000002KdSN; Path=/; Domain=centralcharts.ice.infomaniak.c
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Set-Cookie' : 'AISSessionId=607a3b123d623ec0_17650988_E2DD9mBv_MTg1Ljc0LjcwLjc!_0000002KdSN; Path=/; Domain=centralcharts.ice.infomaniak.ch
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> 
> [W] HttpRequest.h : 141 - no CONTENT_LENGTH found in reply
> [I] URLStream.h : 72 - size: 0
> [I] URLStream.h : 220 - Request written ... waiting for reply
> [I] MetaData.h : 39 - iceMetaint: 0

podaen, 2021-11-07:
> Opening port
> Port open
> [I] URLStream.h : 57 - URLStream.begin http://21303.live.streamtheworld.com/STUDIO_BRUSSEL_128.mp3
> [I] Url.h : 73 - Url::parse
> [I] Url.h : 109 - url->http://21303.live.streamtheworld.com/STUDIO_BRUSSEL_128.mp3
> [I] Url.h : 110 - host->21303.live.streamtheworld.com
> [I] Url.h : 111 - protocol->http
> [I] Url.h : 112 - path->/STUDIO_BRUSSEL_128.mp3
> [I] Url.h : 113 - port->80
> .....
> [I] URLStream.h : 187 - WiFiClient
> [I] URLStream.h : 187 - WiFiClient
> [I] HttpRequest.h : 178 - process connecting to host 21303.live.streamtheworld.com port 80
> [I] HttpRequest.h : 166 - connect 21303.live.streamtheworld.com
> [I] HttpRequest.h : 189 - Free heap: 265060
> [I] HttpHeader.h : 348 - HttpRequestHeader::setValues - path: /STUDIO_BRUSSEL_128.mp3
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Host' : '21303.live.streamtheworld.com'
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Connection' : 'close'
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Accept' : 'audio/mp3'
> [I] HttpHeader.h : 248 - HttpHeader::write
> [I] HttpHeader.h : 357 - HttpRequestHeader::write1stLine
> [I] HttpHeader.h : 372 - -> GET /STUDIO_BRUSSEL_128.mp3 HTTP/1.1
> [I] HttpHeader.h : 185 -  -> Host: 21303.live.streamtheworld.com 
> [I] HttpHeader.h : 185 -  -> Connection: close 
> [I] HttpHeader.h : 185 -  -> Accept: audio/mp3 
> [I] HttpHeader.h : 288 -  -> <CR LF> 
> [I] HttpRequest.h : 213 - Request written ... waiting for reply
> [I] HttpHeader.h : 218 - HttpHeader::read
> [W] HttpHeader.h : 225 - Waiting for data...
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> HTTP/1.0 200 OK
> [I] HttpHeader.h : 443 - HttpReplyHeader::parse1stLine: HTTP/1.0 200 OK
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Expires: Thu, 01 Dec 2003 16:00:00 GMT
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Expires' : 'Thu, 01 Dec 2003 16:00:00 GMT'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Cache-Control: no-cache, must-revalidate
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Cache-Control' : 'no-cache, must-revalidate'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Pragma: no-cache
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Pragma' : 'no-cache'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Access-Control-Allow-Origin' : '*'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Access-Control-Allow-Credentials: true
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Access-Control-Allow-Credentials' : 'true'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Set-Cookie: uuid=dfea88ce-7a38-49cf-b1c2-d94a4e475806; expires=Tue, 07 Dec 2021 09:26:53 GMT; path=/; domain=.live.streamtheworld.com
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Set-Cookie' : 'uuid=dfea88ce-7a38-49cf-b1c2-d94a4e475806; expires=Tue, 07 Dec 2021 09:26:53 GMT; path=/; domain=.live.streamtheworld.com'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Set-Cookie: uuid-s=dfea88ce-7a38-49cf-b1c2-d94a4e475806; expires=Tue, 07 Dec 2021 09:26:53 GMT; path=/; domain=.live.streamtheworld.com
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Set-Cookie' : 'uuid-s=dfea88ce-7a38-49cf-b1c2-d94a4e475806; expires=Tue, 07 Dec 2021 09:26:53 GMT; path=/; domain=.live.streamtheworld.com;
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Content-Type: audio/mpeg
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Content-Type' : 'audio/mpeg'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Connection: close
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Connection' : 'close'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-name: Studio Brussel
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-name' : 'Studio Brussel'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-description: 
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-url: https://stubru.be/
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-url' : 'https://stubru.be/'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-genre: Rock
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-genre' : 'Rock'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-br: 128
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-br' : '128'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Server: MediaGateway 5.6.2-0391.el6
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Server' : 'MediaGateway 5.6.2-0391.el6'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> 
> [W] HttpRequest.h : 141 - no CONTENT_LENGTH found in reply
> [I] URLStream.h : 72 - size: 0

> Opening port
> Port open
> [I] URLStream.h : 57 - URLStream.begin http://22183.live.streamtheworld.com/JOE.mp3
> [I] Url.h : 73 - Url::parse
> [I] Url.h : 109 - url->http://22183.live.streamtheworld.com/JOE.mp3
> [I] Url.h : 110 - host->22183.live.streamtheworld.com
[... 109 lines cut ...]
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-url' : 'https://mnm.be/'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-genre: Pop
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-genre' : 'Pop'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-br: 128
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-br' : '128'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> Server: MediaGateway 5.6.2-0391.el6
> [I] HttpHeader.h : 96 - HttpHeader::put -> 'Server' : 'MediaGateway 5.6.2-0391.el6'
> [I] HttpHeader.h : 155 - HttpHeader::readLine -> 
> [W] HttpRequest.h : 141 - no CONTENT_LENGTH found in reply
> [I] URLStream.h : 72 - size: 0

The icy seems to work. Great job!

[... cut ...]
[I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-description:
[I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-url: https://stubru.be/
[I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-url' : 'https://stubru.be/'
[I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-genre: Rock
[I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-genre' : 'Rock'
[I] HttpHeader.h : 155 - HttpHeader::readLine -> icy-br: 128
[I] HttpHeader.h : 96 - HttpHeader::put -> 'icy-br' : '128'

I am not sure what that means...
[I] MetaData.h : 39 - iceMetaint: 0

pschatzmann, 2021-11-09:
I am planning to have a new ICYStream class which provides the music title from the audio stream. Unfortunately I am still struggling with some currupted memory and at this point of time I don't know yet where the issue is coming from. So I prefer to leave this issue open until this is resolved.

podaen, 2021-11-25:
I am wondering how the csv file is generated...

pschatzmann, 2021-11-25:
The ICYStream is finally working now...

closed 2021-11-25 by pschatzmann