pschatzmann/arduino-audio-tools #39: Testing example player-sd-i2s

podaen, 2021-11-07:
if I comment #define USE_URL_ARDUINO in the AudioConfig I get these issues

> AudioHttp.h:3: In file included from
> AudioTools.h:27: from
> player-sd-i2s.ino:66: from
>  
> URLStreamESP32.h: 169:72: error: 'MethodID' has not been declared
>    void begin(const char* urlStr, const char* acceptMime=nullptr, MethodID action=GET,  const char* reqMime="", const char*reqData="") {
>  
> URLStreamESP32.h: 207:9: error: 'HttpRequest' does not name a type
>    HttpRequest &httpRequest(){
>  
> URLStreamESP32.h: 213:9: error: 'URLStreamDefault' does not name a type
>    URLStreamDefault* p_urlStream=nullptr
>  
> URLStreamESP32.h: 169:88: error: 'GET' was not declared in this scope
>    void begin(const char* urlStr, const char* acceptMime=nullptr, MethodID action=GET,  const char* reqMime="", const char*reqData="") {
> URLStreamESP32.h: In constructor audio_tools::URLStream::URLStream(int)
>  
> URLStreamESP32.h: 148:13: error: 'p_urlStream' was not declared in this scope
>    p_urlStream = new URLStreamDefault(readBufferSize)
>  
> URLStreamESP32.h: 148:31: error: expected type-specifier before 'URLStreamDefault
>    p_urlStream = new URLStreamDefault(readBufferSize)
> URLStreamESP32.h: In constructor audio_tools::URLStream::URLStream(Client&, int)
>  
> URLStreamESP32.h: 154:13: error: 'p_urlStream' was not declared in this scope
>    p_urlStream = new URLStreamDefault(clientPar, readBufferSize)
>  
> URLStreamESP32.h: 154:31: error: expected type-specifier before 'URLStreamDefault
>    p_urlStream = new URLStreamDefault(clientPar, readBufferSize)
> URLStreamESP32.h: In constructor audio_tools::URLStream::URLStream(const char*, const char*, int)
>  
> URLStreamESP32.h: 160:13: error: 'p_urlStream' was not declared in this scope
>    p_urlStream = new URLStreamDefault(network, password, readBufferSize)
>  
> URLStreamESP32.h: 160:31: error: expected type-specifier before 'URLStreamDefault
>    p_urlStream = new URLStreamDefault(network, password, readBufferSize)
> URLStreamESP32.h: In destructor virtual audio_tools::URLStream::~URLStream()
>  
> URLStreamESP32.h: 166:17: error: 'p_urlStream' was not declared in this scope
>    if (p_urlStream!=nullptr) delete p_urlStream
>  
> URLStreamESP32.h: 166:46: error: type '<type error>' argument given to 'delete', expected pointer
>    if (p_urlStream!=nullptr) delete p_urlStream
> URLStreamESP32.h: In member function void audio_tools::URLStream::begin(const char*, const char*, int, const char*, const char*)
>  
> URLStreamESP32.h: 171:13: error: 'p_urlStream' was not declared in this scope
>    p_urlStream->begin(urlStr, acceptMime, action,reqMime, reqData )
> URLStreamESP32.h: In member function void audio_tools::URLStream::end()
>  
> URLStreamESP32.h: 203:13: error: 'p_urlStream' was not declared in this scope
>    p_urlStream->end()
>  
> AudioTools.h:28: In file included from
> player-sd-i2s.ino:66: from
> AudioPlayer.h: In member function virtual Stream* audio_tools::AudioSourceURL::selectStream(int)
>  
> AudioPlayer.h: 307:44: error: call to 'void audio_tools::URLStream::begin(const char*, const char*, int, const char*, const char*)' uses the default argument for parameter 3, which is not yet defined
>    actual_stream->begin(urlArray[pos], mime)
> Error compiling project sources
> Debug build failed for project 'player-sd-i2s'

I don't know there can be a work arround. If not I just leave it uncomment and import wifi and wificlientsecure

podaen, 2021-11-07:
The player seem to play a file I have deleted from my sd card. :-/ Is is still on the sd card but in a subdirectory

Does this contain the filepath?

[I] AudioPlayer.h : 207 - -> isValidAudioFile: '001.mp3': 1

after that it gives a backtrace

```
Exception:Error:
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715:::0x400851c4:invoke_abort
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715:::0x40085441:abort
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_platform.h:54:::0x4008905f:multi_heap_assert
(inlined by) multi_heap_malloc_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:432
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:305:::0x40089614:multi_heap_malloc
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:215:::0x400815d9:heap_caps_malloc
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:325:::0x400817b5:heap_caps_calloc
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1330:::0x400f0530:i2s_create_dma_queue
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1330:::0x400f1e9c:i2s_set_clk
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1330:::0x400f268d:i2s_driver_install
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d2e25:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d2f9d:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d2515:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d23d5:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d1dd3:libhelix
E:\Documents\Visual Studio 2019\Projects\Arduino\player-sd-i2s\Libraries\arduino-libhelix\0.1\arduino-libhelix\src/MP3DecoderHelix.h:111:::(inlined by) libhelix:
E:\Documents\Visual Studio 2019\Projects\Arduino\player-sd-i2s\Libraries\arduino-libhelix\0.1\arduino-libhelix\src/CommonHelix.h(206): error 0x400d4242:libhelix
E:\Documents\Visual Studio 2019\Projects\Arduino\player-sd-i2s\Libraries\arduino-libhelix\0.1\arduino-libhelix\src/CommonHelix.h(119): error 0x400d42b2:libhelix
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d2225:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioPWM/PWMAudioESP32.h(154): error 0x400d21d7:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioCopy.h(202): error 0x400d4ff1:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioCopy.h(88): error 0x400d5106:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioCopy.h(285): error 0x400d5198:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools/AudioPlayer.h:610:::(inlined by) audio_tools:
E:\Documents\Visual Studio 2019\Projects\Arduino\player-sd-i2s\player-sd-i2s/player-sd-i2s.ino(104): error 0x400d52ba:loop()
C:\Users\Dave2Moon\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/main.cpp(23): error 0x400ddbe1:loopTask(void*)
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x40086492:vPortTaskWrapper
0x40078000: ?? ??:0
0x40080400: ?? ??:0
0x400806a8: ?? ??:0
```

pschatzmann, 2021-11-08:
The AudioSourceSdFat uses all files in the indicated directory and it's subdirectories.
So if you indicate the root as start and just move it to a subdirectory, it will still be found...

podaen, 2021-11-10:
The player does not go correctly to the next file and eventually get stuck.

> Opening port
> Port open
> [I] I2SESP32.h : 244 - rx/tx mode: TX
> [I] I2SESP32.h : 245 - sample rate: 44100
> [I] I2SESP32.h : 246 - bits per sample: 16
> [I] I2SESP32.h : 247 - number of channels: 2
> [I] I2SESP32.h : 248 - is_master: Master
> [I] I2SESP32.h : 249 - mode: 0
> [I] I2SESP32.h : 253 - pin bck_io_num: 26
> [I] I2SESP32.h : 254 - pin ws_io_num: 25
> [I] I2SESP32.h : 255 - pin data_num: 22
> [I] AudioPlayer.h : 213 - -> isValidAudioFile: 'test.txt': 0
> [I] AudioPlayer.h : 225 - -> getFile: '': 0
> [W] AudioPlayer.h : 175 - -> selectStream: 0 ''
> [W] AudioPlayer.h : 515 - -> begin: no data found
> [W] AudioPlayer.h : 632 - -> timeout - moving to next stream
> [W] AudioPlayer.h : 166 - -> nextStream: 1
> [I] AudioPlayer.h : 213 - -> isValidAudioFile: 'test.txt': 0
> [I] AudioPlayer.h : 225 - -> getFile: '': 1
> [W] AudioPlayer.h : 175 - -> selectStream: 1 ''

Ps: I am looking now how to list the indexs and filesnames to know what comes next.

pschatzmann, 2021-11-10:
Can you describe your file structure, so that I can replicate the issue. 
I did not run into any issues so far..

podaen, 2021-11-10:
it's an sd.h example

can't dir and index the sdfat for now.

> Opening port
> Port open
> �SD Card Type: SDHC
> SD Card Size: 30436MB
> Listing directory: /
>   FILE: /test.txt  SIZE: 1048576
>   FILE: /foo.txt  SIZE: 13
>   DIR : /22
>   DIR : /01
>   DIR : /02
>   DIR : /03
>   DIR : /04
>   DIR : /05
>   DIR : /06
>   DIR : /07
>   DIR : /08
>   DIR : /09
>   DIR : /10
>   DIR : /21
>   DIR : /23
>   DIR : /24
>   DIR : /25
>   DIR : /26
>   DIR : /27
>   DIR : /28
>   DIR : /29
>   DIR : /30
>   FILE: /5.MP3  SIZE: 8731800
>   FILE: /4.MP3  SIZE: 6221683
>   FILE: /3.MP3  SIZE: 3510577
>   FILE: /010.MP3  SIZE: 3230947
>   FILE: /14.MP3  SIZE: 5714736
>   FILE: /015.MP3  SIZE: 7479477
>   FILE: /2.MP3  SIZE: 29894
>   FILE: /001.MP3  SIZE: 218437
> Creating Dir: /mydir
> Dir created
> Listing directory: /
>   FILE: /test.txt  SIZE: 1048576
>   FILE: /foo.txt  SIZE: 13
>   DIR : /mydir
>   DIR : /22
>   DIR : /01
>   DIR : /02
>   DIR : /03
>   DIR : /04
>   DIR : /05
>   DIR : /06
>   DIR : /07
>   DIR : /08
>   DIR : /09
>   DIR : /10
>   DIR : /21
>   DIR : /23
>   DIR : /24
>   DIR : /25
>   DIR : /26
>   DIR : /27
>   DIR : /28
>   DIR : /29
>   DIR : /30
>   FILE: /5.MP3  SIZE: 8731800
>   FILE: /4.MP3  SIZE: 6221683
>   FILE: /3.MP3  SIZE: 3510577
>   FILE: /010.MP3  SIZE: 3230947
>   FILE: /14.MP3  SIZE: 5714736
[... 335 lines cut ...]
>   FILE: /30/006.MP3  SIZE: 4327495
>   FILE: /30/007.MP3  SIZE: 9386500
>   FILE: /30/008.MP3  SIZE: 11506708
>   FILE: /30/009.MP3  SIZE: 6294857
>   FILE: /30/010.MP3  SIZE: 8834582
>   FILE: /5.MP3  SIZE: 8731800
>   FILE: /4.MP3  SIZE: 6221683
>   FILE: /3.MP3  SIZE: 3510577
>   FILE: /010.MP3  SIZE: 3230947
>   FILE: /14.MP3  SIZE: 5714736
>   FILE: /015.MP3  SIZE: 7479477
>   FILE: /2.MP3  SIZE: 29894
>   FILE: /001.MP3  SIZE: 218437
> Writing file: /hello.txt
> File written
> Appending to file: /hello.txt
> Message appended
> Reading file: /hello.txt
> Read from file: Hello World!
> Deleting file: /foo.txt
> File deleted
> Renaming file /hello.txt to /foo.txt
> File renamed
> Reading file: /foo.txt
> Read from file: Hello World!

pschatzmann, 2021-11-10:
2 things are different from my setup. I did not have any txt files - so I will make a test with this. And your extensions can be in caps. I will need to extend my logic to cope with this

podaen, 2021-11-10:
Does this in the sd.h library give me the index?

			//Serial.print(" POSITION: ");
			//Serial.println(file.position());

I have updated my dir above. You will see all positions are 0. I think they are not indexed yet...

pschatzmann, 2021-11-10:
No, i am walking the file trie recursively until I found the requested position.

ps. I am not using the SD.h library but the SDFAT library
https://www.arduino.cc/en/reference/SD
https://github.com/greiman/SdFat
but the mechanisms are the same...

pschatzmann, 2021-11-10:
I just committed some corrections for the navigation and made the file extension insensitive to caps...

podaen, 2021-11-10:
Ok, I will take a look. I know now how to work with the index in shfat library. 

```
  File dir;
  File file;
  // open the directory containing the file.
  if (!dir.open(dirPath)) { // handle error}
  // Use name of file in the directory - replace O_RDWR with desired options
  if (!file.open(&dir, fileName, O_RDWR)) { // handle error}
  // use uint16_t for FAT, uint32_t for exFAT
  uint32_t index = file.dirIndex();
  file.close();
  // ...
  // open later by index
  if (!file.open(&dir, index, O_RDWR)) { // hnadle error}
```

pschatzmann, 2021-11-10:
In my case I have a complex directory structure (comping from ITunes) with authors, albums and songs and I wanted to have a way to deal with all files in all subdirectories. So if I point to root, I should get all songs of all authors. If I start at an author directory, I get the songs only of this author...

podaen, 2021-11-10:
First thing you need to know with fat you have.

This is an example of a FAT32 witch means that the index is a uint32_t 

[image]

podaen, 2021-11-12:
On what framework IDF are you are now? I cann't acces strcmp_i in 1.0.6...

podaen, 2021-11-12:
Ok, I see now. You made it yourself. Strong move.

podaen, 2021-11-25:
With playing by path, it possible to da that with storing only the position in to the memorie I think. Speed is not an issue. Speed is not an issue wirh playing songs if we think that evry song plays for at least a second.

closed 2021-11-30 by podaen