pschatzmann/arduino-audio-tools #41: which MCU can we use with this library ?

deladriere, 2021-11-09:
Sorry, not really an issue, but which MCU can we use with this library ?
Can I use it on the SAMD21, or it's only for the ESP32?

I want to run the SAM text to speech on the Feather M0
Thanks

pschatzmann, 2021-11-09:
Unfortunately I do not own a SAMD21 processor. If I remember right I have added some logic which should support I2S, but this has not been tested.

You would make me very happy if you pick this up and try to make it work...

deladriere, 2021-11-09:
SAMD21 has an internal DAC, so maybe this should work, but I don't know (yet) how to do.

pschatzmann, 2021-11-09:
I did a quick google search and I think the easiest would be to use the callbacks from SAM and feed them into the Adafruit I2S library....
https://github.com/adafruit/Adafruit_ZeroI2S

podaen, 2021-11-15:
They are not soo supportive when it comes to software. I would choose an other library for the case you have bugs.

closed 2022-04-20 by pschatzmann