pschatzmann/arduino-audio-tools #42: streams-generator-a2dp.ino - not working

craiglindley, 2021-11-15:
I finally got back to this project and was hoping that the recent changes to the library would fix the noisy output I was getting from the bluetooth connection. Instead, now, I cannot get bluetooth to work at all. My code didn't work at all so I brought up this example program to see if it would work. I only changed 2 things in the example.

1. The name of my bluetooth amplifier to "KD-240WF"
2. I had to change  #include  "AudioA2DP.h"  to #include "AudioLibs/AudioA2dp.h" to get the code to compile

Here is the logging info which indicates the bluetooth connection was made but I don't hear anything from my speaker.

[I] AudioA2DP.h : 129 - Connecting to KD-240WF
[I] AudioA2DP.h : 136 - Starting a2dp_source...
[W] AudioA2DP.h : 292 - ==> state: Connecting
[W] AudioA2DP.h : 292 - ==> state: Disconnected
[W] AudioA2DP.h : 292 - ==> state: Connecting
[W] AudioA2DP.h : 292 - ==> state: Connected
[I] AudioA2DP.h : 145 - a2dp_source is connected...
A2DP is connected now...
[I] AudioTypes.h : 33 - sample_rate: 44100
[I] AudioTypes.h : 34 - channels: 2
[I] AudioTypes.h : 35 - bits_per_sample: 16

and no sound is produced at all; not a click or a pop or noise or anything. And yes the volume is turned up.

I have not idea what the issue is.

pschatzmann, 2021-11-15:
I am quite happy with the current A2DP functionality and I have even an MP3 SD player working nicely with the A2DP output. 

Which ESP32 Arduino version are you using ? 
I will try to retest the sketch with the same version...

craiglindley, 2021-11-15:
I'm using version 1.8.16 of the Arduino IDE

in preferences I have this line:
https://dl.espressif.com/dl/package_esp32_index.json

Is this the info you need?

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

pschatzmann, 2021-11-15:
No, open -> tools -> board -> the board manager 
and enter ESP32

you will see esp32 by Espressif Systems version [version number] e.g. 2.0.1

craiglindley, 2021-11-15:
1.0.6

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

craiglindley, 2021-11-15:
Are there other libraries your code uses that I might have old versions of?

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

pschatzmann, 2021-11-15:
For me the sketch was even crashing. I suggest you update the A2DP library and the Arduino Audio Tools to the latest version. 

The update sketch is working now...

craiglindley, 2021-11-15:
OK with the changes you made your example program is now working.
Thank you. I will now try my own code.

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

pschatzmann, 2021-11-16:
I would suggest to start simple and increase complexity in steps. So first instead of going directly for A2DP I would use the CsvStream to output the data. This way you see if the communication is working properly and in the Serial Plotter you see if the quality is ok. 

Don't forget to make e.g. the sender the master (is_master=true) and the other the client (by setting is_master=false).

If you set the log level to info and you see the copy in the log, you know that data is received/processed.

craiglindley, 2021-11-16:
OK I am now back to where I was. esp#1 is sending a sine wave out the i2s
interface and esp32#2 is receiving it on its i2s interface. I used the
CsvStream to view the received data on the serial plotter and can see the
sine wave is broken up and not continuous. When I listen to this I hear the
sine wave but I also hear a bunch of crackles and pops which I believe
correspond to the tearing I see of the sine wave. If I had to guess I would
say a buffer somewhere is not functioning correctly. Maybe the buffer is
filling but not being emptied fast enough so data is being thrown out. I'm
trying to run at 44,100 in stereo so there is a lot of data to move. Both
esp32's are identical and running from the same power supply.

Any ideas where to look next?

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

pschatzmann, 2021-11-17:
If you think that it is related to the speed you can try to decrease the sample_rate: eg to 22050.

I think it is really difficult to figure out what the issue is. I was working today on a ADC and only could make it work by using the apll. If you update to the latest version you can try to use
```
    cfg.use_apll = true;

```
in both sketches. This might help to prevent synchronization issues.

Apart from this there are a couple of settings that you can play with in AudioConfig.h
- I2S_BUFFER_SIZE: number of samples that are buffered
- I2S_BUFFER_COUNT: number of buffers
- DEFAULT_BUFFER_SIZE: used in the copy to define the number of bytes per copy 
- COPY_DELAY_ON_NODATA: milliseconds delay if the copy did not receive any data

craiglindley, 2021-11-17:
OK, I reduced the data rate to 22050 and set apll to true and increased the
I2S_BUFFER_COUNT to 8 and this had a small but noticeable difference. I ran
the CsvStream test again and grabbed the output which I have attached.
Maybe you can explain what I am seeing.

Thanks Phil for your attention to this. I'm sure you have other things to
do.

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.

pschatzmann, 2022-02-17:
closed due to inactivity

closed 2022-02-17 by pschatzmann