pschatzmann/arduino-audio-tools #43: New DAC for your wiki

clarkbremer, 2021-11-18:
Hi Phil.  I just connected the Hiletgo DAC module to my ESP32, and have it working with your library.  I made a branch of your wiki and updated the page, but since I'm not a collaborator, I can't push it up to make a PR.  Are you interested in including it, and if so, how should we proceed.

pschatzmann, 2021-11-18:
If this is just abut updating the Wiki, I suggest that you enter the information about the module in a new discussion. I will take it up from there.

clarkbremer, 2021-11-18:
### HiLetgo PCM5102 I2S IIS Lossless Digital Audio DAC Decoder

[image]

This DAC is very easy to use: Just connect the 3 I2S pins and the power and everything works as expected, w/o changing the configuration!  SCK to ground seems to reduce noise.

DAC  |	ESP32
-----|----------------
SCK  |	GND
BCK  |	BCK (GPIO26)
DIN  |	OUT (GPIO22)
LCK  |	WS (GPIO25)
GND  |	GND
VIN  |	5V

pschatzmann, 2021-11-19:
I have updated the Wiki

closed 2021-11-19 by pschatzmann