pschatzmann/arduino-audio-tools #45: Difficulties with intellisense

podaen, 2021-11-21:
I am using visual studio with the Visual Micro Extension and have a long time difficulties within this library with Intellisense. But now It got worse... I cann't get it right anymore, the autofillin is gone.

What I think is... that it has something to do with the compiler.

I see changes when I comment or change these.

AudioLogger::instance().begin(Serial, AudioLogger::Info);

OR

Visual Micro in the menu-> Core Debug level

podaen, 2021-11-21:
I did a lookup and it seems I can do something first in Visual Micro. So I will do that before I am coming back.

podaen, 2021-11-23:
Phill, could you test if you have the same issues with the lastest library?

When I type "player." I don't have to autofillin anymore of the properties of the player. I have tested my url stream player and had the same issues that was working before.

podaen, 2021-11-23:
I looked into the library and I see there are no fundamental reasons that intellisens stops working within the labrary. Only when I turn on the warnings I see these. It is not that it isn't compiling or running. But these can be related with the autofillin and sometimes strange behavor of the code reqonizion.

> AudioStreams.h: 578:40: error: deleting object of polymorphic class type 'audio_tools::TimerAlarmRepeatingESP32' which has non-virtual destructor might cause undefined behaviour [-Werror=delete-non-virtual-dtor]
>    if (timer!=nullptr) delete timer
> 
> I2SESP32.h: 57:7: error: narrowing conversion of '(uint32_t)cfg.audio_tools::I2SConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Werror=narrowing]
> 
> I2SESP32.h: 57:7: error: narrowing conversion of '(cfg.audio_tools::I2SConfig::use_apll ? (((uint32_t)cfg.audio_tools::I2SConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate) * cfg.audio_tools::I2SConfig::apll_frequency_factor) : 0u)' from 'unsigned int' to 'int' inside { } [-Werror=narrowing]
>  
> AnalogAudioESP32.h: 206:9: error: narrowing conversion of '(uint32_t)cfg.audio_tools::AnalogConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Werror=narrowing]
> 
> MetaData.h: 25:35: error: deleting object of abstract class type 'audio_tools::AbstractMetaData' which has non-virtual destructor will cause undefined behaviour [-Werror=delete-non-virtual-dtor]
>    if (meta!=nullptr) delete meta
> 
> MetaData.h: 59:21: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
>    if (callback!=nullptr & meta != nullptr) {
>  
> AudioEffectsSuite.h: 980:82: error: base 'audio_tools::ModulationBaseClass' will be initialized after [-Werror=reorder]
>    DelayEffectBase(effectsuite_t(extSampleRate) * .031), SimpleLPF(0.0001, 4) {
> 
> AudioEffectsSuite.h: 980:82: error:   base 'audio_tools::DelayEffectBase' [-Werror=reorder]
> 
> AudioEffectsSuite.h: 978:3: error:   when initialized here [-Werror=reorder]
>    SimpleChorus(int extSampleRate=44100)
> 
> SoundGenerator.h: 81:13: error: statement has no effect [-Werror=unused-value]
>    def.bits_per_sample!=sizeof(T)*8

pschatzmann, 2021-11-23:
I am afraid that I haven's used Windows for more then 10 years now.
Visual Micro is definitely not supported here...

See my comments marked after =>
AudioStreams.h: 578:40: error: deleting object of polymorphic class type 'audio_tools::TimerAlarmRepeatingESP32' which has non-virtual destructor might cause undefined behaviour [-Werror=delete-non-virtual-dtor]
if (timer!=nullptr) delete timer
=> I will take care of this. Though in real life I don't expect this to be an issue in Arduino because objects usually are never destroyed.

I2SESP32.h: 57:7: error: narrowing conversion of '(uint32_t)cfg.audio_tools::I2SConfig::.audio_tools::AudioBaseInfo::sample_rate' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Werror=narrowing]
=> this depends on the IDF version. I was making sure to comply with the latest defintion.

I2SESP32.h: 57:7: error: narrowing conversion of '(cfg.audio_tools::I2SConfig::use_apll ? (((uint32_t)cfg.audio_tools::I2SConfig::.audio_tools::AudioBaseInfo::sample_rate) * cfg.audio_tools::I2SConfig::apll_frequency_factor) : 0u)' from 'unsigned int' to 'int' inside { } [-Werror=narrowing]
=> this depends on the IDF version. I was making sure to comply with the latest defintion.

AnalogAudioESP32.h: 206:9: error: narrowing conversion of '(uint32_t)cfg.audio_tools::AnalogConfig::.audio_tools::AudioBaseInfo::sample_rate' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Werror=narrowing]
=> this depends on the IDF version. I was making sure to comply with the latest defintion.

MetaData.h: 25:35: error: deleting object of abstract class type 'audio_tools::AbstractMetaData' which has non-virtual destructor will cause undefined behaviour [-Werror=delete-non-virtual-dtor]
if (meta!=nullptr) delete meta
=> I will take care of this. Though in real life I don't expect this to be an issue in Arduino because objects usually are never destroyed.

MetaData.h: 59:21: error: suggest parentheses around comparison in operand of '&' [-Werror=parentheses]
if (callback!=nullptr & meta != nullptr) {
=> this is a nasty one!, will be corrected

AudioEffectsSuite.h: 980:82: error: base 'audio_tools::ModulationBaseClass' will be initialized after [-Werror=reorder]
DelayEffectBase(effectsuite_t(extSampleRate) * .031), SimpleLPF(0.0001, 4) {
=> I took this logic from another library - I will correct this.

AudioEffectsSuite.h: 980:82: error: base 'audio_tools::DelayEffectBase' [-Werror=reorder]
=> I took this logic from another library - I will correct this.

AudioEffectsSuite.h: 978:3: error: when initialized here [-Werror=reorder]
SimpleChorus(int extSampleRate=44100)
=> I took this logic from another library - I will look into this.

SoundGenerator.h: 81:13: error: statement has no effect [-Werror=unused-value]
def.bits_per_sample!=sizeof(T)*8
=> will be corrected

pschatzmann, 2021-11-23:
I just committed the correction which should take care of the warnings....

podaen, 2021-11-23:
All errors are gone now. And it is compiling with the warnings on but still no autofillin on "player."

podaen, 2021-11-23:
I did some tests on an other machine and I had no issues on that. I am looking in to what can be problem.

closed 2021-11-23 by podaen

podaen, 2021-11-24:
I found the issue after removing all software and SDK for my computer and reinstalled it. But it wasn't that. I was out of idees an take a look at the audiotools.h file and I removed the namespace form it. And voila, now it's working. I think you have cleaned a bit too much. ;-)

//namespace audio_tools {

#if defined(__linux__) || defined(_WIN32) || defined(__APPLE__)
typedef PortAudioStream DefaultStream;
#elif defined(ESP32) || defined(ESP8266) || defined(__SAMD21G18A__)
typedef I2SStream DefaultStream;
#endif

//}