pschatzmann/arduino-audio-tools #46: player sd a2dp

yannk0, 2021-11-23:
for an installation i need to send mp3 from sd card to a bluetoothspeaker/receiver. the code compiles so far but the programm seems to stop at one point. the console shows:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4

(stops here, not working)

when i use streams-generator a2dp, it says:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4
[W] AudioA2DP.h : 292 - ==> state: Connecting
[W] AudioA2DP.h : 292 - ==> state: Connected
A2DP is connected now...
[W] AudioA2DP.h : 201 - is_a2dp_active -> true with 63488 bytes

( this works)

the libs are all latest from git repo...arduino 11.8.16 expressive 2.0.1

any hints?

pschatzmann, 2021-11-23:
Set the loglevel to debug and then you should see in with method it get's stuck.
Are you sure that the SD is working properly ?

yannk0, 2021-11-23:
nopt 100% sure. sdtest shows the files......but...you  write somewhere  to connect it to 5v while in some other howto he says the card must be connected to 3.3v. (i soldered the cardadapter directly to the esp32)

pschatzmann, 2021-11-23:
Maybe this depends on the module. I tried it first with 3.3V and for me this did not work...

yannk0, 2021-11-23:
they say the sdcard will die of its on 5V...(?) 
how do i set the loglevel to debug for the esp32?

pschatzmann, 2021-11-23:
AudioLogger::instance().begin(Serial, AudioLogger::Debug);

And did you double check the chipSelect  and speedMHz  parameters of the AudioSourceSdFat class ?

yannk0, 2021-11-23:
where do i do that?
i only installed the dependency libs and choose my board, doit esp32 devkit v1...i am not aware of further steps...

pschatzmann, 2021-11-23:
Just use the  player-sd-a2dp example and adapt the log level...

yannk0, 2021-11-23:
i see no difference, 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4

it does compile but there are quite a lot of messages i the compiler window. post them here? 

i might try/risk the card on 5v ( later)...it seems as if it does list the files but play them gets stuck...

pschatzmann, 2021-11-23:
Look at the spec of your module. I am using this one https://hobbycomponents.com/adapters/578-microsd-card-adapter-with-level-shifters - and here the power supply must be 4.5V ~ 5.5V.

What messages are you getting ?

yannk0, 2021-11-23:
In file included from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SStream.h:9,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:17,
                 from /home/yann/Arduino/000player-sd-a2dp/000player-sd-a2dp.ino:14:
/home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h: In member function 'void audio_tools::I2SBase::begin(audio_tools::I2SConfig)':
/home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h:55:27: warning: narrowing conversion of '(cfg.audio_tools::I2SConfig::use_apll ? (((uint32_t)cfg.audio_tools::I2SConfig::<anonymous>.audio_tools::AudioBaseInfo::sample_rate) * cfg.audio_tools::I2SConfig::apll_frequency_factor) : 0)' from 'eps32_i2s_sample_rate_type' {aka 'unsigned int'} to 'int' inside { } [-Wnarrowing]
             .fixed_mclk = (eps32_i2s_sample_rate_type) (cfg.use_apll ? cfg.sample_rate * cfg.apll_frequency_factor : 0 )
                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h: In member function 'i2s_comm_format_t audio_tools::I2SBase::toCommFormat(audio_tools::I2SFormat)':
/home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h:226:40: warning: 'I2S_COMM_FORMAT_PCM_LONG' is deprecated [-Wdeprecated-declarations]
             return (i2s_comm_format_t) I2S_COMM_FORMAT_PCM_LONG;
                                        ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/yann/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h:8,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SStream.h:9,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:17,
                 from /home/yann/Arduino/000player-sd-a2dp/000player-sd-a2dp.ino:14:
/home/yann/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:100:5: note: declared here
     I2S_COMM_FORMAT_PCM_LONG  __attribute__((deprecated)) = 0x08, /*!< PCM Long, (I2S_COMM_FORMAT_PCM | I2S_COMM_FORMAT_PCM_LONG) correspond to `I2S_COMM_FORMAT_STAND_PCM_LONG`*/
     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SStream.h:9,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:17,
                 from /home/yann/Arduino/000player-sd-a2dp/000player-sd-a2dp.ino:14:
/home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h:228:40: warning: 'I2S_COMM_FORMAT_PCM_SHORT' is deprecated [-Wdeprecated-declarations]
             return (i2s_comm_format_t) I2S_COMM_FORMAT_PCM_SHORT;
                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/yann/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/driver/include/driver/i2s.h:16,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SESP32.h:8,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioI2S/I2SStream.h:9,
                 from /home/yann/Arduino/libraries/arduino-audio-tools/src/AudioTools.h:17,
                 from /home/yann/Arduino/000player-sd-a2dp/000player-sd-a2dp.ino:14:
/home/yann/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32/include/hal/include/hal/i2s_types.h:99:5: note: declared here
     I2S_COMM_FORMAT_PCM_SHORT __attribute__((deprecated)) = 0x04, /*!< PCM Short, (I2S_COMM_FORMAT_PCM | I2S_COMM_FORMAT_PCM_SHORT) correspond to `I2S_COMM_FORMAT_STAND_PCM_SHORT`*/
     ^~~~~~~~~~~~~~~~~~~~~~~~~
Der Sketch verwendet 1198793 Bytes (91%) des Programmspeicherplatzes. Das Maximum sind 1310720 Bytes.
Globale Variablen verwenden 37996 Bytes (11%) des dynamischen Speichers, 289684 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.

yannk0, 2021-11-23:
i just used the cardholder that came with the sd card. no extra module

yannk0, 2021-11-23:
i read the word "format" quite often

pschatzmann, 2021-11-23:
I suggest to try some simple SD audio example first and then only move to A2DP when this is working:
 e,g, player-sd-analog. This one you can check with some simple earphones.

Did you double check the chipSelect parameter of the AudioSourceSdFat class ?

yannk0, 2021-11-23:
"Did you double check the chipSelect parameter of the AudioSourceSdFat class ?"
where?

pschatzmann, 2021-11-23:
If you look in the constructor the chipSelect is assigned some default value. I think it is  VSPI-CS0 (GPIO 05)
This needs to correspond with your wiring..

yannk0, 2021-11-23:
with player-sd-analog its shows the same console messages...maybe that indicates some issue with card?
could that stop the programmloop?

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1284
load:0x40078000,len:12836
load:0x40080400,len:3032
entry 0x400805e4

pschatzmann, 2021-11-23:
Yes, If I remember right I had the same problem util I could make the SD work: I needed to decrease the speed.
But I doubt that this will help you because the current value is already very slow.

https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_audio_source_sd_fat.html

yannk0, 2021-11-23:
the guy in the direct connect howto says something about a variation to connect the card spi-mmc or such that would be much faster. could that be the reason, that the card is just not fast enought for audio processing in this mode?

yannk0, 2021-11-23:
https://www.youtube.com/watch?v=e1xOgZsnAuw

yannk0, 2021-11-23:
yes

pschatzmann, 2021-11-23:
Right now I am clueless. You could try to move the global variables into the setup after the logger setup and comment out the loop. Just in the hope that you might see some error messages this way....

yannk0, 2021-11-23:
ok ...i try some more things....ask more later

yannk0, 2021-11-23:
one last question..if i connect the card in sdmmc mode..would the player sketch work with it as it is?

pschatzmann, 2021-11-23:
That would be a question for https://github.com/greiman/SdFat.
I am just doing the following in my code of FileAccessSdFat
sd.begin(SdSpiConfig(chipSelect, DEDICATED_SPI, SD_SCK_MHZ(speedMHz))

yannk0, 2021-11-23:
ok, i see. is this speed enough for the sd-a2dp function?:
1048576 bytes read for 2890 ms
1048576 bytes written for 4696 ms

pschatzmann, 2021-11-23:
I think so, if i calclated correctly that would give 93000 samples per second.
I think you should concentrate on the AudioSourceSdFat class.  Try to  create a simple sketch with logging activated and and use the AudioSourceSdFat class as local variable and try out some methods...

yannk0, 2021-12-01:
> Maybe this depends on the module. I tried it first with 3.3V and for me this did not work...

i am now back opn this...by some mistake i freid the esp32...got a new one...while testing alternatives...i came to a post about connecting a dfplayer to an arduino...where he wrote the 3.3v out has not enough power for the card reader and connected it via a voltage divider 680ohm/1k  to 5v. maybe thats the reason why my thing di not work. did you use the vin of the esp32 as 5v source?

pschatzmann, 2021-12-01:
I am using a module which has a built in voltage regulator. https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sd-a2dp

In the meantime I have also changed the logic, so that the card initialization is not happening in the constructor any more. So you might get some error messages with the latest version.

yannk0, 2021-12-01:
i now even hear a sound coming out of the speaker(!)..but only a few fragments the it sops...and this in the console:
AudioA2DP.h : 203 - Waiting for free buffer space - available: 62976

pschatzmann, 2021-12-01:
Try to deactivate the logging both in Arduino and in the Sketch by setting the log level to Warning.

yannk0, 2021-12-01:
its working now!

pschatzmann, 2021-12-01:
Cool

yannk0, 2021-12-01:
i need to let it choose one random track out of the mp3 folder, fade in..and then play tracks as long the installtions runs ( depending on the timer) and then fade out...maybe some tips for the commands to the player?

pschatzmann, 2021-12-01:
What was the solution for the SD drive ?

yannk0, 2021-12-01:
> Cool

yes...

pschatzmann, 2021-12-01:
Look at the documentation:
https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_audio_player.html
https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_audio_source_sd_fat.html

yannk0, 2021-12-01:
> What was the solution for the SD drive ?
i used a modul with 5v input connected to the vin...the sd-adapter directly with voltage divider to vin seems to have problems left...

yannk0, 2021-12-01:
i named the tracks 0001.mp3, 0002.mp3 etc but it only play the first track. are there naming rules?

yannk0, 2021-12-01:
it stops saying this:
[W] AudioPlayer.h : 754 - -> timeout - moving to next stream
[W] AudioPlayer.h : 215 - -> nextStream: 1
[W] AudioPlayer.h : 224 - -> selectStream: 1 ''
 where to find the commands to communicated with the player?...i am a bit lost in the documentation

pschatzmann, 2021-12-02:
Unfortunately I dont have any time today to look into this. 
But since you plan to implement some random access you might consider to use the CallbackSource
https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_audio_source_callback.html

The player automatically calls selectStream(0) for the first entry and then nextStream() to get the next files

pschatzmann, 2021-12-04:
I just committed a correction. I hope this helps in your case as well

yannk0, 2021-12-07:
the player now plays the other tracks too. problem is the sketch is combined with the RBD-motor library for to slowly spin up the motor while play the tracks ( chants of the humpback whale) out of the mp3 folder and fade out when the motor stops. it turn a three aces rotating object with the bt speaker inside. the ramp function is gone when the play plays. also because the cods gets big and should have place for ota as well i thought to split it for the esp32 for the sou and a nano for the motor...but how to do the connection? the player also slows down uart ( because  i do it wrong) here the sketch:

[RBDmotorSdA2dpPlayer_ino.txt](https://github.com/pschatzmann/arduino-audio-tools/files/7670285/RBDmotorSdA2dpPlayer_ino.txt)

yannk0, 2021-12-07:
here the sketch with player and uart

esp32 receiver:

[SdA2dpPlayer_uart_esp32_nano_ino.txt](https://github.com/pschatzmann/arduino-audio-tools/files/7670306/SdA2dpPlayer_uart_esp32_nano_ino.txt)
 
nano sender:

[nano_uart_esp32_sender_ino.txt](https://github.com/pschatzmann/arduino-audio-tools/files/7670335/nano_uart_esp32_sender_ino.txt)

yannk0, 2021-12-07:
maybe uart is not the right thing so i thought best would be spi as the sd is an spi device already but where to insert the nano as new divice? the xamples are not really helpful...or don't work

yannk0, 2021-12-07:
or even better i2c as i learned that its no problem to have both at the same time?

pschatzmann, 2021-12-07:
I rather tend to think that the problem is the motor library and that you might be able to do all on the ESP32.
I tend to think you would be better off if you use something which is based on the built in ESP32 pwm functionality.
If the sketch is getting too big you can select Partition Scheme Hugh App ...

If you want to connect 2 devices, Serial is the simplest option.

mario44, 2022-01-08:
When compiling under Arduino the file: player-sd_a2dp-audiokit.ino
I have the following error In function 'void loop()':
67:13: error: could not convert 'player' from 'audio_tools :: AudioPlayer' to 'bool'

mario44, 2022-01-09:
I changed the player to player_active :
```
void loop() {
  if (player_active) {
    player.copy();
  } else {
    // feed watchdog
    delay(10);
  }
  kit.processActions();
}
```

closed 2022-02-17 by pschatzmann