pschatzmann/arduino-audio-tools #54: Base-adc-a2dp has low audio on bletooth heaphones

dev-techshlok, 2021-11-26:
Hello @pschatzmann 

First of all, thank you for providing us with such beautiful work. 

I used the example Base-adc-a2dp, it worked fine. But the audio level on my headphone is very low.  I tried using set Volume to 255 as well but still the same audio. 

Is there anything I can do?

pschatzmann, 2021-11-26:
Have a look at the scaler. If you define a factor > 1.0 it the input is getting amplified. e.g. Try with 1.0, 2.0 until you get some acceptable sound. If the factor is too big, the result is getting clipped which leads to a bad audio quality....

dev-techshlok, 2021-11-27:
In which file, can you give a small example, please?

pschatzmann, 2021-11-27:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-adc-a2dp

dev-techshlok, 2021-11-27:
Thank you for the fastest response.

If you don't mind I have a few more questions.

1. I changed the factor to 2.0, the volume got increased but as you said noise increased also LEFT Chanel on headphones starts to fade in low volume. When I changed to factor to 5 on LEFT there was no audio but the right had full volume.

2. Also unable to use GPIO 34 as no audio is received with ADC to Serial plotter. Only receives data on GPIO35 pin of esp.

3. Also, these are the values I am getting at CSVstreamer
```

[I] AudioCopy.h : 282 - StreamCopy::copy 1024 bytes - in 1 hops
-691, -691
-691, -691
-691, -697
-697, -698
-698, -705
-705, -691
-691, -640
-640, -691
-691, -641
-641, -581
-581, -516
-516, -434
-434, -434
-434, -434
-434, -386
-386, -386
-386, -376
-376, -371
-371, -338
-338, -371
-371, -389
-389, -410
-410, -355
-355, -263
-263, -231
-231, -214
-214, -203
-203, -200
-200, -227
-227, -197
-197, -115
-115, -35
-35, -66
-66, -115
-115, -115
-115, -36
-36, -33
-33, -139
-139, -161
-161, -35
-35, 78
78, 39
39, 14
14, 78
78, 141
141, 124
124, 80
80, 85

```

Any suggestions? 

5. I am connecting two ESPs.  
- First ESP (Receives audio stream on Bluetooth sink mode and transmits over internal DAC on pins 25,26.)
- Second ESP collects the audio stream on ADC pin GPIO35 and sends it to a Bluetooth headphone. 
- It works fine but this has a low volume problem

- Second Scenario was I tried using ESP-ESP with I2S as input-output but it just gave audio with a fading-like effect like once you hear music then noise then again music.

Can you suggest something?

pschatzmann, 2021-11-27:
You use the ConverterScaler<int16_t> scaler(factor, offset, clippingMaxValue (e.g. 32767) ) to get the correct output from the input. It applies the offset first and then calculates the factor. The goals should be that the center is 0 and the result values are between -32768 and 32767 (which would give the maximum volume). Please do not forget to remove any filters and scalers from the CSV Sketch, because you want to see the real values!

If the log above is coming from the ADC, then you really receive 2 channels. If you use the AnalogAudioStream you can recieve the 2 channels with GPIO35 and 36.

Analog output and input has usually noise problems and I would expect that the quality with I2S to be much better. If you have some issues with I2S you could try to use the apll by setting config.use_apll = true.

dev-techshlok, 2021-11-27:
Ok ill try this.
But also can you exmplain a bit about how to use this

pschatzmann, 2021-11-27:
You would set this I2S configuration parameter both in your sending and receiving sketch. 
In my understanding this is using a separate clock with is a bit more accurate...

dev-techshlok, 2021-11-27:
which two examples would be good to use.
ESP 1 - Sink
ESP 2 - Source

Right now using 

ESP 1 - Sink - ESP32A2DP library ([bt_music_receiver_simple](https://github.com/pschatzmann/ESP32-A2DP/blob/main/examples/bt_music_receiver_simple/bt_music_receiver_simple.ino))
ESP 2 - source - [Base_i2s a2dp](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-i2s-a2dp) 

Can you suggest which one should I use. in order to make it work and also use these use_apll settings.

It would be really helpful for me.

Thank you

pschatzmann, 2021-11-27:
ESP1: In the Readme of the ESP32-A2DP project there is an apll example. Just make sure that you use 16 as bits_per_sample and not 32!

ESP2: same: just use
  auto cfg = i2s.defaultConfig(RX_MODE);
  cfg. is_master = false;
  cfg.use_apll = true;
  i2s.begin(cfg);

dev-techshlok, 2021-11-27:
Thank you.
The noise stopped so does the music.
I have modified the code according to your guidance.

**ESP1 code - sink** 

```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {

  static i2s_config_t i2s_config = {
    .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
    .sample_rate = 44100, // updated automatically by A2DP
    .bits_per_sample = (i2s_bits_per_sample_t)16,
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
    .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_STAND_I2S),
    .intr_alloc_flags = 0, // default interrupt priority
    .dma_buf_count = 8,
    .dma_buf_len = 64,
    .use_apll = true,
    .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
  };
 

    
  a2dp_sink.set_i2s_config(i2s_config);
  a2dp_sink.start("MyMusic");
}

void loop() {
}

```

**ESP 2 code - source**

```
/**
   @file base-i2s-a2dp.ino
   @author Phil Schatzmann
   @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/base-i2s-a2dp/README.md

   @author Phil Schatzmann
   @copyright GPLv3
*/

// Add this in your sketch or change the setting in AudioConfig.h
#define USE_A2DP

#include "AudioTools.h"

using namespace audio_tools;

/**
   @brief We use a ADS1015 I2S microphone as input and send the data to A2DP
   Unfortunatly the data type from the microphone (int32_t)  does not match with the required data type by A2DP (int16_t),
   so we need to convert.
*/

BluetoothA2DPSource a2dp_source;
I2S<int32_t> i2s;
ChannelConverter<int32_t> converter(&convertFrom32To16);
ConverterFillLeftAndRight<int32_t> bothChannels;
const size_t max_buffer_len = 1024;
int32_t buffer[max_buffer_len][2];

// callback used by A2DP to provide the sound data
int32_t get_sound_data(Frame* data, int32_t len) {
  size_t req_len = min(max_buffer_len, (size_t) len);

  // the microphone provides data in int32_t -> we read it into the buffer of int32_t data
  size_t result_len = i2s.read(buffer, req_len);

  // we have data only in 1 channel but we want to fill both
  bothChannels.convert(buffer, result_len);

  // convert buffer to int16 for A2DP
  converter.convert(buffer, data, result_len);
  return result_len;
}

// Arduino Setup
void setup(void) {
  Serial.begin(115200);

  // start i2s input with default configuration

  Serial.println("starting I2S...");
  auto cfg = i2s.defaultConfig(RX_MODE);
  cfg.is_master = false;
  cfg.use_apll = true;
  i2s.begin(cfg);

  // start the bluetooth
  Serial.println("starting A2DP...");
  a2dp_source.start("JBL T450BT", get_sound_data);
}

// Arduino loop - repeated processing
void loop() {
}
```

**PIns used ESP 1 - sink** 

```
BCK - 26
WS - 25
Data out - 22
```
**PIns used ESP 2 -Source** 
```
BCK - 14
WS - 15
Data in- 32
```

Can you suggest what might be happening?

pschatzmann, 2021-11-27:
I think the issue is in the second sketch: You should remove all conversion logic. 

But here again I would recommend to first try to make the CSV output work.

I am usually using the Stream API with CsvStream and if this is working as expected I replace it with a A2DPStream: 
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-csv
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-a2dp
Don't use the config settings from the examples but just start with what you currently have.

pschatzmann, 2021-11-27:
I was double checking the functionality and for me everything is working fine.
Are you sure that you used the correct sample rate which is 44100 ?

I updated the example with a correct sound example and also added 8 bit support.

dev-techshlok, 2021-11-29:
Hello Is there any way we can transmit the received audio from I2S to DAC of the same ESP? Because the Example 
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-csv
worked fine.

I was able to receive good looking data on plotter.

dev-techshlok, 2021-11-29:
Hello @pschatzmann 

It worked like charm! like you said. 

I just converted i2s-CSV to i2s-a2dp. All worked perfectly.

Thank you so much.

closed 2021-11-29 by dev-techshlok