pschatzmann/arduino-audio-tools #56: Reset last Bluetooth connection like in ESP32A2dp library set_auto_reconnect(false);

dev-techshlok, 2021-11-29:
Hello @pschatzmann 

I worked with examples for streaming audio from ADC to A2DP. 
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-adc-a2dp

Here when I wanted to reset last Bluetooth connection i used this line.
`a2dp_source.set_auto_reconnect(false);`

Then I switched to I2S on this, which worked well for me.
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-a2dp

But here I don't have the option to stop auto reconnection.

Can you suggest how should I do it?

Thank you.

GuidoMeinke, 2021-11-30:
How did you compile streams-i2s-a2dp? I get some errors because of the folder structure.

pschatzmann, 2021-11-30:
I suggest to use the latest version and then please report the errors that you are getting. It might be that since I have more (optional) libraries installed, I don't catch all issues...

pschatzmann, 2021-11-30:
If I understand correctly there is still the open question how to set the set_auto_reconnect() when using an A2DPStream.
On the Stream you can use the source() method to get the A2DPSource object. so it would be 
```
a2dp_stream.source(). set_auto_reconnect(false);
```

dev-techshlok, 2021-12-01:
Hello @pschatzmann 

That Worked like a charm. Thank you so much.

closed 2021-12-01 by dev-techshlok