pschatzmann/arduino-audio-tools #60: Arduino compiling error

GuidoMeinke, 2021-11-30:
Hello,

when i want to compile your example I get following issue:
`
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
In file included from C:\Users\guido\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioEffects/AudioEffects.h:7:0,
 Benutzt: C:\Users\guido\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
                 from C:\Users\guido\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:29,
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
                 from C:\Users\guido\Documents\Arduino\libraries\arduino-audio-tools-main\examples\examples-stream\streams-i2s-a2dp\streams-i2s-a2dp.ino:13:
C:\Users\guido\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioEffects/STKEffects.h:5:20: fatal error: Effect.h: No such file or directory
compilation terminated.
exit status 1
`
What am i doing wrong?
The way I see it Arduino doesn't really handle the folder structure? Because all the files are there.

I use Arduino 1.8.16 in Windows 10

pschatzmann, 2021-11-30:
I think in the latest version this issue has been resolved...

pschatzmann, 2021-11-30:
Can you maybe give me some lines how you get there because with the current set up this should not be possible.
Have you maybe used a #define USE_STK anywhere in your code ?

GuidoMeinke, 2021-11-30:
just used your example and give it a try.
I did nothing on your code

GuidoMeinke, 2021-11-30:
it is 
`/**
 * ------------------------------------------------------------------------- 
 * @brief Activate Other Tools
 */

//#define USE_STK
//#define USE_SDFAT
//#define USE_DELTASIGMA 
/**`

as i said, out of the zip, give it a try

pschatzmann, 2021-11-30:
I deleted the Arduino-STK library and tested the sketch. It is working fine.
So either you have a #define USE_STK in AudioConfig.h or you work with an old version. 
You can double check this:
If you look in AudioEffect/AudioEffects.h you should have 
```
#pragma once

#include "AudioTools/SoundGenerator.h"
#include "AudioTools/Vector.h"
#include "AudioEffects/AudioEffect.h"
#include "AudioEffects/AudioEffectsSuite.h"
#ifdef  USE_STK
#include "AudioEffects/STKEffects.h"
#endif

namespace audio_tools {

```

pschatzmann, 2021-11-30:
And please note that if you import a zip file, there is not automatic update! So I would recommend to use the git clone method. Then you can update it easily with git update....

GuidoMeinke, 2021-11-30:
thank you very much, you were right i had an old version

pschatzmann, 2021-11-30:
no worries

podaen, 2021-11-30:
You could play with the parameters to get optimal results. And what you don't use you beter comment. As Phill says you could use git to stay up to date with the latest version.

closed 2021-11-30 by pschatzmann