pschatzmann/arduino-audio-tools #62: Print metadata from the sd player

podaen, 2021-12-01:
Soo we have the metadata from the url player working with icy... Now I tried to recieving data in the player-sd-i2s example from printMetaData, but nothing so far.

pschatzmann, 2021-12-01:
That's strange, I remember having tested this and that it was working. 
Are you sure that your mp3 files contain metadata ?
In my streams-url_mp3-metadata2.ino test I point to a file which contains metadata. Maybe you can use that one to double check.

I will retest as soon I have the player setup ready on the Raspberry Pico...

podaen, 2021-12-01:
Yeah, but I diden't test if further. Can you give an example with the callback like you sad here  #27 

> you can do the approach that I used in my player to register a callback in the copy and process the additional output from there

I used the one of the example player, but it isn't working for me...

pschatzmann, 2021-12-01:
the exmple player_sd_i2s was working for me with all my mp3 files.

podaen, 2021-12-01:
I looked and test it again. Not for me.

Only by doing this...

> player.setMetadataCallback(printMetaData);

> void printMetaData(MetaDataType type, const char* str, int len) {
> 	Serial.print("==> ");
> 	Serial.print(MetaDataTypeStr[type]);
> 	Serial.print(": ");
> 	Serial.println(str);
> }

pschatzmann, 2021-12-01:
Did you also test with the file that I have indicated: this definitively has metadata defined.
Right now my guess would be that your files do not contain any metdata.

podaen, 2021-12-01:
This is metadata right?

[image]

> Did you also test with the file that I have indicated

Which file?

podaen, 2021-12-01:
> In my streams-url_mp3-metadata2.ino

Cann't find this one

podaen, 2021-12-01:
> I will retest as soon I have the player setup ready on the Raspberry Pico...

I am working on an esp32.

pschatzmann, 2021-12-01:
Yes I know, but right now I am working on Picos and it does not make any difference.
The file is https://pschatzmann.github.io/arduino-audio-tools/resources/audio.mp3
Can you share your file as well ?

podaen, 2021-12-01:
you are definetly talking about other data than I do. I downloaded the file from your url link and if I do right click - properties... I get no information on Title, album or what ever.

pschatzmann, 2021-12-01:
I am using https://en.wikipedia.org/wiki/ID3
It would be interesting to know what your format is...

podaen, 2021-12-01:
Wait a minute... I have seen this before.

podaen, 2021-12-01:
Now I am confused,

[image]

As you can see below the picture, the file contains ID3 v1 and v2 information. What the information is I can not tell.

pschatzmann, 2021-12-01:
I have implemented all versions, but not tested them all. 
I was happy when all my files were working and called it a day.

The test file I have just produced with some free web service.

podaen, 2021-12-01:
You can make your own attribute information by rightclick on the file and fill in the properties. I think this is the most common...

pschatzmann, 2021-12-01:
Hmm, not on osx and linux...
I haven't used Windows for more the 20 years now.

podaen, 2021-12-01:
I have not much experence on that, but maybe you could download an music player. there you will find properties. I use media monkey.

pschatzmann, 2021-12-01:
Your right in Audacity I don't see any metadata for that file either...

podaen, 2021-12-01:
what I could do is copy the file with filled in attributes to an usb stick and try it on an old auto radio to see if the information is displayed. I know it works with a2dp and spotify. diden't test it with mp3 extensions.

pschatzmann, 2021-12-01:
I was just checking with my other files which also work - and there I see the metadata also in Audacity.
I think the best thing is if you share one of your files which is not working and we concentrate on that

pschatzmann, 2021-12-01:
Just had a quick look at the source code. I am currently only supporting  frame_header.encoding == 0 || frame_header.encoding == 3; which is ASCII and UTF8. I think Windows uses some strange cp1252 ...

podaen, 2021-12-01:
I have send you the file I was talking about.

pschatzmann, 2021-12-01:
I try to access it but it seems that you still need to approve my request...

podaen, 2021-12-01:
Does it work now?

pschatzmann, 2021-12-01:
yes, thanks...

pschatzmann, 2021-12-01:
I tried the player with my files: - it's working
```
19:11:06.798 -> [I] AudioPlayer.h : 768 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int))
19:11:06.798 -> [I] AudioPlayer.h : 270 - directory: /Music
19:11:06.833 -> [I] AudioPlayer.h : 261 - -> isValidAudioFile: '01 ?guas de mar?o.mp3': 1
19:11:06.833 -> [I] AudioPlayer.h : 337 - ==> found: '01 ?guas de mar?o.mp3' at index 0
19:11:06.833 -> [W] AudioPlayer.h : 215 - -> selectStream: 0 '01 ?guas de mar?o.mp3'
19:11:06.833 -> libhelix -  -> invalid frame size: 0 / max: 1600
19:11:06.833 -> ==> Title: Elis & Tom
19:11:06.833 -> ==> Genre: Revival
19:11:06.833 -> [I] AudioCopy.h : 96 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
19:11:06.867 -> libhelix -  -> invalid frame size: 0 / max: 1600
19:11:06.867 -> libhelix -  -> invalid frame size: 0 / max: 1600
19:11:06.867 -> [I] AudioCopy.h : 96 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
19:11:06.867 -> [I] AudioPlayer.h : 633 - sample_rate: 44100
19:11:06.901 -> [I] AudioPlayer.h : 634 - bits_per_sample: 16
19:11:06.901 -> [I] AudioPlayer.h : 635 - channels: 2
19:11:06.901 -> [I] I2SStream.h : 103 - virtual voi
```
I created a test sketch with your file: -> its working as well

```
18:54:10.225 -> [I] I2SConfig.h : 63 - rx/tx mode: TX
18:54:10.225 -> [I] I2SConfig.h : 64 - port_no: 0
18:54:10.225 -> [I] I2SConfig.h : 65 - is_master: Master
18:54:10.225 -> [I] I2SConfig.h : 66 - sample rate: 44100
18:54:10.225 -> [I] I2SConfig.h : 67 - bits per sample: 16
18:54:10.225 -> [I] I2SConfig.h : 68 - number of channels: 2
18:54:10.260 -> [I] I2SConfig.h : 69 - i2s_format: I2S_STD_FORMAT
18:54:10.260 -> [I] I2SConfig.h : 77 - pin_ws: 15
18:54:10.260 -> [I] I2SConfig.h : 78 - pin_bck: 14
18:54:10.260 -> [I] I2SConfig.h : 79 - pin_data: 22
18:54:10.260 -> ==> Title: Talespin (Promo)
18:54:10.260 -> ==> Album: 0sfx Louie's Place
18:54:10.260 -> ==> Genre: Soundtrack
18:54:10.260 -> [I] I2SStream.h : 103 - virtu
```

podaen, 2021-12-03:
I am not getting it... Did you enabled something in the configuration file or in the sdfat library?

pschatzmann, 2021-12-03:
No, I am wondering as well what it could be.
Here is the test sketch: 

```
// set this in AudioConfig.h or here after installing https://github.com/pschatzmann/arduino-libhelix.git
#define USE_HELIX 

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include <SPI.h>
#include <SdFat.h>

using namespace audio_tools;  

//                            -> EncodedAudioStream -> I2SStream
// URLStream -> MultiOutput -|
//                            -> MetaDataPrint

File audioFile;
MetaDataPrint outMeta; // final output of metadata
I2SStream i2s; // I2S output
EncodedAudioStream out2dec(&i2s, new MP3DecoderHelix()); // Decoding stream
MultiOutput out(outMeta, out2dec);
StreamCopy copier(out, audioFile); // copy url to decoder
const int chipSelect=CS;

// callback for meta data
void printMetaData(MetaDataType type, const char* str, int len){
  Serial.print("==> ");
  Serial.print(toStr(type));
  Serial.print(": ");
  Serial.println(str);
}

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  // setup file
  SD.begin(SdSpiConfig(PIN_CS, DEDICATED_SPI, SD_SCK_MHZ(2)));
  //audioFile = SD.open("/Music/Conquistadores.mp3");
  audioFile = SD.open("/test/002.mp3");

  // setup metadata
  outMeta.setCallback(printMetaData);
  outMeta.begin()

  // setup i2s
  auto config = i2s.defaultConfig(TX_MODE);
  i2s.begin(config);

  // setup I2S based on sampling rate provided by decoder
  out2dec.setNotifyAudioChange(i2s);
  out2dec.begin();

}

void loop(){
  copier.copy();
}
```

podaen, 2021-12-03:
Now I see. You need to update the sd example. I think it will work now.

podaen, 2021-12-03:
Cann't make it work for the player

> const char* startFilePath = "/";
> const char* ext = "mp3";
> AudioSourceSdFat source(startFilePath, ext, SD_CS_PIN,16);
> File audioFile;
> MetaDataPrint outMeta; // final output of metadata
> I2SStream i2s;
> VolumeOutput volume(i2s);
> LogarithmicVolumeControl lvc(0.1);
> //MP3DecoderHelix decoder;
> EncodedAudioStream out2dec(&volume, new MP3DecoderHelix()); // Decoding stream
> //AudioPlayer player(source, i2s, decoder);
> AudioPlayer player(source, volume, out2dec);
> MultiOutput out(outMeta, out2dec);
> StreamCopy copier(out, audioFile); // copy url to decoder

out2dec is not valid for the player.

podaen, 2021-12-03:
and if I make it work for the player

> MP3DecoderHelix decoder;
> //EncodedAudioStream out2dec(&volume, new MP3DecoderHelix()); // Decoding stream
> AudioPlayer player(source, volume, decoder);
> MultiOutput out(outMeta, decoder);
> StreamCopy copier(out, audioFile); // copy url to decoder

It doesn't accept MultiOutput out(outMeta, decoder);

pschatzmann, 2021-12-03:
in the player   player.setMetadataCallback(printMetaData); should be good enough. It was working for me in the player_sd_i2s.ino example

podaen, 2021-12-03:
nop

pschatzmann, 2021-12-03:
Can you double check why in your case the AudioPlayer the following code on line 594 is not executed ?
```
                    if (meta_active) {
                        copier.setCallbackOnWrite(decodeMetaData, this);
                    }
```

podaen, 2021-12-03:
it seems that audioplayer is not called. If I do a check when I select stream the meta_active returns 1.

podaen, 2021-12-03:
setMetadataCallback in the audioplayer is called

pschatzmann, 2021-12-03:
if you set the log level to Debug is the decodeMetaData() method called ?
That should happen after every copy step

podaen, 2021-12-03:
This works too in the MetaDataID3 class

```
void begin() {
        Serial.println("test");
        id3v1.begin();
        id3v2.begin();
    }
```

podaen, 2021-12-03:
> Opening port
> Port open
> end setup
> here
> [D] I2SStream.h : 87 - void audio_tools::I2SStream::begin(audio_tools::I2SConfig)
> [D] I2SESP32.h : 38 - void audio_tools::I2SBase::begin(audio_tools::I2SConfig)
> [I] I2SConfig.h : 63 - rx/tx mode: TX
> [I] I2SConfig.h : 64 - port_no: 0
> [I] I2SConfig.h : 65 - is_master: Master
> [I] I2SConfig.h : 66 - sample rate: 44100
> [I] I2SConfig.h : 67 - bits per sample: 16
> [I] I2SConfig.h : 68 - number of channels: 2
> [I] I2SConfig.h : 69 - i2s_format: I2S_STD_FORMAT
> [I] I2SConfig.h : 77 - pin_ws: 25
> [I] I2SConfig.h : 78 - pin_bck: 26
> [I] I2SConfig.h : 79 - pin_data: 22
> [D] I2SESP32.h : 93 - begin - started
> [I] AudioPlayer.h : 779 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int))
> [D] AudioPlayer.h : 599 - virtual bool audio_tools::AudioPlayer::begin(int, bool)
> [D] AudioOutput.h : 310 - void audio_tools::EncodedAudioStream::begin()
> [D] CodecMP3Helix.h : 76 - virtual void audio_tools::MP3DecoderHelix::begin()
> [D] AudioPlayer.h : 207 - virtual void audio_tools::AudioSourceSdFat::begin()
> [I] AudioPlayer.h : 281 - directory: /
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: Playlist.csv with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: 'Playlist.csv': 0
> [D] AudioPlayer.h : 360 - -> close: 'Playlist.csv'
> [D] AudioPlayer.h : 340 - -> processing: 22 with index 0
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: 001.mp3 with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: '001.mp3': 1
> [I] AudioPlayer.h : 348 - ==> found: '001.mp3' at index 0
> [D] AudioPlayer.h : 360 - -> close: '22'
> [D] AudioPlayer.h : 340 - -> processing: 01 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '01'
> [D] AudioPlayer.h : 340 - -> processing: 02 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '02'
> [D] AudioPlayer.h : 340 - -> processing: 03 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '03'
> [D] AudioPlayer.h : 340 - -> processing: 04 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '04'
> [D] AudioPlayer.h : 340 - -> processing: 05 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '05'
> [D] AudioPlayer.h : 340 - -> processing: 06 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '06'
> [D] AudioPlayer.h : 340 - -> processing: 07 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '07'
> [D] AudioPlayer.h : 340 - -> processing: 08 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '08'
> [D] AudioPlayer.h : 340 - -> processing: 09 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '09'
> [D] AudioPlayer.h : 340 - -> processing: 10 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '10'
> [D] AudioPlayer.h : 340 - -> processing: 21 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '21'
> [D] AudioPlayer.h : 340 - -> processing: 23 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '23'
> [D] AudioPlayer.h : 340 - -> processing: 24 with index 1
[... 380 lines cut ...]
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
> [D] I2SStream.h : 128 - virtual size_t audio_tools::I2SStream::writeExt(const uint8_t*, size_t)
[... cut ...]
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
> [D] I2SStream.h : 128 - virtual size_t audio_tools::I2SStream::writeExt(const uint8_t*, size_t)
> [D] I2SESP32.h : 110 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes

pschatzmann, 2021-12-03:
I don't get it - so decodeMetaData() is called.
Will you try with the same file that you have sent to me ?

podaen, 2021-12-03:
This is with that file and the example shetch with volume control

> Opening port
> Port open
> [D] I2SStream.h : 87 - void audio_tools::I2SStream::begin(audio_tools::I2SConfig)
> [D] I2SESP32.h : 38 - void audio_tools::I2SBase::begin(audio_tools::I2SConfig)
> [I] I2SConfig.h : 63 - rx/tx mode: TX
> [I] I2SConfig.h : 64 - port_no: 0
> [I] I2SConfig.h : 65 - is_master: Master
> [I] I2SConfig.h : 66 - sample rate: 44100
> [I] I2SConfig.h : 67 - bits per sample: 16
> [I] I2SConfig.h : 68 - number of channels: 2
> [I] I2SConfig.h : 69 - i2s_format: I2S_STD_FORMAT
> [I] I2SConfig.h : 77 - pin_ws: 25
> [I] I2SConfig.h : 78 - pin_bck: 26
> [I] I2SConfig.h : 79 - pin_data: 22
> [D] I2SESP32.h : 93 - begin - started
> [I] AudioPlayer.h : 779 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int))
> [D] AudioPlayer.h : 599 - virtual bool audio_tools::AudioPlayer::begin(int, bool)
> [D] AudioOutput.h : 310 - void audio_tools::EncodedAudioStream::begin()
> [D] CodecMP3Helix.h : 76 - virtual void audio_tools::MP3DecoderHelix::begin()
> [D] AudioPlayer.h : 207 - virtual void audio_tools::AudioSourceSdFat::begin()
> [I] AudioPlayer.h : 281 - directory: /
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: Playlist.csv with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: 'Playlist.csv': 0
> [D] AudioPlayer.h : 360 - -> close: 'Playlist.csv'
> [D] AudioPlayer.h : 340 - -> processing: 22 with index 0
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: 001.mp3 with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: '001.mp3': 1
> [I] AudioPlayer.h : 348 - ==> found: '001.mp3' at index 0
> [D] AudioPlayer.h : 360 - -> close: '22'
> [D] AudioPlayer.h : 340 - -> processing: 01 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '01'
> [D] AudioPlayer.h : 340 - -> processing: 02 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '02'
> [D] AudioPlayer.h : 340 - -> processing: 03 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '03'
> [D] AudioPlayer.h : 340 - -> processing: 04 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '04'
> [D] AudioPlayer.h : 340 - -> processing: 05 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '05'
> [D] AudioPlayer.h : 340 - -> processing: 06 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '06'
> [D] AudioPlayer.h : 340 - -> processing: 07 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '07'
> [D] AudioPlayer.h : 340 - -> processing: 08 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '08'
> [D] AudioPlayer.h : 340 - -> processing: 09 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '09'
> [D] AudioPlayer.h : 340 - -> processing: 10 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '10'
> [D] AudioPlayer.h : 340 - -> processing: 21 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '21'
> [D] AudioPlayer.h : 340 - -> processing: 23 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '23'
> [D] AudioPlayer.h : 340 - -> processing: 24 with index 1
[... 84 lines cut ...]
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
[... cut ...]
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
> [D] I2SStream.h : 128 - virtual size_t audio_tools::I2SStream::writeExt(const uint8_t*, size_t)
> [D] I2SESP32.h : 110 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes

podaen, 2021-12-03:
here is the shetch

```
// set this in AudioConfig.h or here after installing https://github.com/pschatzmann/arduino-libhelix.git
#define USE_HELIX 
#define USE_SDFAT

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

using namespace audio_tools;

const char* startFilePath = "/";
const char* ext = "mp3";

AudioSourceSdFat source(startFilePath, ext);
I2SStream i2s;
VolumeOutput volume(i2s);
LogarithmicVolumeControl lvc(0.1);
MP3DecoderHelix decoder;
//AudioPlayer player(source, i2s, decoder);
AudioPlayer player(source, volume, decoder);

void printMetaData(MetaDataType type, const char* str, int len) {
	Serial.print("==> ");
	Serial.print(toStr(type));
	Serial.print(": ");
	Serial.println(str);
}

void setup() {
	Serial.begin(115200);
	//AudioLogger::instance().begin(Serial, AudioLogger::Debug);

	// setup output
	auto cfg = i2s.defaultConfig(TX_MODE);
	i2s.begin(cfg);

	// setup player
	//source.setFileFilter("*Bob Dylan*");
	player.setMetadataCallback(printMetaData);
	player.begin();
	volume.setVolume(0.1);
}

void loop() {
	player.copy();
	//Serial.print("xPortGetFreeHeapSize: ");
	//Serial.println(xPortGetFreeHeapSize());//290116//Since ESP32 has 328 KiB of data RAM
	//Serial.println("");
}
```

podaen, 2021-12-03:
This is with the file I send you and with the example shetch.

> Opening port
> Port open
> [D] I2SStream.h : 87 - void audio_tools::I2SStream::begin(audio_tools::I2SConfig)
> [D] I2SESP32.h : 38 - void audio_tools::I2SBase::begin(audio_tools::I2SConfig)
> [I] I2SConfig.h : 63 - rx/tx mode: TX
> [I] I2SConfig.h : 64 - port_no: 0
> [I] I2SConfig.h : 65 - is_master: Master
> [I] I2SConfig.h : 66 - sample rate: 44100
> [I] I2SConfig.h : 67 - bits per sample: 16
> [I] I2SConfig.h : 68 - number of channels: 2
> [I] I2SConfig.h : 69 - i2s_format: I2S_STD_FORMAT
> [I] I2SConfig.h : 77 - pin_ws: 25
> [I] I2SConfig.h : 78 - pin_bck: 26
> [I] I2SConfig.h : 79 - pin_data: 22
> [D] I2SESP32.h : 93 - begin - started
> [I] AudioPlayer.h : 779 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int))
> [D] AudioPlayer.h : 599 - virtual bool audio_tools::AudioPlayer::begin(int, bool)
> [D] AudioOutput.h : 310 - void audio_tools::EncodedAudioStream::begin()
> [D] CodecMP3Helix.h : 76 - virtual void audio_tools::MP3DecoderHelix::begin()
> [D] AudioPlayer.h : 207 - virtual void audio_tools::AudioSourceSdFat::begin()
> [I] AudioPlayer.h : 281 - directory: /
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: Playlist.csv with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: 'Playlist.csv': 0
> [D] AudioPlayer.h : 360 - -> close: 'Playlist.csv'
> [D] AudioPlayer.h : 340 - -> processing: 22 with index 0
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: 001.mp3 with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: '001.mp3': 1
> [I] AudioPlayer.h : 348 - ==> found: '001.mp3' at index 0
> [D] AudioPlayer.h : 360 - -> close: '22'
> [D] AudioPlayer.h : 340 - -> processing: 01 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '01'
> [D] AudioPlayer.h : 340 - -> processing: 02 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '02'
> [D] AudioPlayer.h : 340 - -> processing: 03 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '03'
> [D] AudioPlayer.h : 340 - -> processing: 04 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '04'
> [D] AudioPlayer.h : 340 - -> processing: 05 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '05'
> [D] AudioPlayer.h : 340 - -> processing: 06 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '06'
> [D] AudioPlayer.h : 340 - -> processing: 07 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '07'
> [D] AudioPlayer.h : 340 - -> processing: 08 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '08'
> [D] AudioPlayer.h : 340 - -> processing: 09 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '09'
> [D] AudioPlayer.h : 340 - -> processing: 10 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '10'
> [D] AudioPlayer.h : 340 - -> processing: 21 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '21'
> [D] AudioPlayer.h : 340 - -> processing: 23 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '23'
> [D] AudioPlayer.h : 340 - -> processing: 24 with index 1
[... 228 lines cut ...]
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
[... cut ...]
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
> [D] I2SStream.h : 128 - virtual size_t audio_tools::I2SStream::writeExt(const uint8_t*, size_t)
> [D] I2SESP32.h : 110 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes

podaen, 2021-12-03:
Here is my shetch with addapted volume control

```
//* @file player - sd - i2s.ino
//* @brief see https ://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-sd-i2s/README.md
//*
//*@author Phil Schatzmann
//* @copyright GPLv3
//* /

// set this in AudioConfig.h or here after installing https://github.com/pschatzmann/arduino-libhelix.git
#define USE_HELIX 
#define USE_SDFAT

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

using namespace audio_tools;

const char* startFilePath = "/";
const char* ext = "mp3";

AudioSourceSdFat source(startFilePath, ext);
I2SStream i2s;
VolumeOutput volume(i2s);
LogarithmicVolumeControl lvc(0.1);
MP3DecoderHelix decoder;
//AudioPlayer player(source, i2s, decoder);
AudioPlayer player(source, volume, decoder);

void printMetaData(MetaDataType type, const char* str, int len) {
	Serial.print("==> ");
	Serial.print(toStr(type));
	Serial.print(": ");
	Serial.println(str);
}

void setup() {
	Serial.begin(115200);
	AudioLogger::instance().begin(Serial, AudioLogger::Debug);

	// setup output
	auto cfg = i2s.defaultConfig(TX_MODE);
	i2s.begin(cfg);

	// setup player
	//source.setFileFilter("*Bob Dylan*");
	player.setMetadataCallback(printMetaData);
	player.begin();
	volume.setVolume(0.1);
}

void loop() {
	player.copy();
	//Serial.print("xPortGetFreeHeapSize: ");
	//Serial.println(xPortGetFreeHeapSize());//290116//Since ESP32 has 328 KiB of data RAM
	//Serial.println("");
}
```

pschatzmann, 2021-12-03:
Something is really different between my and your code:
```
        virtual bool begin(int index=0, bool isActive = true) {
            LOGD(LOG_METHOD);
            bool result = false;

            // navigation supoort
            autonext = p_source->isAutoNext();

            // start dependent objects
            p_out_decoding->begin();
            p_source->begin();
            meta_out.begin();

    ```

In your log you get:

    ```

[D] I2SESP32.h : 93 - begin - started
[I] AudioPlayer.h : 779 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void ()(audio_tools::MetaDataType, const char, int))
[D] AudioPlayer.h : 599 - virtual bool audio_tools::AudioPlayer::begin(int, bool)
[D] AudioOutput.h : 310 - void audio_tools::EncodedAudioStream::begin()
[D] CodecMP3Helix.h : 76 - virtual void audio_tools::MP3DecoderHelix::begin()
[D] AudioPlayer.h : 207 - virtual void audio_tools::AudioSourceSdFat::begin()
[I] AudioPlayer.h : 281 - directory: /
[D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
    ```

So the meta_out.begin(); seems to be missing

podaen, 2021-12-03:
This is the log from the file I send you.

> Opening port
> Port open
> �end setup
> here
> [D] I2SStream.h : 87 - void audio_tools::I2SStream::begin(audio_tools::I2SConfig)
> [D] I2SESP32.h : 38 - void audio_tools::I2SBase::begin(audio_tools::I2SConfig)
> [I] I2SConfig.h : 63 - rx/tx mode: TX
> [I] I2SConfig.h : 64 - port_no: 0
> [I] I2SConfig.h : 65 - is_master: Master
> [I] I2SConfig.h : 66 - sample rate: 44100
> [I] I2SConfig.h : 67 - bits per sample: 16
> [I] I2SConfig.h : 68 - number of channels: 2
> [I] I2SConfig.h : 69 - i2s_format: I2S_STD_FORMAT
> [I] I2SConfig.h : 77 - pin_ws: 25
> [I] I2SConfig.h : 78 - pin_bck: 26
> [I] I2SConfig.h : 79 - pin_data: 22
> [D] I2SESP32.h : 93 - begin - started
> [I] AudioPlayer.h : 779 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int))
> [D] AudioPlayer.h : 599 - virtual bool audio_tools::AudioPlayer::begin(int, bool)
> [D] AudioOutput.h : 310 - void audio_tools::EncodedAudioStream::begin()
> [D] CodecMP3Helix.h : 76 - virtual void audio_tools::MP3DecoderHelix::begin()
> [D] AudioPlayer.h : 207 - virtual void audio_tools::AudioSourceSdFat::begin()
> [I] AudioPlayer.h : 281 - directory: /
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: Playlist.csv with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: 'Playlist.csv': 0
> [D] AudioPlayer.h : 360 - -> close: 'Playlist.csv'
> [D] AudioPlayer.h : 340 - -> processing: 22 with index 0
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 0
> [D] AudioPlayer.h : 340 - -> processing: 001.mp3 with index 0
> [I] AudioPlayer.h : 272 - -> isValidAudioFile: '001.mp3': 1
> [I] AudioPlayer.h : 348 - ==> found: '001.mp3' at index 0
> [D] AudioPlayer.h : 360 - -> close: '22'
> [D] AudioPlayer.h : 340 - -> processing: 01 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '01'
> [D] AudioPlayer.h : 340 - -> processing: 02 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '02'
> [D] AudioPlayer.h : 340 - -> processing: 03 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '03'
> [D] AudioPlayer.h : 340 - -> processing: 04 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '04'
> [D] AudioPlayer.h : 340 - -> processing: 05 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '05'
> [D] AudioPlayer.h : 340 - -> processing: 06 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '06'
> [D] AudioPlayer.h : 340 - -> processing: 07 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '07'
> [D] AudioPlayer.h : 340 - -> processing: 08 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '08'
> [D] AudioPlayer.h : 340 - -> processing: 09 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '09'
> [D] AudioPlayer.h : 340 - -> processing: 10 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '10'
> [D] AudioPlayer.h : 340 - -> processing: 21 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
> [D] AudioPlayer.h : 360 - -> close: '21'
> [D] AudioPlayer.h : 340 - -> processing: 23 with index 1
> [D] AudioPlayer.h : 329 - void audio_tools::AudioSourceSdFat::getFileAtIndex(audio_tools::AudioFile, size_t, size_t&, audio_tools::AudioFile&): 1
[... 310 lines cut ...]
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes
> [D] AudioPlayer.h : 816 - static void audio_tools::AudioPlayer::decodeMetaData(void*, void*, size_t), 11268
> [I] AudioCopy.h : 96 - StreamCopy::copy 11268 -> 11268 -> 11268 bytes - in 1 hops
> [D] AudioPlayer.h : 744 - virtual void audio_tools::AudioPlayer::copy()
> [D] AudioCopy.h : 71 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
[... cut ...]
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioOutput.h : 412 - virtual size_t audio_tools::VolumeOutput::write(const uint8_t*, size_t)
> [D] AudioStreams.h : 277 - virtual size_t audio_tools::BufferedStream::write(const uint8_t*, size_t): 4608
> [D] I2SStream.h : 128 - virtual size_t audio_tools::I2SStream::writeExt(const uint8_t*, size_t)
> [D] I2SESP32.h : 110 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
> [D] I2SESP32.h : 117 - i2s_write 4608 -> 4608 bytes

podaen, 2021-12-03:
in the MetaDataId3.h I check...

```
    /// provide the (partial) data which might contain the meta data
    size_t write(const uint8_t* data, size_t len){
        if (armed){ 
            Serial.print("Status: ");
            Serial.println(status);
            switch(status){
                case TagNotFound:
                    processTagNotFound(data,len);
                    break;
                case PartialTagAtTail:
                    processPartialTagAtTail(data,len);
                    break;
                case TagFoundPartial:
                    Serial.print("len: ");
                    Serial.println(len);
                    processTagFoundPartial(data,len);
                    break;
                default:                
                    // do nothing
                    break;
            }
        }
        return len;
    }
```

Here below it doesn't contain anything

```
    /// We have the beginning of the metadata and need to process the remainder
    void processTagFoundPartial(const uint8_t* data, size_t len) {
        //String str = (char*)data;//gives me special charachters
        //Serial.print(str);
        if (tag!=nullptr){
            Serial.print("tag: ");
            int remainder = sizeof(ID3v1) - use_bytes_of_next_write;
            memcpy(tag,data+use_bytes_of_next_write,remainder);
            processNotify();                 
            use_bytes_of_next_write = 0;
           
        } else if (tag_ext!=nullptr){
            Serial.print("tag_ext");
            int remainder = sizeof(ID3v1Enhanced) - use_bytes_of_next_write;
            memcpy(tag_ext,data+use_bytes_of_next_write,remainder);
            processNotify();                 
            use_bytes_of_next_write = 0; 
            
        }
    }
```

The serial monitor

> Opening port
> Port open
> end setup
> here
> [W] AudioPlayer.h : 224 - -> selectStream: 0 '001.mp3'
> tracknumber: 0
> Playlist opened
> read read song form playlist
> pfileName: 002.mp3
> set path
> [W] AudioPlayer.h : 231 - -> selectStream: /002.mp3
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 0
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Status: 2
> len: 11268
> Next Stream

It takes a long time before it see a status change.

podaen, 2021-12-03:
> So the meta_out.begin(); seems to be missing
> 
> MetaDataID3 has
> 
> ```
>     void begin() {
>         LOGD(LOG_METHOD);
>         id3v1.begin();
>         id3v2.begin();
>     }
> ```
> 
> So there should be a line in the log!

I checked the MetadataID3.h file and indeed I have an old version. Very strang because I updated this morning. I take a look what is going on.

pschatzmann, 2021-12-03:
This should not be. With your big write size, it should find it after the first write in the processing for processTagNotFound() in your first Status: 0

pschatzmann, 2021-12-03:
Maybe we should add a  tag_processed = false; to the begin method!

podaen, 2021-12-03:
I will do that, but first I need to get the latest version of the MetaDataID3.h file. There is something within my git extension why it is not doing it. My githtml is uptodate... I have updated the extension not so long a go and I think I do something wrong. I says it's up to date but it issen't.

podaen, 2021-12-03:
I see now it was uptodate... I think you are using

https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioTools/MetaDataID3.h

and I am using

https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioMetaData/MetaDataID3.h

pschatzmann, 2021-12-03:
Oh I did not notice. I thought I moved everything to the new directory!

podaen, 2021-12-03:
glad we find something.

pschatzmann, 2021-12-03:
They are almost identical so this should  not be the source of the issue however

podaen, 2021-12-03:
Are the both needed? otherwise clean it up first. Then I will look what could be the issue.

pschatzmann, 2021-12-04:
I just committed the correction...

podaen, 2021-12-04:
It for sure entering processTagNotFound and I set this tag_processed = false; but no changes.

> Port closed
> Uploading to I/O board
> Opening port
> Port open
> �end setup
> here
> [W] AudioPlayer.h : 224 - -> selectStream: 0 '001.mp3'
> tracknumber: 0
> Playlist opened
> read read song form playlist
> pfileName: 002.mp3
> set path
> [W] AudioPlayer.h : 231 - -> selectStream: /002.mp3
> status: 0 processTagNotFound ID3V1: status: 0 
> processTagNotFound ID3V2: status: 0 processTagNotFound ID3V1: status: 0 
...
> processTagNotFound ID3V2: status: 0 processTagNotFound ID3V1: status: 0 
> processTagNotFound ID3V2: status: 2 status: 0 
...
> processTagNotFound ID3V2: status: 2 status: 0 
> processTagNotFound ID3V2: Next Stream

I analysed the file and it seems the tag is ID3V2.3

[image]

podaen, 2021-12-04:
Here... more tags

[image]

podaen, 2021-12-04:
I did some lookup and it means when there is ID3v2.3. It contains v2 and v3 and that one of those could be displayed. Mostly the highest version.

[50 more replies]

closed 2021-12-08 by podaen