pschatzmann/arduino-audio-tools #63: 2 ESP32 1. Streams-url_mp3-i2s --> 2. streams-i2s-a2dp

GuidoMeinke, 2021-12-01:
Hello Phil,

your sketches working like a charm, but unfortunately i get no signal on my BT-Speakers. should these examples working out of the box?

Because when i touch the pins of the a2dp source my BT speaker is scratching, but there is no dataflow between these two ESP32.

is it a wiring issue or is there any way to get it working?

thanks in advance

pschatzmann, 2021-12-01:
I have never tried this out myself and I have my doubts if this will work reliably.  But I have a couple of points to double check:
- only one device must be defined as master: I suggest to use the sender as master (cfg.is_master=true) and the receiver as client (cfg.is_master = false);
- double check the data format. It must match with the requirements from A2DP
  - sample_rate: 44100
  - bits_per_sample: 16
  - channels: 2
- you might want to activate the pll (cfg. use_apll = true)

There is a discussion https://github.com/pschatzmann/arduino-audio-tools/discussions/26

The final advice is start simple and test each step (e.g. with the output to a CSVStream. Otherwise you have no chance to indentify what the issue is.

pschatzmann, 2023-04-26:
https://github.com/pschatzmann/arduino-audio-tools/discussions/149
I2S between 2 ESP32 seems to have issues.
Closed due to inactivity

closed 2023-04-26 by pschatzmann