pschatzmann/arduino-audio-tools #64: I2S Audio transmission between two ESP32s stuttering?

JackBlair87, 2021-12-12:
Hello everyone, 

I am working on a project that receives audio inputs from multiple phones over bluetooth, performs some computation with that audio, and then transmits the output to multiple bluetooth speakers. Right now I am using the ESP32 module, and seeing as it can only connect to one A2DP audio device over bluetooth at a time, I am attempting to wire an I2S channel between two of them, so one can receive and one can send. Currently (just trying to stream one line of audio) my setup looks like this.

Phone -->A2DP Bluetooth --> ESP32 1 --> I2S --> ESP32 2--> A2DP Bluetooth --> Bluetooth Speaker

I have been able stream audio packets to the speaker, but the output is jitters and lags to the point where it is useless. I wanted to see if anyone had any advice as to better approach this problem, or recreate my current issue. I'll attach my code and wiring below, but I believe the issue has something to do with the I2S transmission, since I tested both sending and receiving a sin wave and the bluetooth side of things is perfect. 

Bluetooth receiving end:

```
#define USE_A2DP

#include "AudioTools.h"

using namespace audio_tools;  

A2DPStream a2dpStream = A2DPStream::instance(); // access A2DP as stream
I2SStream i2sStream;                            // Access I2S as stream
StreamCopy copier(i2sStream, a2dpStream); // copy from A2DP to I2S
ConverterFillLeftAndRight<int16_t> filler(RightIsEmpty); // fill both channels

void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Warning);

    i2sStream.begin(i2sStream.defaultConfig(TX_MODE));
    
    auto cfgA2DP = a2dpStream.defaultConfig(RX_MODE);
    a2dpStream.setVolume(3);
    a2dpStream.begin(RX_MODE, "RowUrBoat");
}

void loop() {
    copier.copy();
}
```

Bluetooth Sending end:
```
#define USE_A2DP

#include "AudioTools.h"

using namespace audio_tools;  

I2SStream i2sStream;                            // Access I2S as stream
A2DPStream a2dpStream = A2DPStream::instance(); // access A2DP as stream
StreamCopy copier(a2dpStream, i2sStream); // copy i2sStream to a2dpStream
ConverterFillLeftAndRight<int16_t> filler(RightIsEmpty); // fill both channels

void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Info);
    
    auto cfgA2DP = a2dpStream.defaultConfig(TX_MODE);
    a2dpStream.setVolume(3);
    a2dpStream.begin(TX_MODE, "OontZ Angle 3 DS 7BA");
    
    a2dpStream.setNotifyAudioChange(i2sStream); // i2s is using the info from a2dp
    i2sStream.begin(i2sStream.defaultConfig(RX_MODE));

}

void loop() {
    copier.copy(filler);
}
```

Wiring:
ESP32 1         ESP32 2
D14                 D14
D15                 D15
D32                 D22
D22                 D32

podaen, 2021-12-16:
Why are you wireless connecting to you spreaker?

podaen, 2021-12-26:
Since you diden't react, I think you have found an other way.

JackBlair87, 2021-12-26:
Sorry, I didn't see your message. I need to connect to my speakers wirelessly as a constraint of my system. Both pairs of sending and receiving devices will be moving around independently of each other and can't have wires between them.

podaen, 2021-12-27:
I ask you on an other way, Is it physical possible to connect other media sources to your spreaker, without thinking that they move arround. I am currious what the reason is why you won't connect with wires to your spreaker. Moving arround is not an issue if you attach the esp to your spreaker.

I would advice you to do it like this

Phone -->A2DP Bluetooth --> ESP32 --> I2S -->DECODER --> Bluetooth Speaker

Later on you can attach addition esp to you setup to connect with mulie phones at the same time.

JackBlair87, 2021-12-27:
Connecting with wires to my speakers isn't possible for me because they are commercial (off the shelf) bluetooth speakers. I imagine you are thinking of classic speakers with a positive an negative lead, but instead they are like this https://www.amazon.com/JBL-Portable-Wireless-Bluetooth-Speaker/dp/B014QQAOUE and can only have a bluetooth connection. 

For now, I restate the question- How can I transmit audio over i2s between two ESP32s? The code in the original post created lag and stutters, so can you replicate the issue or offer different code that could work?

podaen, 2021-12-27:
Bad luck... I don't think it will properly work with two a2dp connections on the same line. I have recoverd an old spreaker from my hifi installation. As crazy it will sound, this will work.

> How can I transmit audio over i2s between two ESP32s?

Sorry. No experence on that.

podaen, 2021-12-27:
Watch this #63 topic for the i2s connection.

alexus2033, 2023-01-12:
Hello,
in my [test with A2DP Transmitter](/pschatzmann/arduino-audio-tools/pull/569) I had a lot of trouble with ESP Core configuration.
It might help, if you check and set this Menu-Settings in the Arduino IDE:
```
"Tools" -> "Events run on" -> "Core 0"
"Tools" -> "Arduino runs on" -> "Core 1"
```

podaen, 2023-01-26:
I can confirm, for some internet tools you need to set that right or some data will not be transmitted to an AP. Don't know If it could be checked by code, but it will help. That's true!

pschatzmann, 2023-04-26:
https://github.com/pschatzmann/arduino-audio-tools/discussions/149
I2S between 2 ESP32 seems to have issues.
Closed due to inactivity

closed 2023-04-26 by pschatzmann