pschatzmann/arduino-audio-tools #66: SD board

podaen, 2021-12-27:
The sd board of the example is made to use on an Arduino. This board need a power supply of 5v and since the exp32 is 3.3v... the miso line needs a resistor of 1k to make the reset work with multieple spi devices. PS: The 5v pin of the esp32 is not regulated by the esp32. There for I would not recommend it.

https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sd-i2s#sd-car

Below is a 3.3v SD reader that's direct compatable with the esp32

[image]

pschatzmann, 2021-12-28:
I was double checking the spec of my module: 

https://www.aliexpress.com/item/32949126408.html?spm=a2g0o.productlist.0.0.3638155bpIASEi&algo_pvid=fa678ade-6d17-4bcd-821e-c499efb8e115&algo_exp_id=fa678ade-6d17-4bcd-821e-c499efb8e115-0&pdp_ext_f=%7B%22sku_id%22%3A%2266198527324%22%7D&pdp_pi=-1%3B0.28%3B-1%3BCHF+1.72%400%3BCHF%3Bsearch-mainSearch

As far as I can see it has a built in logic level converter which makes it compatible with 3.3V logic. The power supply does not matter.

podaen, 2021-12-28:
It has a level converter but if you look closer you will see that the 5v line is not only going to the level converter. I m pretty sure I m right on this. For example a display with a level converter has the that too, still you have to add a voltage devider to make it work.

podaen, 2022-01-03:
[image]

pschatzmann, 2022-01-03:
That's fine - the only critical lines are the signal lines. The CS, SCK, MOSI  signal lines that are going out of the esp32 to the module  are fine because the ESP32 provides 3V logic. 
The only line which matters is the signal going from the module to the ESP32 which is the MISO line: I was double checking it with my oscilloscope and I can confirm that Vmax is around 3.3 V

closed 2022-01-03 by pschatzmann

podaen, 2022-01-03:
I don't have the tools with me to measure this out. Than it would be the current flow of the other spi devices on the same line that causes this issue. But still out of my experence I can tell you I diden't work on 3.3v, it need to be powered on 5v otherwise it won't work. I still waiting on my delivery for the more Appropriate device to work on an esp32.

pschatzmann, 2022-01-03:
Yes I confirm that it needs to be powered by 5V.  
3.3V is not sufficient to power this module.

podaen, 2022-01-05:
I did some research and found this...

http://elm-chan.org/docs/mmc/mmc_e.html

The sd card reader I first posted is even worse. It doesn't have a buffer chip with tri-state capabilities.

The one you posted has a buffer chip with tri-state but the pullup on the miso line need to be done by the user to avoid ini problems.

This module is the best I could find for the esp32. because it can be powered on 3.3v, has a buffer chip with tri-state capabilities. Note: The bufferchip is the same from your example so it probably needs also a pullup resistor on the miso line.

https://nl.aliexpress.com/item/1005001380769059.html?spm=a2g0o.9042311.0.0.2b6f4c4dBzd68e

podaen, 2022-01-05:
And than I found this...

https://nl.aliexpress.com/item/32787679017.html?spm=a2g0o.productlist.0.0.7db967b5MtrKQM&algo_pvid=823a08bb-f32f-4161-a1a7-9e6aa197931a&algo_exp_id=823a08bb-f32f-4161-a1a7-9e6aa197931a-22&pdp_ext_f=%7B"sku_id"%3A"-1"%7D

accourding to the documentation it can be powered between 3.2V - 5.5V

podaen, 2022-01-05:
I got this module with the buildin resistors on the miso finaly working. You don't need to add external resistors. The build in pullup resistors just work fine.

> [image] [image]

Add those lines at the top of the setup().

	pinMode(5, OUTPUT);  //SS pin not set as output in the spi library(sd card reader)
	pinMode(19, INPUT_PULLUP);  //miso needs a pullup (sd card reader)

These are the default gpios for a VSPI on an esp32 and could be different.

podaen, 2022-01-16:
I did some tests to work with multiple spi devices (sd card an RFID card reader) and this board bellow works the best with it. But there is one think you have to keep in mind.  When you are using the sd player and have openend a file, the spi is still in use for that devices. As long it is in use for one of your devices. It can't be used for an other device. You need to close the (chip) file before you can use an other divice. It means when you are playing a song from the sd card you need wait until it's done or just stop playing the song.
[image]

@pschatzmann We need to close the file when the song is done. file.close(); in to the class AudioPlayer. Not sure how we can get the file in to the  AudioPlayer class. Now the file is closed when we do selectStream(). We need to do this earlier...

pschatzmann, 2022-01-19:
This does't sound right to me. You should be able to connect and work with multiple SPI devices in parallel. Please note that  each device must be managed via it's own CS pin on the Microcontroller.

podaen, 2022-01-19:
I have leads too... The rfid works before start to play audio, after the player has played the rfid doesn't work anymore. Also for multiple readers I need to close down the reader before by lower the reset pin, I did that.  The sdfat closes itself when all files are closed and the file of the stream is not closed when it's done playing. Player.stop() doesn't work on that. You can do a test by close down the spi by doing SPI.end() (SD.end(), file.close()); I did some tests to be sure. It won't work once the player has played a song because of the reason I have mentioned.

podaen, 2022-01-19:
Could you please look in to that how we could close down the open files, soo the spi communication could be closed down of the sd card reader?

pschatzmann, 2022-01-19:
Did you try to use your own SdSpiConfig for AudioSourceSdFat. You might use the constructor which is using the DEDICATED_SPI setting which limits you to one SPI device....

podaen, 2022-01-19:
Not yet. I will take a look. Thanks

podaen, 2022-01-19:
I am confused... I have found it at sdfatconfig.h and in sdspidriver.h. Do I need to change them both?

pschatzmann, 2022-01-19:
just use the following constructor:
        AudioSourceSdFat(const char* startFilePath, const char* ext, SdSpiConfig &config) {

podaen, 2022-01-19:
Having the same issue. Can not begin Transaction of the other spi device. This is what they say on the side of the rfid.

> In SPI, the communication "happens" only when the CS (chip select) signal is pulled low (by the master), acting as an "addressing" signal. In this way, you can have many chips sharing the MISO/MOSI/CLK signals.
> In idle state (no communication at all), all the CS signals should be pulled high. When the master wants to speak with a slave, it (the master) selects the slave by pulling the corresponding CS signal low, sends/receives the data and pulls the CS high.
> Exactly this is happening in the MFRC522 library - when the communication ends, the CS returns to high. And exactly in the same way should happen in the case of SD communication also - the SD's CS (SS) signal should return to high after communication finishes.

pschatzmann, 2022-01-19:
Was this  after using SdSpiConfig(chipSelect, SHARED_SPI, SD_SCK_MHZ(speedMHz)) ?

podaen, 2022-01-19:
I just changed it in to the  sdspidriver.h. 

Now I tried

```
#define SPI_SPEED SD_SCK_MHZ(4)
SdSpiConfig spiconfig (SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(4));
AudioSourceSdFat SDsource(startFilePath, ext, spiconfig);
```
And I have some improvment, the code keeps running. But the rfid still doesn't scan. I am looking in to the rfid what could be the problem.

podaen, 2022-01-19:
Soo the chip selection pin is turned high now?

pschatzmann, 2022-01-19:
It is my understanding that when using SHARED_SPI, the CS is pulled low (only) whenever a SD command is executed and is pulled high right after command has executed. 

To make sure that it is high you just need to make sure that no SD commands execute: e.g. by calling stop().

podaen, 2022-01-19:
If the command is exectuded I would be able to end the spi I think. When I was programming my playlist I could not close the spi before evry file was closed. I will try tomorow if can now end the spi after playing a song and stop().

podaen, 2022-01-20:
I am sorry. I was wrong. I did some test on what I was saying, and it appears that the spi can be ended even when a file is open. it It was a long time a go I programmed the playlist. But there was something why I coulden't end it. I did now the test over on end the spi in my program with the sdspiconfig and it can be end now.

A good spi transaction looks like this for multiple spi devices 

    SPI.beginTransaction(SPISettings(x, x, x));
    digitalWrite(chipSelectPin, LOW);
    SPI.transfer(mybyte1);
    SPI.transfer(mybyte2);
    digitalWrite(chipSelectPin, HIGH);
    SPI.endTransaction();

This is what I found in the sdfat library

```
bool SdSpiCard::readSingle(uint32_t sector, uint8_t* dst) {
  // use address if not SDHC card
  if (type() != SD_CARD_TYPE_SDHC) {
    sector <<= 9;
  }
  if (cardCommand(CMD17, sector)) {
    error(SD_CARD_ERROR_CMD17);
    goto fail;
  }
  if (!readData(dst, 512)) {
    goto fail;
  }
  spiStop();
  return true;

 fail:
  spiStop();
  return false;
}
//------------------------------------------------------------------------------
bool SdSpiCard::readStart(uint32_t sector) {
  if (type() != SD_CARD_TYPE_SDHC) {
    sector <<= 9;
  }
  if (cardCommand(CMD18, sector)) {
    error(SD_CARD_ERROR_CMD18);
    goto fail;
  }
  return true;

 fail:
  spiStop();
  return false;
}
```

If the readStart is used the spi doesn't stop without a fail. Then the read stop need to be called. 

```
bool SdSpiCard::readStop() {
  if (cardCommand(CMD12, 0)) {
    error(SD_CARD_ERROR_CMD12);
    goto fail;
  }
  spiStop();
  return true;

 fail:
  spiStop();
  return false;
}
```
I am not sure when the readstart is called, but I was once told on the side of the sdfat: if you want to end what you are doing, you should close all open files then call sd.end(). Currently the only function of sd.end() is to call end() in the SPI library.

podaen, 2022-01-20:
I checked on the rfid and it looks like this.

> SPI.beginTransaction(SPISettings(x, x, x));
> digitalWrite(chipSelectPin, LOW);
> SPI.transfer(mybyte1);
> SPI.transfer(mybyte2);
> digitalWrite(chipSelectPin, HIGH);
> SPI.endTransaction();

podaen, 2022-01-20:
Out of what I experienced... it is not the player.begin that is the issue for mutiple spi devices.

pschatzmann, 2022-01-20:
Have you tested both cases after the change to SHARED_SPI ? I would expect your described behaviour when using DEDICATED_SPI.

If you absolutely need to release all file related resources currently the best way is to work with pointers and allocated AudioSourceSdFat with new and delete it after you do not use it any more.

podaen, 2022-01-20:
Ah... My mistake. By testing I have added rfid reset. This was the reason the rfid reader still diden't worked. Now it does.

I have tested it now on two ways, 

1/ use the audioplayer seperatly from the rfid reader. when the audio stops start scanning and this works without a problem.

2/use the rfid reader while I am playing a song...
Evrything works but the sound starts chuncking and it slows down. The rfid is only using 25ms to scan, not sure what causing this.

podaen, 2022-01-20:
Maybe it is because I don't work with a buffer chip right now. I have 4 types of sd readers... I want to test the second of below. It can be powered on 3.3v and have a buffer chip.

[image]

podaen, 2022-01-20:
Soo I did test with the second of below with the bufferchip and it plays audio, but the rfid reader doesn't work anymore in both cases (without playing audio and with playing audio). To be sure I changed back to the mini and than it works.

podaen, 2022-01-20:
I have now tested the second from above and this does the same than the one underneath it. it plays also on 3.3v but with multiple device it stops the other devices.

podaen, 2022-01-21:
Maybe this is the case... The rfid reader doesn't have a bufferchip with tri-state capabilities like the sd card reader. This would be a proper way to connect more than two rfid devices (what I don't have).

[image]