pschatzmann/arduino-audio-tools #67: Turn off player for A2DP

podaen, 2021-12-27:
I want to switch between the player and the a2dp by turning of the player like this. I think I closed evrything, still I get stuck by starting the a2dp.

```
void A2DPOnOFF(bool BlnOn) {
	if (BlnOn == true) {
		player.stop();
		player.end();
		i2s.end();
		Serial.println("activate A2DP");       //here it stops!!!
		a2dp_sink.start("BluetoothSpreaker");
		Serial.println("A2DP active");
	}
	else
	{
		a2dp_sink.stop();
		a2dp_sink.end();
		auto cfg = i2s.defaultConfig(TX_MODE);
		cfg.i2s_format = I2S_LEFT_JUSTIFIED_FORMAT;
		i2s.begin(cfg);
		player.begin(0, false);
		player.setAutoNext(false);
		volume.setVolume(0.1);
	}
}
```

> [D] AudioPlayer.h : 665 - virtual void audio_tools::AudioPlayer::stop()
> [D] AudioPlayer.h : 638 - virtual void audio_tools::AudioPlayer::end()
> [D] AudioOutput.h : 323 - void audio_tools::EncodedAudioStream::end()
> [D] CodecMP3Helix.h : 85 - virtual void audio_tools::MP3DecoderHelix::end()
> [I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
> [D] I2SStream.h : 96 - void audio_tools::I2SStream::end()
> [D] I2SESP32.h : 58 - void audio_tools::I2SBase::end()
> activate A2DP
> Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Decoder exception

> Exception:Error:
> 0x4000c46c: ?? ??:0
> 0x4000c477: ?? ??:0
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:102:::0x4010be8d:gatt_init
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_init.c:97:::0x4010b440:btu_init_core
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:278:::0x4010b0d7:btu_task_start_up
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:226:::0x4010b117:btu_task_thread_handler
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x400901de:vPortTaskWrapper
> 0x40078000: ?? ??:0
> 0x40080400: ?? ??:0
> 0x400806a8: ?? ??:0

podaen, 2021-12-28:
It doesn't crash anymore when I do SPI.end(); Still the a2dp won't start because I can't detected it.

podaen, 2021-12-29:
I tried to following link to make a beter spi connection

https://www.pjrc.com/better-spi-bus-design-in-3-steps/

But still I can't make it work. What I could do is to surpress the multi spi devices and test with only the sd card reader and the A2DP. An other thing would be that I use a card reader that is ment to work on 3.3v. But I m waiting on my order for now.

on boot I work with the SD player of audiotools. After a button is pushed I turn off the player and start A2DP but it won't start.

This is how I end it and giving me the best result for now.

		player.end();
		SPI.end();
		//i2s.end();
		Serial.println("activate A2DP");
		//a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
		a2dp_sink.start("BluetoothSpreaker");
		Serial.println("A2DP active");

podaen, 2021-12-29:
Just found out that it's not the the a2dp sink that won't start... It's the spi that won't end.

podaen, 2021-12-29:
cann't end the spi because of the sd player. Could be a hardware issue. But when I leave the player out it still crashes when I try to begin A2DP. I think I need to start a project with only the a2dp library and the sd player to clear this out.

pschatzmann, 2021-12-29:
I committed some corrections and I think I have it working now.
- A2DP do not try to reconnect when end is in progress
- AudioSourceSdFat AudioFs sd is initalized only once

podaen, 2021-12-29:
You forgot the revert the config file.

> #define I2S_BUFFER_COUNT 20 // 20

pschatzmann, 2021-12-29:
Corrected

podaen, 2021-12-29:
I mad a new project with only the sd player and the esp32-a2dp. now it works without close down the sd player. I have added a the other library I m using in the other project and it still works. I am thinking of the task I have created are the problem to start the  ESP32-A2DP sink.

This was the error I was getting...

```
Exception:Error:
0x4000c46c: ?? ??:0
0x4000c477: ?? ??:0
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:102:::0x4010be8d:gatt_init
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_init.c:97:::0x4010b440:btu_init_core
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:278:::0x4010b0d7:btu_task_start_up
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:226:::0x4010b117:btu_task_thread_handler
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x400901de:vPortTaskWrapper
0x40078000: ?? ??:0
0x40080400: ?? ??:0
0x400806a8: ?? ??:0
```

podaen, 2021-12-29:
The esp exception error descriptor comfirms with what I m thinking. Is there an interfering with my task in the shetch and ESP32-A2DP library?

pschatzmann, 2021-12-29:
I don't know your code, so it's difficult to say. In the A2DP library there is a task which is dispatching the messages.

podaen, 2021-12-29:
I have added the task to my new created project and this is what I found out. There are two reasons why it crashes with the esp32-a2dp library.

1/ To small delay in the loop
2/ To high stacksize in the task

After trying to optimize it, I can connect with my phone to the esp32, but after that it crashes again.

Strang test results:
Stacksize on 50000 = crashes immediately
Stacksize on 30000 = crashes immediately
Stacksize on 10000 = crashes after connecting (PortGetFreeHeapSize: 178992)

```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;
int8_t IntStat = 0;
int8_t IntConnected = 0;

#include <MP3DecoderHelix.h>
#include <helix_pgm.h>
#include <helix_log.h>
#include <CommonHelix.h>
#include <AACDecoderHelix.h>

#include <SPI.h>
#include <SdFat.h>
//audiotools
#include <AudioTools.h>
#include <AudioConfig.h>

//SD
const int SD_CS_PIN = SS;
//#define SPI_SPEED SD_SCK_MHZ(4)
// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 1

#if SD_FAT_TYPE == 0
SdFat sd;
File dir;
File file;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
File32 dir;
File32 file;
File32 DbFile;
//File32 TxtFileWrite;
#elif SD_FAT_TYPE == 2
SdExFat sd;
ExFile dir;
ExFile file;
#elif SD_FAT_TYPE == 3
SdFs sd;
FsFile dir;
FsFile file;
#else  // SD_FAT_TYPE
#error invalid SD_FAT_TYPE
#endif  // SD_FAT_TYPE
// Store error strings in flash to save RAM.
#define error(s) sd.errorHalt(&Serial, F(s))
const uint32_t MAX_FILE_SIZE_MiB = 100;  // 100 MiB file.
const uint32_t MAX_FILE_SIZE = MAX_FILE_SIZE_MiB << 20;
char seperator[2] = ";";
const int MaxTitleLength = 80;// 40;
const int TitleIndex = 1000;

#undef _WIN32

#define USE_HELIX 
#define USE_SDFAT

using namespace audio_tools;

const char* startFilePath = "/";
const char* ext = "mp3";
AudioSourceSdFat source(startFilePath, ext, SD_CS_PIN, 4);//16//4

I2SStream i2s;
VolumeOutput volume(i2s);
LogarithmicVolumeControl lvc(0.1);
MP3DecoderHelix decoder;
//AudioPlayer player(source, i2s, decoder);
AudioPlayer player(source, volume, decoder);

TaskHandle_t Task1;

// the setup function runs once when you press reset or power the board
void setup() {
	

	xTaskCreate(
		Task1code,   /* Task function. */
		"Task1",     /* name of task. */
		10000,       /* Stack size of task */
		NULL,        /* parameter of the task */
		1,           /* priority of the task */
		&Task1);      /* Task handle to keep track of created task */

	delay(500);
}

// the loop function runs over and over again until power down or reset
void loop() {

}

void Task1code(void* pvParameters) {
	Serial.begin(115200);// 115200);//57600//921600//2000000
	while (!Serial);

	//AudioLogger::instance().begin(Serial, AudioLogger::Info);
        //AudioLogger::instance().begin(Serial, AudioLogger::Warning);
	//AudioLogger::instance().begin(Serial, AudioLogger::Debug);
	Serial.println("here");
	// setup output
	auto cfg = i2s.defaultConfig(TX_MODE);
	//cfg.i2s_format = I2S_LEFT_JUSTIFIED_FORMAT;
	i2s.begin(cfg);

	player.begin(0, false);
	player.setAutoNext(false);//moet na begin staan
	//player.begin();
	volume.setVolume(0.1);

	//a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
	a2dp_sink.start("MyMusic");
	//a2dp_sink.set_volume(0.1);

	Serial.println("end setup");

	for (;;) {//infinit loop
		delay(1000);
	}
}
```

pschatzmann, 2021-12-29:
I suggest that you also activate the ESP32 error Logging. 
You can't start the A2DP Sink and I2SStream at the same time and I would expect that you see plenty of Error Messages in the log. You either make sure that only one or the other is started or you deactivate I2S in A2DP and use the I2SStream via a callback (see my example https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/player-sd_a2dp-audiokit)

But I don't think this is the source of your issues.  Some more thoughts: 
- Since A2DP is running it's separate task I would take it out of your task and use it only for the player.
- I am also wondering what forced you to use your own task in the first place ?

podaen, 2021-12-29:
> * I am also wondering what forced you to use your own task in the first place ?

 I started it when I was working with the sd player. This is only a fraction of my project. My project is hugh and I could only have the default loopstack of 8192, that was to low.

> * I think the default stack size is 8192 - can it be that you are running out of heap ?

The reason that I could not increase the stack was because it has to be defined like this.

`#define LOOP_STACK_SIZE 60000`

and than added in here, instead of writing a number in here.

```
	xTaskCreate(
		Task1code,   /* Task function. */
		"Task1",     /* name of task. */
		LOOP_STACK_SIZE,       /* Stack size of task */
		NULL,        /* parameter of the task */
		1,           /* priority of the task */
		&Task1);      /* Task handle to keep track of created task */
```

> You can't start the A2DP Sink and I2SStream at the same time

Humm... I'm not using the I2S as an input, I am using it as an output. deactivating the I2S I could not do because I need it to output to my spreaker. It could be if it's used as input. I think this is what you mean.

> https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/player-sd_a2dp-audiokit)

This will help.

> * Since A2DP is running it's separate task I would take it out of your task and use it only for the player.

Soo use the buttons to start 1 task for A2DP and 1 task for the player?

podaen, 2021-12-30:
I did the player.end() but still getting stuck by starting the A2DP on this line: init_bluetooth();

how can I turn on the logger for the esp32-a2dp library? it's not just by adding this

#include "esp_log.h"
esp_log_level_set(BT_AV_TAG, ESP_LOG_DEBUG);

pschatzmann, 2021-12-30:
In Arduino you activate the logging in Tools -> Core Debug Level.  I still think that it will be pure luck if you have 2 functionalities running the I2S output setup - One of them will fail - and it is not really defined what will happen.

I am still of the opinion that most likely you are running out of heap: You can easily verify this by moving the Bluetooth initialization before the player. Then bluetooth might work and the player might fail. 

I would also suggest that you double check your local variables which are the source of trouble in the first place. Usually you only need a big stack if you have recursive methods. But I assume you use manly arrays that you could put on the heap.

podaen, 2021-12-30:
Just cleared it out. Turning on the debug level needs this line on an arduino IDE

Serial.setDebugOutput(true);

and enabling in the menu tools of the Arduino IDE

podaen, 2021-12-30:
I don't think it's the gap. I checked it.

xPortGetFreeHeapSize: 128920

This is the exception decoder message that I get.

```
Exception:Error:
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2520:::0x400900b8:vQueueDelete
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715:::0x4008ee88:invoke_abort
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715:::0x4008f105:abort
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/semaphore.c:75:::0x400f1525:osi_sem_free
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/fixed_queue.c:88:::0x40120e77:fixed_queue_free
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/fixed_queue.c:67:::0x40120ee0:fixed_queue_new
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_main.c:113:::0x40107fde:gatt_init
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_init.c:97:::0x40107540:btu_init_core
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:278:::0x401071d7:btu_task_start_up
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:226:::0x40107217:btu_task_thread_handler
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1):::0x400901ba:vPortTaskWrapper
0x40078000: ?? ??:0
0x40080400: ?? ??:0
0x400806a8: ?? ??:0
```

I'm not getting any wiser on that. Please check my code again if you can find anything. I think something is still active and therefore crashes on this line

```
    if (esp_bluedroid_init() != ESP_OK) {
        ESP_LOGE(BT_AV_TAG,"Failed to initialize bluedroid");
        return false;
    }
```

pschatzmann, 2021-12-30:
Are you sure that 128920 is enough for bluetooth ?

podaen, 2021-12-30:
You where right. I can increase the stack size for my task more than 10000. But still crashing after connecting to my phone.

> Opening port
> Port open
> V�xPortGetFreeHeapSize: 178920
> 
> here
> [W] AudioPlayer.h : 227 - -> selectStream: 0 '001.mp3'
> player end
> activate A2DP
> [D][BluetoothA2DPSink.cpp:123] start(): start
> [I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 123488
> [I][BluetoothA2DPSink.cpp:136] start(): Device name will be set to 'BluetoothSpreaker'
> [D][BluetoothA2DPSink.cpp:1002] init_nvs(): init_nvs
> [D][BluetoothA2DPCommon.cpp:113] get_last_connection(): get_last_connection
> [D][BluetoothA2DPCommon.cpp:137] get_last_connection(): => b4:ce:40:2c:e1:f1
> [I][BluetoothA2DPSink.cpp:184] init_i2s(): init_i2s
> [I][BluetoothA2DPSink.cpp:186] init_i2s(): init_i2s is_i2s_output
> [D][BluetoothA2DPSink.cpp:266] init_bluetooth(): init_bluetooth
> [I][BluetoothA2DPSink.cpp:271] init_bluetooth(): controller initialized
> [I][BluetoothA2DPSink.cpp:280] init_bluetooth(): bluedroid initialized
> [I][BluetoothA2DPSink.cpp:289] init_bluetooth(): bluedroid enabled
> [D][BluetoothA2DPSink.cpp:387] app_task_start_up(): app_task_start_up
> [D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
> [D][BluetoothA2DPSink.cpp:1108] ccall_app_task_handler(): ccall_app_task_handler
> [D][BluetoothA2DPSink.cpp:358] app_task_handler(): app_task_handler
> [D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
> [D][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, sig 0x1, 0x0
> [I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
> [D][BluetoothA2DPSink.cpp:333] app_work_dispatched(): app_work_dispatched
> [D][BluetoothA2DPSink.cpp:1138] ccall_av_hdl_stack_evt(): ccall_av_hdl_stack_evt
> [D][BluetoothA2DPSink.cpp:830] av_hdl_stack_evt(): av_hdl_stack_evt evt 0
> [D][BluetoothA2DPSink.cpp:835] av_hdl_stack_evt(): av_hdl_stack_evt av_hdl_stack_evt BT_APP_EVT_STACK_UP
> [D][BluetoothA2DPSink.cpp:844] av_hdl_stack_evt(): AVRCP controller initialized!
> [D][BluetoothA2DPSink.cpp:879] av_hdl_stack_evt(): connect_to_last_device
> [D][BluetoothA2DPCommon.cpp:177] connect_to_last_device(): connect_to_last_device
> [D][BluetoothA2DPSink.cpp:884] av_hdl_stack_evt(): set_scan_mode_connectable(true)
> [I][BluetoothA2DPCommon.cpp:255] set_scan_mode_connectable(): set_scan_mode_connectable true
> [D][BluetoothA2DPSink.cpp:1120] ccall_app_a2d_callback(): ccall_app_a2d_callback
> [D][BluetoothA2DPSink.cpp:900] app_a2d_callback(): app_a2d_callback
> [D][BluetoothA2DPSink.cpp:904] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
> [D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
> [D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
> [D][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, sig 0x1, 0x0
> [I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
> [D][BluetoothA2DPSink.cpp:333] app_work_dispatched(): app_work_dispatched
> [D][BluetoothA2DPSink.cpp:1152] ccall_av_hdl_a2d_evt(): ccall_av_hdl_a2d_evt
> [D][BluetoothA2DPSink.cpp:532] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
> [D][BluetoothA2DPSink.cpp:536] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
> [D][BluetoothA2DPSink.cpp:637] handle_connection_state(): handle_connection_state evt 0
> [I][BluetoothA2DPSink.cpp:642] handle_connection_state(): partner address: b4:ce:40:2c:e1:f1
> [I][BluetoothA2DPSink.cpp:650] handle_connection_state(): A2DP connection state: Connecting, [b4:ce:40:2c:e1:f1]
> [I][BluetoothA2DPSink.cpp:722] handle_connection_state(): ESP_A2D_CONNECTION_STATE_CONNECTING
> [I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 5784
> A2DP active
> end setup
> [D][BluetoothA2DPSink.cpp:1120] ccall_app_a2d_callback(): ccall_app_a2d_callback
> [D][BluetoothA2DPSink.cpp:900] app_a2d_callback(): app_a2d_callback
> [D][BluetoothA2DPSink.cpp:904] app_a2d_callback(): app_a2d_callback ESP_A2D_CONNECTION_STATE_EVT
> [D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 16
> [D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
> [D][BluetoothA2DPSink.cpp:365] app_task_handler(): app_task_handler, sig 0x1, 0x0
> [I][BluetoothA2DPSink.cpp:368] app_task_handler(): app_task_handler, APP_SIG_WORK_DISPATCH sig: 1
> [D][BluetoothA2DPSink.cpp:333] app_work_dispatched(): app_work_dispatched
> [D][BluetoothA2DPSink.cpp:1152] ccall_av_hdl_a2d_evt(): ccall_av_hdl_a2d_evt
> [D][BluetoothA2DPSink.cpp:532] av_hdl_a2d_evt(): av_hdl_a2d_evt evt 0
> [D][BluetoothA2DPSink.cpp:536] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
> [D][BluetoothA2DPSink.cpp:637] handle_connection_state(): handle_connection_state evt 0
> [I][BluetoothA2DPSink.cpp:642] handle_connection_state(): partner address: b4:ce:40:2c:e1:f1
> [I][BluetoothA2DPSink.cpp:650] handle_connection_state(): A2DP connection state: Disconnected, [b4:ce:40:2c:e1:f1]
[... 47 lines cut ...]
> [D][BluetoothA2DPSink.cpp:536] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_CONNECTION_STATE_EVT
> [D][BluetoothA2DPSink.cpp:637] handle_connection_state(): handle_connection_state evt 0
> [I][BluetoothA2DPSink.cpp:642] handle_connection_state(): partner address: b4:ce:40:2c:e1:f1
> [I][BluetoothA2DPSink.cpp:650] handle_connection_state(): A2DP connection state: Connected, [b4:ce:40:2c:e1:f1]
[... cut ...]
> [D][BluetoothA2DPSink.cpp:541] av_hdl_a2d_evt(): av_hdl_a2d_evt ESP_A2D_AUDIO_STATE_EVT
> [D][BluetoothA2DPSink.cpp:610] handle_audio_state(): handle_audio_state evt 1
> [I][BluetoothA2DPSink.cpp:612] handle_audio_state(): A2DP audio state: Started
> [I][BluetoothA2DPSink.cpp:623] handle_audio_state(): i2s_start
> /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:706 (xQueueCreateCountingSemaphore)- assert failed!
> abort() was called at PC 0x4008fa8b on core 0

pschatzmann, 2021-12-30:
Try to set it to 8192 - increasing the stack size is reducing the available heap

podaen, 2021-12-30:
I am connected now. I had to end the i2s.

> 		player.end();
> 		i2s.end();
> 		a2dp_sink.start("BluetoothSpreaker");

podaen, 2021-12-30:
Yes it works.

Now I need to get a seprate stack size for the A2DP and for the player. So if a button is pressed I need to delete the task and start a new one I think.

podaen, 2021-12-30:
If I delete the task the memory is still in use. I was affraid for this. because I had that same issue before when I runned my code. 

For example the std is clearing memory by his self. This was the correction I made...

```
char* subString(char* s, int index, int n) {
	std::string Str = std::string(s);
	std::string sub2 = Str.substr(index, n - index);

	//https://www.techiedelight.com/convert-std-string-char-cpp/
	std::vector<char> chars(sub2.begin(), sub2.end());
	chars.push_back('\0');

	char* c = &chars[0];
	return c;

	//Serial.println("sub2");
	//Serial.println(sub2.c_str());
	//char* c = const_cast<char*>(sub2.c_str());
	//char* c = strcpy(new char[sub2.length() + 1], sub2.c_str());
	//const char* c = sub2.c_str();
	//Serial.println("c");
	//Serial.println(c);
	//return c;//.c_str();
	//return sub2.c_str();
	// 
	////char* res = new char[n + 1];
	//char res[n + 1];
	////sprintf(res, "%.*s", n, s + index);
	////sprintf(res, "%.*s", n, s + index);
	//memcpy(res, &s[n], n-index);
	//res[n - index] = '\0';
	////Serial.println(res);
	////delete[] res;
	//return res;
}
```

For example sprintf is ussing a lot of memory and is hard to clear it.

pschatzmann, 2021-12-30:
Releasing a2dp memory is a little bit a mess  - if you call end(true) it will release - but you can't restart it again.
The only reliable way is - not to release - or to restart the processor.

podaen, 2021-12-30:
> The only reliable way is - not to release - or to restart the processor.

I need that memory and I need to switch over and over again, so not release it is not an option. Soo the only thing I could do is restart the processor. I think I can do that with my current setup. Please take a look at what I sad and try to use std. I know it's not always possible but I saved a lot of memory with me.

pschatzmann, 2021-12-30:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/controller_vhci.html#_CPPv229esp_bt_controller_mem_release13esp_bt_mode_t

podaen, 2021-12-30:
IDF 1.0.6 in commen is using sprintf for sure and I cann't find where they clear it out. Take a search in the entire solution, you will see. What I mean is that I saw it in the str.h file with things like sprintf and strcmp.

pschatzmann, 2021-12-30:
sprintf does not allocate any memory and my str.h library does not allocate any memory as well. It is just copying and formatting data which has been allocated by the user before!

By the way your local std::string might be the cause of your stack problems. If you use it with new you can put it on the heap...

podaen, 2021-12-30:
I have tested this

char* subString(char* s, int index, int n) { 	std::string Str = std::string(s); 	std::string sub2 = Str.substr(index, n - index); 	//https://www.techiedelight.com/convert-std-string-char-cpp/ 	std::vector<char> chars(sub2.begin(), sub2.end()); 	chars.push_back('\0'); 	char* c = &chars[0]; 	return c;
}
And it doesn't use memory at all after deleting the task.

pschatzmann, 2021-12-30:
That's right - but you need to reserve stack memory from the heap which will not be available for other use.  If this is good or bad really depends on how much memory you have to spare...

podaen, 2021-12-30:
That's also right. But I don't want that for my application. What I want is use as less memory after deleting a task. And if I still could start it, is it a plus.

podaen, 2021-12-31:
Yep, it's working. I test it with ESP.restart and now I have my memory back.

closed 2021-12-31 by podaen