pschatzmann/arduino-audio-tools #69: A2DP source - set_auto_reconnect option

thanks-for-your-help, 2021-12-30:
Hello,

Thanks for the works on library.

I use both ESP32-A2DP and arduino audio tools.

I want to use a command from ESP32-A2DP in an example from arduino audio tools library : 

set_auto_reconnect

I want to set it to false.

Is there a way ?

Best regards

podaen, 2021-12-30:
You are making a topic in audiotools. Autoreconnect is a part of the esp32-a2dp library. A side of that, I think it's in a2dp_sink.start(name,reconnect).

pschatzmann, 2021-12-30:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect
https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_a2_d_p_stream.html
If you use the A2DPStream you can access the sink by using the sink() method.

podaen, 2022-01-03:
did it work?

thanks-for-your-help, 2022-01-03:
Where is the auto reconnect option in arduino audio tools documentation ?????

https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_a2_d_p_stream.html

It exist and work in ESP32-A2DP. I don't find it in arduino audio tools

pschatzmann, 2022-01-03:
I thought this should be pretty obvious - you just put the 2 things together:

A2DPStream a2dp;
a2dp.sink(). set_auto_reconnect(false, false);

thanks-for-your-help, 2022-01-03:
I'am sorry, I'am i'am better in hardware then software.

Note :  I can't change the sink parameters. I want the source change the sink parameters.

Could I use this in this example ? Where place theses 2 lines ?

/**
 * @file streams-i2s-a2dp.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/stream-i2s-a2dp/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

// Add this in your sketch or change the setting in AudioConfig.h
#define USE_A2DP

#include "AudioTools.h"

using namespace audio_tools;  

I2SStream i2sStream;                            // Access I2S as stream
A2DPStream a2dpStream = A2DPStream::instance(); // access A2DP as stream
StreamCopy copier(a2dpStream, i2sStream); // copy i2sStream to a2dpStream
ConverterFillLeftAndRight<int16_t> filler(RightIsEmpty); // fill both channels

// Arduino Setup
void setup(void) {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Info);
    
    // start bluetooth
    Serial.println("starting A2DP...");
    auto cfgA2DP = a2dpStream.defaultConfig(TX_MODE);
    cfgA2DP.name = "LEXON MINO L";
    a2dpStream.begin(cfgA2DP);

    // start i2s input with default configuration
    Serial.println("starting I2S...");
    a2dpStream.setNotifyAudioChange(i2sStream); // i2s is using the info from a2dp
    i2sStream.begin(i2sStream.defaultConfig(RX_MODE));

}

// Arduino loop - copy data
void loop() {
    copier.copy(filler);
}

pschatzmann, 2022-01-03:
It's actually only one line and you need to add it somewhere before the a2dpStream.begin()
in the example it would be 
a2dpStream.source(). set_auto_reconnect(false, false);

thanks-for-your-help, 2022-01-03:
Ok, I will try tomorrow. 

Thanks-for-your-help

thanks-for-your-help, 2022-01-04:
It seem not working if for example we have 3 sinks with same names (with 3 differents adress) and try connect / disconnect one after one.

I can connect first one. I disconnect first one.
When trying connecting second one the connection seem ok  but program don't continue and stream not working.

Same with third one.

Here is a copy of the logs when trying to connect with second and third : 

[W] AudioA2DP.h : 291 - ==> state: Connecting
[W] AudioA2DP.h : 291 - ==> state: Disconnected
[W] AudioA2DP.h : 291 - ==> state: Connecting
[W] AudioA2DP.h : 291 - ==> state: Connected
[W] AudioA2DP.h : 291 - ==> state: Disconnected
[W] AudioA2DP.h : 291 - ==> state: Connecting
[W] AudioA2DP.h : 291 - ==> state: Disconnected
[W] AudioA2DP.h : 291 - ==> state: Connecting
[W] AudioA2DP.h : 291 - ==> state: Connected
[W] AudioA2DP.h : 291 - ==> state: Disconnected

I think the is A2DP active check is not performed if sink name is the same then previously connected. Could this fonction detect adress change and check is A2DP active ?

pschatzmann, 2022-02-17:
It is my understanding that the original question has been answered.
Your second topic is not supported

closed 2022-02-17 by pschatzmann