pschatzmann/arduino-audio-tools #70: Switch SD player <-> Url player

podaen, 2021-12-31:
I am going to create the AudioConfigLocal.h, with my personal settings.

How could I switch between the sd player and the url player?

SD player

> AudioSourceSdFat source(startFilePath, ext, SD_CS_PIN,5);//16

Url player

> URLStream urlStream(wifi, password);
> AudioSourceIcyUrl source(urlStream, urls, "audio/mp3");

podaen, 2021-12-31:
It seem inpossible this way for me. I think we have to make a general class with evrything in in it and switch on a variable that is set in to the ino.

pschatzmann, 2021-12-31:
I suggest that you create your own subclass of AudioPlayer and add the setAudioSource() method. This way you will be able to switch between different sources.

If you dont' find some major issues with this we could later add it to the generic AudioPlayer.

podaen, 2021-12-31:
It will be a good excerse to learn about classes. I think I can do that. I will give it a try.

podaen, 2021-12-31:
How do we will call the class?

pschatzmann, 2021-12-31:
I suggest to do this in your sketch

podaen, 2021-12-31:
AudioSourceIcyUrl needs to intregraded...

pschatzmann, 2021-12-31:
I am afraied that I don't understand: In your sketch you can add:
class MyAudioPlayer : public AudioPlayer {
}; 
And you have all the functionality of the original AudioPlayer plus your additional functionality

podaen, 2021-12-31:
I think I am starting too...

podaen, 2021-12-31:
is it normal we have two time the same in the audioplayer?

        AudioPlayer(AudioSource& source, AudioStream& output, AudioDecoder& decoder) {
            LOGD(LOG_METHOD);
            this->p_source = &source;
            this->volume_out.begin(output);
            this->p_out_decoding = new EncodedAudioStream(volume_out, decoder);
            this->p_final_stream = &output;

            // notification for audio configuration
            decoder.setNotifyAudioChange(*this);
        }

podaen, 2021-12-31:
This is a hard one for forwarding. Can you help me?

```
	virtual void setMetadataCallback(void (*callback)(MetaDataType type, const char* str, int len), ID3TypeSelection sel = SELECT_ID3) {
		setMetadataCallback((type, str, len), sel)
	}
```

podaen, 2022-01-01:
I having two issues while creating the class.

1/no default constructer excist for the class audio_tools::AudioPlayer
2/MyAudioPlayer player(SDSource, UrlSource, volume, decoder); no instance of constructer AudioPlayer::AudioPlayer match the argument list
```
const char* startFilePath = "/";
const char* ext = "mp3";
AudioSourceSdFat SDSource(startFilePath, ext, SD_CS_PIN, 4);//16//4

const char* urls[] = {
	"http://23553.live.streamtheworld.com:80/RADIO_2_VLAAMSBR_128.mp3",//ok
	"http://20723.live.streamtheworld.com/QFOUTERADIO.mp3",//ok
	"http://25443.live.streamtheworld.com:80/QMUSIC.mp3",//ok
	"http://22183.live.streamtheworld.com/JOE.mp3",//ok
	"http://21303.live.streamtheworld.com/STUDIO_BRUSSEL_128.mp3",//ok
	"http://icecast-servers.vrtcdn.be/stubru_tgs-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/stubru_tijdloze-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/ketnetradio-high.mp3",//ok
	"http://22353.live.streamtheworld.com/MNM_128.mp3",//ok
	"http://icecast-servers.vrtcdn.be/mnm_90s00s-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/mnm_hits-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/mnm_urb-high.mp3",//ok
	"http://fred.torontocast.com:1050/stream",//ok
	"http://icecast-servers.vrtcdn.be/radio1_classics_high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/radio2_benebene-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/radio2_unwind-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/stubru_bruut-high.mp3",//ok
	"http://icecast-servers.vrtcdn.be/stubru_untz-high.mp3",//ok
	"http://randstad.wirelessbelgie.be:8330/stream.mp3"//ok
};
const char* wifi = "ssid";
const char* password = "pass";
URLStream urlStream(wifi, password);
AudioSourceIcyUrl UrlSource(urlStream, urls, "audio/mp3");

class MyAudioPlayer : public AudioPlayer {
public:
	MyAudioPlayer(AudioSource& SDsource, AudioSource& URLsource, AudioStream& output, AudioDecoder& decoder) {
		LOGD(LOG_METHOD);
		this->p_SDsource = &SDsource;
		this->p_Urlsource = &URLsource;
		this->volume_out.begin(output);
		this->p_out_decoding = new EncodedAudioStream(volume_out, decoder);
		this->p_final_stream = &output;
		// notification for audio configuration
		decoder.setNotifyAudioChange(*this);
	}
	/// Default destructor
	~MyAudioPlayer() {
		if (p_out_decoding != nullptr) {
			delete p_out_decoding;
		}
	}
	virtual bool begin(int index = 0, bool isActive = true, bool url = false) {
		LOGD(LOG_METHOD);
		bool result = false;
		//p_Url = url;
		if (url == true) {
			this->p_source = p_Urlsource;
		}
		else
		{
			this->p_source = p_SDsource;
		}
		// navigation supoort
		autonext = p_source->isAutoNext();

		// start dependent objects
		p_out_decoding->begin();
		p_source->begin();
		meta_out.begin();

		if (index >= 0) {
			p_input_stream = p_source->selectStream(index);
			if (p_input_stream != nullptr) {
				if (meta_active) {
					copier.setCallbackOnWrite(decodeMetaData, this);
				}
				copier.begin(*p_out_decoding, *p_input_stream);
				timeout = millis() + p_source->timeoutAutoNext();
				active = isActive;
				result = true;
			}
			else {
				LOGW("-> begin: no data found");
				active = isActive;
				result = false;
			}
		}
		else {
			LOGW("-> begin: no stream selected");
			active = isActive;
			result = false;
		}
		return result;
	}
	virtual void end() {
		end();
	}
	//virtual void setAudioSource() {

	//}
	virtual void setAudioInfo(AudioBaseInfo info) {
		setAudioInfo(info);
	}
	virtual void play() {
		play();
	}
	virtual void stop() {
		stop();
	}
	virtual bool next(int offset) {
		next(offset);
	}
	virtual bool setIndex(int idx) {
		setIndex(idx);
	}
	virtual bool setPath(char* path) {
		setPath(path);
	}
	virtual bool previous(int offset = 1) {
		previous(offset);
	}
	virtual bool setStream(Stream* input) {
		setStream(input);
	}
	virtual bool isActive() {
		isActive();
	}
	operator bool() {
		return isActive();
	}
	virtual void setVolume(float volume) {
		setVolume(volume);
	}
	virtual float volume() {
		volume();
	}
	virtual void setAutoNext(bool next) {
		setAutoNext(next);
	}
	virtual void copy() {
		copy();
	}
	virtual void setMetadataCallback(void (*callback)(MetaDataType type, const char* str, int len), ID3TypeSelection sel = SELECT_ID3) {
		setMetadataCallback(callback, sel);
	}
	//virtual void setVolumeControl(VolumeControl& vc) {

	//}
protected:
	//bool p_Url = false;
	AudioSource* p_SDsource = nullptr;
	AudioSource* p_Urlsource = nullptr;

	static void decodeMetaData(void* obj, void* data, size_t len) {
		decodeMetaData(obj, data, len);
	}
};

I2SStream i2s;
VolumeOutput volume(i2s);
LogarithmicVolumeControl lvc(0.1);
MP3DecoderHelix decoder;

MyAudioPlayer player(SDSource, UrlSource, volume, decoder);
```

podaen, 2022-01-01:
I run the code without the new class, just like this AudioPlayer player(SDSource, volume, decoder); and it crashes because I added

URLStream urlStream(wifi, password);
AudioSourceIcyUrl UrlSource(urlStream, urls, "audio/mp3");

```
Exception:Error:
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112:::0x401ef773:__cxxabiv1
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715:::0x4008f648:invoke_abort
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715:::0x4008f8c5:abort
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112:::0x401ef7ba:std
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87:::0x401eee93:__cxa_throw
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54:::0x401ef1ca:operator new(unsigned int)
/builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_opv.cc:32:::0x401eef31:operator new[](unsigned int)
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools.h(86): error 0x400d866d:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools/Buffers.h:339:::(inlined by) audio_tools:
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioHttp/URLStreamESP32.h:17:::(inlined by) audio_tools:
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools.h(86): error 0x400d875e:audio_tools
E:\Documents\Visual Studio 2019\Projects\Arduino\libraries\arduino-audio-tools\src/AudioTools.h(86): error 0x400d8af9:_GLOBAL__sub_I_keys
(inlined by) _GLOBAL__sub_I_keys at E:\Documents\Visual Studio 2019\Projects\Arduino\A2DP_with_SD-player\A2DP_with_SD-player/A2DP_with_SD-player.ino:544
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:510 (discriminator 3):::0x401d178b:do_global_ctors
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:396:::0x4008495d:start_cpu0_default
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/cpu_start.c:262:::0x40084b8c:call_start_cpu0
0x40079247: ?? ??:0
0x400792ad: ?? ??:0
0x400792b8: ?? ??:0
0x40079465: ?? ??:0
0x400806da: ?? ??:0
0x40007c31: ?? ??:0
0x4000073d: ?? ??:0
0x40078000: ?? ??:0
0x40080400: ?? ??:0
0x400806a8: ?? ??:0
```

podaen, 2022-01-02:
> I run the code without the new class, just like this AudioPlayer player(SDSource, volume, decoder); and it crashes because I added
> 
> URLStream urlStream(wifi, password); AudioSourceIcyUrl UrlSource(urlStream, urls, "audio/mp3");
> exception decode message...

It was because of my tasks I have in my project. I just had to comment:

//#define USE_URLSTREAM_TASK

podaen, 2022-01-02:
> 1/no default constructer excist for the class audio_tools::AudioPlayer

solved it by added 

```
        AudioPlayer() {
            LOGD(LOG_METHOD);
        }
```

Still looking how to solve this

> MyAudioPlayer player(SDSource, UrlSource, volume, decoder); no instance of constructer AudioPlayer::AudioPlayer match the argument list

podaen, 2022-01-02:
> MyAudioPlayer player(SDSource, UrlSource, volume, decoder); no instance of constructer AudioPlayer::AudioPlayer match the argument list

My mistake, it had too be...

MyAudioPlayer(AudioSource& SDsource, AudioSource& URLsource, AudioPrint& output, AudioDecoder& decoder) {

podaen, 2022-01-02:
Ok. It works. Now I want to make AudioConfigLocal.h with all different definitions. I am thinking of defining them by calling from the setup() soo I could choose whitch one to defined, but without succes. The are both defined, condition or not...

I tried...

```
#define DEFAULT_BUFFER_SIZE

void PlayerConfig(bool BlnUrl) {

#if BlnUrl==true
#define DEFAULT_BUFFER_SIZE 11268//1024 // 2048
#else
#define DEFAULT_BUFFER_SIZE 11268//1024 // 2048
#endif

if (BlnUrl == true) {
	 DEFAULT_BUFFER_SIZE = 11268;//1024 // 2048
}
else
{
	DEFAULT_BUFFER_SIZE = 11268;//1024 // 2048
}
```

I don't understand how to switch or redefin between those settings. Please help

podaen, 2022-01-02:
Ha I got it working like this...

if (BlnUrl == false) {
#define DEFAULT_BUFFER_SIZE 11268//1024 // 2048
	Serial.println("BlnUrl-false");
}
else {
#define DEFAULT_BUFFER_SIZE 2048//1024 // 2048
	Serial.println("BlnUrl-true");
}

just need to test it.

podaen, 2022-01-02:
Yes, Phill... we nailed it, after six months of work we have 3 type of streams on 1 esp  ESP32-A2DP - SD player - URL player. I will do a cleanup of my class and imported in to the player. These are some small changes. A new constructer, method and two sources.

closed 2022-01-04 by podaen