pschatzmann/arduino-audio-tools #71: Support of PCM1808 ?

franki29, 2022-01-02:
Hi, I like to use the streams-i2s-webserver_wav example, but I do not know how to set the configuration right. As my pcm1808 board has no clock generator, I need to use the Master mode (Slave mode for pcm1808) and the clock for mclk at PIN 0.

The ADC provide a 24 bit pcm output and additional I need it to be converted to 16bit wav

Thanks for helping
Frank

podaen, 2022-01-02:
I did a quick lookup from you decoder and it only suports 24bit as far I can see. I don't know if it could be converted, if you succed in that it could work. This library only works on 16bit not higher. Here is a list of all supported decoders for now, maybe it gives you an idea how to connect to your board.

https://github.com/pschatzmann/ESP32-A2DP/wiki/External-DAC

franki29, 2022-01-02:
Hi, thanks for your information but I it is a ADC , not a DAC. And if I read the documentation right, there is a 24bit to 16 bit converter in the Audio tools, but I do not know how to configure it in the sample.

podaen, 2022-01-02:
Never used it, can't help you on that. Adc is connecting a mic. Right? I planning to work with a I2s mic, these are extremly fast. Analoog, esp and sound... don't have some good experience with that.

pschatzmann, 2022-01-03:
I have used the following logic to generate the clock signal on pin 0 or 1 in my https://github.com/pschatzmann/arduino-audiokit project. This might work for your purpose as well ?
```
esp_err_t i2s_mclk_gpio_select(i2s_port_t i2s_num, gpio_num_t gpio_num)
{
    if (i2s_num >= I2S_NUM_MAX) {
        KIT_LOGE( "Does not support i2s number(%d)", i2s_num);
        return ESP_ERR_INVALID_ARG;
    }
    if (gpio_num != GPIO_NUM_0 && gpio_num != GPIO_NUM_1 && gpio_num != GPIO_NUM_3) {
        KIT_LOGE( "Only support GPIO0/GPIO1/GPIO3, gpio_num:%d", gpio_num);
        return ESP_ERR_INVALID_ARG;
    }
    KIT_LOGI( "I2S%d, MCLK output by GPIO%d", i2s_num, gpio_num);
    if (i2s_num == I2S_NUM_0) {
        if (gpio_num == GPIO_NUM_0) {
            PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
            WRITE_PERI_REG(PIN_CTRL, 0xFFF0);
        } else if (gpio_num == GPIO_NUM_1) {
            PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
            WRITE_PERI_REG(PIN_CTRL, 0xF0F0);
        } else {
            PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0RXD_U, FUNC_U0RXD_CLK_OUT2);
            WRITE_PERI_REG(PIN_CTRL, 0xFF00);
        }
    } else if (i2s_num == I2S_NUM_1) {
        if (gpio_num == GPIO_NUM_0) {
            PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
            WRITE_PERI_REG(PIN_CTRL, 0xFFFF);
        } else if (gpio_num == GPIO_NUM_1) {
            PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0TXD_U, FUNC_U0TXD_CLK_OUT3);
            WRITE_PERI_REG(PIN_CTRL, 0xF0FF);
        } else {
            PIN_FUNC_SELECT(PERIPHS_IO_MUX_U0RXD_U, FUNC_U0RXD_CLK_OUT2);
            WRITE_PERI_REG(PIN_CTRL, 0xFF0F);
        }
    }
    return ESP_OK;
}
```

To convert the data format you can use e.g. the [FormatConverterStream](https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_format_converter_stream.html)

pschatzmann, 2022-01-03:
I just committed some additional logic for the ESP32 I2S configuration 
The following should generate e.g. a 8Mz signal in pin 1:
  config.use_apll = true;
  config.fixed_mclk = 8000000;
  config.pin_mck = 1;

franki29, 2022-01-03:
Hallo , vielen Dank.
Thank you very much for your help.
I will test it and come back to you.

Br. Frank

franki29, 2022-01-06:
Hello, testing the CSV example, I can see a good result in the data, using 32 Bit and I2S_STD_FORMAT.
But I am struggling getting a sound If I use the settings for the WAV server example. 
As I am not a programmer, I do not know what to do , especially to convert the 32 bit data into 16 bit.
Hope you can send me a example how to do it.

I am trying with this settings, but no luck: ( I try to connect with VLC player)

Serial log:
I] I2SConfig.h : 66 - rx/tx mode: RX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 32
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_MSB_FORMAT
[I] I2SConfig.h : 75 - use_apll: true
[I] I2SConfig.h : 76 - fixed_mclk: 11289600
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 32
I2S started
[I] CodecWAV.h : 436 - is_streamed! because length is 2147418112
....................starting I2S...
[I] I2SConfig.h : 66 - rx/tx mode: RX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 32
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_MSB_FORMAT
[I] I2SConfig.h : 75 - use_apll: true
[I] I2SConfig.h : 76 - fixed_mclk: 11289600
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 32
I2S started
[I] CodecWAV.h : 436 - is_streamed! because length is 2147418112
.....
IP address: xxxxxxxxxxxx
[I] AudioServer.h : 198 - New Client.
[I] AudioServer.h : 338 - sendReply - Returning WAV stream...
[I] CodecWAV.h : 436 - is_streamed! because length is 4294967295
[I] CodecWAV.h : 480 - Writing Header
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

The sketch I am using:

#include "AudioTools.h"

using namespace audio_tools;  

//AudioEncoderServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("ssid","lpassword"); // the same a above

I2SStream i2sStream;    // Access I2S as stream

void setup(){
 Serial.begin(115200);
AudioLogger::instance().begin(Serial, AudioLogger::Info);
     // start i2s input with default configuration
Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_MSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 44100;
  config.channels = 2;
  config.is_master = true;
  config.bits_per_sample = 32;
  config.use_apll = true;
  config.fixed_mclk = 11289600 ;
  config.pin_mck = 0;
  i2sStream.begin(config);
 Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.doLoop();  
}

Thanks for helping
Frank

pschatzmann, 2022-01-10:
WAV should support 32 bit - I rather tend to think that your sample rate is too high.

franki29, 2022-01-11:
Hi, I tested down to 8000, no luck.
CSV Sample is still working

franki29, 2022-01-21:
Hi, me again. As I could not get it to work I bought a different DAC similar shown in the CSV example and with this I can hear the sound clear but with some drops, so I think I have to change the buffer size, but I will open a new issue if I can not find a solution for the drops.
Br. Frank

closed 2022-01-21 by franki29

pschatzmann, 2022-01-21:
I suggest to increase the log level to Warning.

franki29, 2022-01-21:
Already done.
looking at a older issue I add:
#define I2S_BUFFER_SIZE 1024//256
#define I2S_BUFFER_COUNT 40//10
#define DEFAULT_BUFFER_SIZE 2048//1024

to the sketch not sure if this really override the standard settings , but it looks like as it is working a little bit better

pschatzmann, 2022-01-21:
To be sure that the updated values are used you can change these settings in AudioConfig.h or create your local AudioConfigLocal.h file and put them there

franki29, 2022-01-23:
Some news : without changing the buffer size but using ACEncoderFDK , i can manage to play the music without drops.
I did not test it over a longer time, but 20 minutes runs without drops. As the example of the streams-i2s-webserver_aac runs into a memory problem, I used the aac lib that I have found in a older issue. This changed lib does not compile with the newest version of your tools, but with a very dirty trick, I can compile it. First I compile the "Standard" version, than I change the library for aac found in the older discussion and compile again. I can see a error E] AACEncoderFDK.h : 323 - Unable to open encoder: Memory allocation failed␊in the serial console, but it is working and it plays, no idea why and how :-)

franki29, 2022-01-23:
Of course it would be great to get it work without those tricks as this solution is a very cheap way to add a analogue audio input for my Ikea "Sonos" speaker.

franki29, 2022-01-23:
OK found the issue with the help of vernonet why the current aac lib did not work for me.

I changed the AACENC_ERROR rc = aacEncOpen(&handle, 0, channels); to AACENC_ERROR rc = aacEncOpen(&handle, 1, channels); and I change also the int out_size = 20480; to int out_size = 2048; as recommended.

Now it pays with my Sonos Ikea speakers, but not every time, I could not yet figured out why sometimes it is working and sometimes not, using a PC with VLC Player it is working all time, but it takes time until the buffering seems to work.

franki29, 2022-01-23:
Here the code I am using:
#define USE_FDK
#include "AudioTools.h"
#include "AudioCodecs/CodecAACFDK.h"

using namespace audio_tools;

//AudioEncoderServer server(new WAVEncoder(),"ssid","password");
AudioEncoderServer server(new AACEncoderFDK(),"ssid","password");  

I2SStream i2sStream;    // Access I2S as stream

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_MSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 44100;
  config.channels = 2;
  config.is_master = 1;
  config.bits_per_sample = 16;
  config.use_apll = true;
  config.fixed_mclk = 0 ;
  config.pin_mck = 0;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config);
}

// Arduino loop
void loop() {
  // Handle new connections
  server.doLoop();
}

podaen, 2022-01-23:
Add this line to your setup. 

AudioLogger::instance().begin(Serial, AudioLogger::Debug);

And post your log.

pschatzmann, 2022-01-23:
I dont think that this will help. You can adjust the log level in helix by changing HELIX_LOG_LEVEL in fdk_log.h

franki29, 2022-01-24:
Hi thank you both for your information. 
Heklix log level did not show anything but debug level for the Audiologger shows that the client stops. 
So the code seems to be running. I figure out using the Sonos app, the problem is gone. 
Currently I am using a IR remote that is controlled by node red to control my Ikea sonos speaker and it looks like that this is causing the problem of the sporadic working. So for me it is working now. 
The ADC chip I am using is a  AKM5720. Even it says it is a 24bit, it can also run in 
16bit mode. And your code is running perfectly with this chip in slave mode if you use the MLCK provided by the ESp32.
Of course you can not use it for watching TV as the delay is up to 2 seconds. 
Thanks you very much for helping
Frank
[image]

sevenberyl, 2023-02-10:
i use pcm1808 on my esp32,  it can work well , here is some sample code:

`I2SStream in;
FormatConverter<int32_t, int16_t> converter(&NumberConverter::convertFrom32To16);

// setting input
void initInput(){
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgRx.sample_rate = SAMPLE_RATE; //44100
  cfgRx.bits_per_sample = ADC_BIT_PER_SAMPLE; //32
  cfgRx.pin_mck = ADC_PIN_MCK; //3
  cfgRx.pin_bck = ADC_PIN_BCK; //27
  cfgRx.pin_ws = ADC_PIN_WS; //25
  cfgRx.pin_data = ADC_PIN_DATA; //26
  cfgRx.auto_clear = true;
  in.begin(cfgRx);
  xTaskCreate(auxModeBufferReadTask, "bufferReadTask", 1024 * 12, NULL, adc_task_priority, &adcBufferReadHandle);
}

//then
void auxModeBufferReadTask(void *args) {
  const size_t max_buffer_len = 1024;
  const size_t max_buffer_bytes = max_buffer_len * sizeof(int32_t) * 2;
  uint8_t buffer[max_buffer_bytes] = { 0 };
  while (true) {
    // 32bit->16bit
    size_t req_bytes = min(max_buffer_bytes, 2048 * 2 * sizeof(int16_t));
    size_t result_bytes = in.readBytes((uint8_t *)buffer, req_bytes);
    converter.convert(buffer, buffer, result_bytes);
    int len = result_bytes / 2;
    // ..
   process buffer....
  }
}`

pschatzmann, 2023-02-10:
Well, I don't like this because it is way too complicated. Why don't you just request 16 bits in the first place?
If you really want to convert i suggest to use the NumberFormatConverterStream class!

sevenberyl, 2023-02-10:
> Well, I don't like this because it is way too complicated. Why don't you just request 16 bits in the first place? If you really want to convert i suggest to use the NumberFormatConverterStream class instead!

16bit is not work on pcm1808,
NumberFormatConverterStream class mybe better,
i try it

samoyedfranek, 2024-10-22:
> i use pcm1808 on my esp32, it can work well , here is some sample code:
>
> `I2SStream in; FormatConverter<int32_t, int16_t> converter(&NumberConverter::convertFrom32To16);
>
> // setting input void [...]

Hi can you provide entire code? i cant write or find a working examplle for pcm1808 and esp32

benni-t, 2024-11-27:
> i use pcm1808 on my esp32, it can work well , here is some sample code:
>
> `I2SStream in; FormatConverter<int32_t, int16_t> converter(&NumberConverter::convertFrom32To16);
>
> // setting input void [...]

> i use pcm1808 on my esp32, it can work well , here is some sample code:
>
> `I2SStream in; FormatConverter<int32_t, int16_t> converter(&NumberConverter::convertFrom32To16);
>
> // setting input void [...]

Hey buddy, can i contact you? I need help with the pcm1808 =)

Best, Benni

sevenberyl, 2024-12-18:
> Hey buddy, can i contact you? I need help with the pcm1808 =)
>
> Best, Benni

you can contact me at sevenminiapp@126.com