pschatzmann/arduino-audio-tools #76: Guru Mediation Error when attempting to open A2DP sink

programminghoch10, 2022-01-09:
Hello,
I have been using your [ESP32-A2DP](https://github.com/pschatzmann/ESP32-A2DP) library without any issues.
Now I wanted to add some sound effects and for example a startup sound.
I figured using this library would be the thing I want, piping the A2DP stream into I2S and also joining in a generator/file stream.
First goal would be to migrate the project to just copy the A2DP stream to I2S.

However I'm failing at setting up the A2DP stream.
Here's my code:
```cpp
#include <Arduino.h>
#define USE_A2DP
#include <AudioTools.h>

using namespace audio_tools;

A2DPStream a2dp = A2DPStream::instance();
// I2SStream i2s;
// StreamCopy copier(a2dp, i2s);

void setup() {
  // Serial.begin(115200);
  // AudioLogger::instance().begin(Serial, AudioLogger::Info);

  a2dp.begin(RX_MODE, "MyMusic");
  // a2dp.setVolume(0.5f);

  // auto config = i2s.defaultConfig(TX_MODE);
  // config.sample_rate = 44100;
  // config.channels = 2;
  // config.bits_per_sample = 16;
  // config.pin_bck = 32;
  // config.pin_data = 17;
  // config.pin_ws = 16;
  // i2s.begin(config);
}

void loop() {
  if (a2dp.sink().is_connected()) {
    // copier.copy();
  }
}
```
As you can see I got streamcopy and stuff setup, but it's all commented out because I was investigating following issue:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][BluetoothA2DPSink.cpp:1054] set_volume(): set_volume 100
[D][BluetoothA2DPSink.cpp:123] start(): start
[I][BluetoothA2DPCommon.cpp:207] log_free_heap(): Available Heap: 166280
[I][BluetoothA2DPSink.cpp:136] start(): Device name will be set to 'MyMusic'
[D][BluetoothA2DPSink.cpp:1006] init_nvs(): init_nvs
[D][BluetoothA2DPCommon.cpp:113] get_last_connection(): get_last_connection
[D][BluetoothA2DPCommon.cpp:137] get_last_connection(): => 08:ae:d6:1e:f8:68
[I][BluetoothA2DPSink.cpp:184] init_i2s(): init_i2s
[D][BluetoothA2DPSink.cpp:266] init_bluetooth(): init_bluetooth
[I][BluetoothA2DPSink.cpp:271] init_bluetooth(): controller initialized
[I][BluetoothA2DPSink.cpp:280] init_bluetooth(): bluedroid initialized
[I][BluetoothA2DPSink.cpp:289] init_bluetooth(): bluedroid enabled
[D][BluetoothA2DPSink.cpp:387] app_task_start_up(): app_task_start_up
[D][BluetoothA2DPSink.cpp:310] app_work_dispatch(): app_work_dispatch event 0x0, param len 0
[D][BluetoothA2DPSink.cpp:342] app_send_msg(): app_send_msg
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40090c42  PS      : 0x00060b33  A0      : 0x8008f77c  A1      : 0x3ffc8690  
A2      : 0xa5a5a5f1  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060b20  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0x3ffc8690  
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x3f400e54  
A14     : 0x3ffc8500  A15     : 0x00000006  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xa5a5a5f1  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

ELF file SHA256: 0000000000000000

Backtrace: 0x40090c42:0x3ffc8690 0x4008f779:0x3ffc86c0 0x400d3f67:0x3ffc8700 0x400d324a:0x3ffc8730 0x400d3185:0x3ffc8770 0x400d1f5a:0x3ffc87b0 0x400d207d:0x3ffc87f0 0x400d62da:0x3ffc8830 0x4008fe16:0x3ffc8850
  #0  0x40090c42:0x3ffc8690 in uxPortCompareSet at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4560
      (inlined by) vPortCPUAcquireMutexIntsDisabledInternal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h:86
      (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.h:98
      (inlined by) vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4201
  #1  0x4008f779:0x3ffc86c0 in xQueueGenericSend at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2520
  #2  0x400d3f67:0x3ffc8700 in BluetoothA2DPSink::app_send_msg(app_msg_t*) at .pio/libdeps/wemos_d1_mini32/ESP32-A2DP/src/BluetoothA2DPSink.cpp:348
  #3  0x400d324a:0x3ffc8730 in BluetoothA2DPSink::app_work_dispatch(void (*)(unsigned short, void*), unsigned short, void*, int) at .pio/libdeps/wemos_d1_mini32/ESP32-A2DP/src/BluetoothA2DPSink.cpp:324
  #4  0x400d3185:0x3ffc8770 in BluetoothA2DPSink::start(char const*) at .pio/libdeps/wemos_d1_mini32/ESP32-A2DP/src/BluetoothA2DPSink.cpp:154
  #5  0x400d1f5a:0x3ffc87b0 in audio_tools::A2DPStream::begin(audio_tools::A2DPConfig) at /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:80
  #6  0x400d207d:0x3ffc87f0 in audio_tools::A2DPStream::begin(audio_tools::RxTxMode, char const*) at /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:80
      (inlined by) setup() at src/main.cpp:15
  #7  0x400d62da:0x3ffc8830 in loopTask(void*) at /home/user/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
  #8  0x4008fe16:0x3ffc8850 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```

I'm currently using a `Wemos D1 Mini`, but i guess that doesn't matter, it's a ESP32 afterall.

This is my `lib_deps`, but is including `ESP32-A2DP` even necessary or is it automatically included by `arduino-audio-tools`?
```
lib_deps =
  https://github.com/pschatzmann/arduino-audio-tools
  https://github.com/pschatzmann/ESP32-A2DP
```
I'm on `platformio`'s default platform, using the git variant didn't work, it just wouldn't let me compile it.
```
platform = espressif32
;platform = https://github.com/platformio/platform-espressif32.git # feature/arduino-upstream
;platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git # master
```

Also I'd really appreciate if you could add any example for A2DP-Sink with this library, A2DP-Source is well documented, but I didn't any example implementing A2DP-Sink properly using the Streams API of this library, that I could have transferred over to my project.

podaen, 2022-01-09:
Seems you out of memorie

programminghoch10, 2022-01-10:
@pschatzmann Thank you for your quick response.
I was able to transfer the provided example to my usecase without any issues.
But isn't this still defeating the purpose of `A2DPStream RX_MODE`?

> Now I wanted to add some sound effects and for example a startup sound.

As for my plan: Is there an easy way to merge two audio streams into one?
I would like to not have to mute the A2DP Stream while playing a sound, maybe only lower its volume to 50%.

For anyone in future trying the same thing, here is my resulting code, which works:
```cpp
#include <Arduino.h>
#define USE_A2DP
#include <AudioTools.h>
#include <BluetoothA2DPSink.h>

using namespace audio_tools;

BluetoothA2DPSink a2dp;
I2SStream i2s;

void read_a2dp_stream(const uint8_t* data, uint32_t length) {
  i2s.write(data, length);
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  a2dp.set_stream_reader(read_a2dp_stream, false);
  a2dp.start("MyMusic");

  auto config = i2s.defaultConfig(TX_MODE);
  config.sample_rate = 44100;
  config.channels = 2;
  config.bits_per_sample = 16;
  config.pin_bck = 21;
  config.pin_data = 17;
  config.pin_ws = 16;
  i2s.begin(config);
}

void loop() {}
```

Also I don't know if there can be anything done about it, but I have got some serious clipping with my `UDA1334A` in volume levels above ~75% and absolutely horrendous screeching at 100%. Thinking of limiting the volume digitally to this level and using an analog amplifier for higher volume levels.
The `UDA1334A` also supports an LSB 24-bit bitstream, would that be any better than I2S (which afaik is currently running 16-bits)?
The screeching only happens when the bass kicks in, so I'm thinking there isn't enough dynamic range?
Also my phone tells me A2DP is streaming `SBC 44.1KHz 16bits/sample`. Can that be changed to 24 or 32 bits/sample for more dynamic range?

Still closing this now as the reported issue is resolved. Thank you!

pschatzmann, 2022-01-24:
To address the clipping just decrease the volume by using the set_volume method...