pschatzmann/arduino-audio-tools #81: problems with getting basic-a2dp-audiokit and player-sd-audiokit

aszc79, 2022-01-19:
Hello @pschatzmann

I have some troubles with running basic-a2dp-audiokit and player-sd-audiokit from arduino-audio-tools. I use PlatformIO. I have cloned all necessary libraries and I am able to build and upload both examples successfully. However, when I try to run the uploaded code serial monitor output shows many errors which are i2c related, as well as codec init failure and SD init failure and there is no sound. The hardware I use is ESP32-A1S AI-Thinker Audiokit. BTW, I have used two audiokit boards and the result is the same on each of them.

Below I attach output from basic-a2dp-audiokit and  player-sd-audiokit

basic-a2dp-audiokit:

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[W][esp32-hal-i2c.c:1419] i2cCheckLineState(): invalid state sda(18)=1, scl(23)=0
[D][esp32-hal-i2c.c:1427] i2cCheckLineState(): Recovered after 1 Cycles
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[D][esp32-hal-i2c.c:1353] i2cProcQueue():  Gross Timeout Dead start=0x12, end=0x44, =50, max=50 error=0
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbfeec
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffba738
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=0
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffba7c0 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbdb10
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffba79c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=1
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
Error:   ->Wire.endTransmission: 3
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbfeec
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffba738
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffba7c0 bits=112
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbdb10
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffba79c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
Error:   ->Wire.endTransmission: 4
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbfeec
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffba738
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffba7c0 bits=112
[... 1098 lines cut ...]
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioKit.h : 480 - void audio_tools::AudioKitStream::setupActions()
[I] AudioActions.h : 36 - ActionLogic::add pin: 19
[I] AudioActions.h : 52 - pin 19 -> INPUT_PULLUP
[I] AudioActions.h : 36 - ActionLogic::add pin: 13
[I] AudioActions.h : 52 - pin 13 -> INPUT_PULLUP
[I] AudioActions.h : 36 - ActionLogic::add pin: 27
[I] AudioActions.h : 52 - pin 27 -> INPUT_PULLUP
Error:   audio_hal handle is null
[I] AudioKit.h : 269 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), audio_tools::AudioActions::ActiveLogic, void*)
[I] AudioActions.h : 36 - ActionLogic::add pin: 32
[I] AudioActions.h : 52 - pin 32 -> INPUT_PULLUP
[I] AudioKit.h : 269 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), audio_tools::AudioActions::ActiveLogic, void*)
[I] AudioActions.h : 36 - ActionLogic::add pin: 33
[I] AudioActions.h : 52 - pin 33 -> INPUT_PULLUP
[I] AudioPlayer.h : 733 - setVolume(0.700000)
[I] AudioOutput.h : 450 - setVolume: 0.700000
[E] AudioPlayer.h : 209 - SD.begin failed!

--- exit ---

Please write if some additional information on the issue is necessary.

Thanks in advance.

pschatzmann, 2022-01-19:
What is the exact version information of your audio kit?
Please note that I am only supporting later versions which use the ES8388.
I think the first step should be to have the output example from the audio-kit project working...

aszc79, 2022-01-19:
my audiokit exact version is ESP32 Audio Kit v 2.2 2762 (ESP32-A1S AI-Thinker Audiokit) How can I check if it has  ES8388? I have set the board as you have suggested to 5 and though I still get no audio, the device monitor output is different:

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 16
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 22
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   ->Wire.endTransmission: 2
Error:   codec init failed!
Error:   audio_hal_init
[I] AudioKit.h : 188 - Channels 2->2
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioKit.h : 480 - void audio_tools::AudioKitStream::setupActions()
[I] AudioActions.h : 36 - ActionLogic::add pin: 39
[I] AudioActions.h : 52 - pin 39 -> INPUT_PULLUP
[I] AudioActions.h : 36 - ActionLogic::add pin: 18
[I] AudioActions.h : 52 - pin 18 -> INPUT_PULLUP
[I] AudioActions.h : 36 - ActionLogic::add pin: 5
[I] AudioActions.h : 52 - pin 5 -> INPUT_PULLUP
Error:   audio_hal handle is null
[I] AudioKit.h : 269 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), audio_tools::AudioActions::ActiveLogic, void*)
[I] AudioActions.h : 36 - ActionLogic::add pin: 23
[I] AudioActions.h : 52 - pin 23 -> INPUT_PULLUP
[I] AudioKit.h : 269 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), audio_tools::AudioActions::ActiveLogic, void*)
[I] AudioActions.h : 36 - ActionLogic::add pin: 19
[I] AudioActions.h : 52 - pin 19 -> INPUT_PULLUP
[I] AudioPlayer.h : 733 - setVolume(0.700000)
[I] AudioOutput.h : 450 - setVolume: 0.700000
[E] AudioPlayer.h : 209 - SD.begin failed!

Should I order newer board with ES8388?

pschatzmann, 2022-01-21:
I am using V2.2 3478.
I just tried the MP3 AudioPlayer with a SD Drive with PlatformIO and for me it is working with the following platformio.ino:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =  https://github.com/pschatzmann/arduino-audio-tools, https://github.com/pschatzmann/arduino-libhelix, https://github.com/pschatzmann/arduino-audiokit, https://github.com/greiman/SdFat
lib_ldf_mode = deep+
build_flags =  -DCORE_DEBUG_LEVEL=5 -DAUDIOKIT_BOARD=5 -DUSE_SDFAT -DUSE_HELIX
monitor_speed = 115200
```

You could try to comment out the #define AUDIOKIT_USE_WIRE line in the AudioKitSettings.h which will use the native ESP32 I2C implementation - though my hopes are small that this will make a difference.

pschatzmann, 2022-01-21:
Strange: I was getting the 'Error: ->Wire.endTransmission: 2' errors l when I used -DAUDIOKIT_BOARD=1. 
Are you sure that you were using -DAUDIOKIT_BOARD=5

I was reading that there as some AI Thinker Audiokit versions which have a different i2c pin assignment: You can also try to use GPIO_NUM_18 and GPIO_NUM_23 in board_pins_config.c;

        i2c_config->sda_io_num = GPIO_NUM_33; // GPIO_NUM_18
        i2c_config->scl_io_num = GPIO_NUM_32; // GPIO_NUM_23

aszc79, 2022-01-21:
I have changed pins assignment to 18 and 23 and had no luck so I have changed them back to 33 and 32. Surprisingly the log from device monitor has changed. Now it is:

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 16
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 22
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   lib/arduino-audiokit/src/audio_hal/i2c_bus.c:111 (i2c_bus_write_bytes):I2C Bus WriteReg Error
Error:   codec init failed!
Error:   audio_hal_init
[E] AudioKit.h : 183 - begin faild: please verify your AUDIOKIT_BOARD setting

My platformio.ini is:

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =   Wire, https://github.com/pschatzmann/arduino-audio-tools, https://github.com/pschatzmann/arduino-libhelix, https://github.com/pschatzmann/arduino-audiokit, https://github.com/greiman/SdFat
lib_ldf_mode = deep+
build_flags =  -DCORE_DEBUG_LEVEL=5 -DAUDIOKIT_BOARD=5 -DUSE_SDFAT -DUSE_HELIX
monitor_speed = 115200

As you can see, I have added Wire to my platformio.ini and I have also added <Wire.h> to audio_hal/src/i2c_wire.cpp , as I have not been able to build project without it. 

Strangely, I obtain previous behavior, i.e:
Error: ->Wire.endTransmission: 2

aszc79, 2022-01-21:
I am almost sure that it has ac101 (I have managed to run https://github.com/Yveaux/AC101/tree/master/examples/ESP32AudioKit). I am getting two ESP32-LyraT V4.3 at the end of next week and try my luck with these. I will check it and get back to you.

pschatzmann, 2022-01-24:
I committed a DRAFT implementation for the AC101 support
Unfortunately I will not be able to test this because I don't have such a board...

aszc79, 2022-01-24:
I have cloned recent arduino-audio-tools (with ac101). I had to change few lines to get the source compile without errors. However still no luck. The output from device monitor is:

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 16
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 22
Warning: reset succeed
[I] AudioKit.h : 191 - Channels 2->2
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioKit.h : 521 - void audio_tools::AudioKitStream::setupActions()
[I] AudioActions.h : 38 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 54 - pin 5 -> INPUT_PULLUP
[I] AudioKit.h : 274 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 38 - ActionLogic::add pin: 13 / logic: 0
[I] AudioActions.h : 54 - pin 13 -> INPUT_PULLUP
[I] AudioKit.h : 274 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 38 - ActionLogic::add pin: 5 / logic: 3
[I] AudioKit.h : 274 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 38 - ActionLogic::add pin: 18 / logic: 3
[I] AudioActions.h : 57 - pin 18 -> INPUT
[I] AudioKit.h : 274 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 38 - ActionLogic::add pin: 23 / logic: 3
[I] AudioActions.h : 57 - pin 23 -> INPUT
[I] AudioKit.h : 274 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 38 - ActionLogic::add pin: 19 / logic: 3
[I] AudioActions.h : 57 - pin 19 -> INPUT
[I] AudioPlayer.h : 773 - setVolume(0.700000)
[I] AudioOutput.h : 450 - setVolume: 0.700000
[E] AudioPlayer.h : 218 - SD.begin failed with cs=13!
begin() failed
Do not reformat the SD.
No card, wrong chip select pin, or wiring error?
SdError: 0X1,0XFF

I am prettty sure that sd card, wiring and chip select pin are OK as I am able to hear audio using another library.  How should I proceed?

pschatzmann, 2022-01-24:
This is most likely an issue with the dip switches which are in the wrong position. Try to switch them all on (and if this is not helping all off).

You can also deactivate the SD with 
build_flags = -DAUDIOKIT_SETUP_SD=0

aszc79, 2022-01-24:
They were all set to on. I have tried and set all the dip switches off. I did not help. I also checked with an example from an other library and there was no audio. After having switched dip switches the example from the other library works.  But still no luck with AAT.  Still SD.begin failed with cs=13! Please find my board_defs.h for AC101 attached below. Is it ok? 

#ifndef _AUDIO_BOARD_DEFINITION_H_
#define _AUDIO_BOARD_DEFINITION_H_

#include "audio_hal.h"

/* SD card related */
#define PIN_AUDIO_KIT_SD_CARD_CS 13
#define PIN_AUDIO_KIT_SD_CARD_MISO 2
#define PIN_AUDIO_KIT_SD_CARD_MOSI 15
#define PIN_AUDIO_KIT_SD_CARD_CLK  14
#define SD_CARD_INTR_GPIO GPIO_NUM_34

#define I2S_SCLK                    27
#define I2S_LCLK                    26
#define I2S_DSIN                    25
#define I2S_DOUT                    35

#define I2C_CLK                     32
#define I2C_DATA                    33

#define GPIO_PA_EN                  GPIO_NUM_21
#define GPIO_SEL_PA_EN              GPIO_SEL_21

#define HEADPHONE_DETECT GPIO_NUM_5
#define PA_ENABLE_GPIO GPIO_NUM_21

#define GREEN_LED_GPIO GPIO_NUM_22
#define BLUE_LED_GPIO GPIO_NUM_19

#define BUTTON_REC_ID GPIO_NUM_36
#define BUTTON_MODE_ID GPIO_NUM_13

/* Touch pad related */
#define TOUCH_SEL_SET GPIO_SEL_19
#define TOUCH_SEL_PLAY GPIO_SEL_23
#define TOUCH_SEL_VOLUP GPIO_SEL_18
#define TOUCH_SEL_VOLDWN GPIO_SEL_5

#define TOUCH_SET GPIO_NUM_19
#define TOUCH_PLAY GPIO_NUM_23
#define TOUCH_VOLUP GPIO_NUM_18
#define TOUCH_VOLDWN GPIO_NUM_5

/* KEY1 tp 6 */
/*
#define PIN_KEY1 BUTTON_REC_ID
#define PIN_KEY2 BUTTON_MODE_ID
#define PIN_KEY3 TOUCH_SEL_SET
#define PIN_KEY4 TOUCH_SEL_PLAY
#define PIN_KEY5 TOUCH_SEL_VOLUP
#define PIN_KEY6 TOUCH_SEL_VOLDWN
*/

#define PIN_KEY1 36
#define PIN_KEY2 13
#define PIN_KEY3 19
#define PIN_KEY4 23
#define PIN_KEY5 18
#define PIN_KEY6 5

#define BUTTON_REC_ID             PIN_KEY1
#define BUTTON_MODE_ID            PIN_KEY2
#define BUTTON_SET_ID             PIN_KEY3
#define BUTTON_PLAY_ID            PIN_KEY4
#define BUTTON_VOLDOWN_ID         PIN_KEY5
#define BUTTON_VOLUP_ID           PIN_KEY6

extern audio_hal_func_t AUDIO_CODEC_AC101_CODEC_HANDLE;

#define AUDIO_CODEC_DEFAULT_CONFIG() {       \
    .adc_input = AUDIO_HAL_ADC_INPUT_LINE1,  \
    .dac_output = AUDIO_HAL_DAC_OUTPUT_ALL,  \
    .codec_mode = AUDIO_HAL_CODEC_MODE_BOTH, \
    .i2s_iface = {                           \
        .mode = AUDIO_HAL_MODE_SLAVE,        \
        .fmt = AUDIO_HAL_I2S_NORMAL,         \
        .samples = AUDIO_HAL_48K_SAMPLES,    \
        .bits = AUDIO_HAL_BIT_LENGTH_16BITS, \
    },                                       \
};

#define INPUT_KEY_NUM     6

#define INPUT_KEY_DEFAULT_INFO() {                      \
     {                                                  \
        .type = PERIPH_ID_BUTTON,                       \
        .user_id = INPUT_KEY_USER_ID_REC,               \
        .act_id = BUTTON_REC_ID,                        \
    },                                                  \
    {                                                   \
        .type = PERIPH_ID_BUTTON,                       \
        .user_id = INPUT_KEY_USER_ID_MODE,              \
        .act_id = BUTTON_MODE_ID,                       \
    },                                                  \
    {                                                   \
        .type = PERIPH_ID_TOUCH,                        \
        .user_id = INPUT_KEY_USER_ID_SET,               \
        .act_id = BUTTON_SET_ID,                        \
    },                                                  \
    {                                                   \
        .type = PERIPH_ID_TOUCH,                        \
        .user_id = INPUT_KEY_USER_ID_PLAY,              \
        .act_id = BUTTON_PLAY_ID,                       \
    },                                                  \
    {                                                   \
        .type = PERIPH_ID_TOUCH,                        \
        .user_id = INPUT_KEY_USER_ID_VOLUP,             \
        .act_id = BUTTON_VOLUP_ID,                      \
    },                                                  \
    {                                                   \
        .type = PERIPH_ID_TOUCH,                        \
        .user_id = INPUT_KEY_USER_ID_VOLDOWN,           \
        .act_id = BUTTON_VOLDOWN_ID,                    \
    }                                                   \
}

#endif

aszc79, 2022-01-24:
I have added 

SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

to the main.cpp of my example - no I can hear some audio but on 1sec repeatedly. Could you advise me what I could do?

Pleas find output of device monitor attached below:

  --- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 5 Active 1 - 1
[I] AudioKit.h : 324 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 306 - incrementVolume: -2 -> 38
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 18 Active 1 - 0
[I] AudioKit.h : 315 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 306 - incrementVolume: 2 -> 40
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 23 Active 1 - 0
[W] AudioPlayer.h : 228 - -> nextStream: 1
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '': 1 - closed
[W] AudioPlayer.h : 237 - -> selectStream: 1 ''
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[I] AudioActions.h : 115 - processActions: 19 Active 1 - 1
[W] AudioPlayer.h : 228 - -> nextStream: -1
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '1.mp3': 0 - open
[W] AudioPlayer.h : 237 - -> selectStream: 0 '1.mp3'
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 5 Active 1 - 1
[I] AudioKit.h : 324 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 306 - incrementVolume: -2 -> 38
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 18 Active 1 - 0
[I] AudioKit.h : 315 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 306 - incrementVolume: 2 -> 40
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 23 Active 1 - 0
[W] AudioPlayer.h : 228 - -> nextStream: 1
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '': 1 - closed
[W] AudioPlayer.h : 237 - -> selectStream: 1 ''
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[I] AudioActions.h : 115 - processActions: 19 Active 1 - 1
[W] AudioPlayer.h : 228 - -> nextStream: -1
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '1.mp3': 0 - open
[W] AudioPlayer.h : 237 - -> selectStream: 0 '1.mp3'
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[... 99 lines cut ...]
[I] AudioKit.h : 306 - incrementVolume: -2 -> 38
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 18 Active 1 - 0
[I] AudioKit.h : 315 - static void audio_tools::AudioKitStream::actionVolumeUp(bool, int, void*)
[I] AudioKit.h : 306 - incrementVolume: 2 -> 40
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: 23 Active 1 - 0
[W] AudioPlayer.h : 228 - -> nextStream: 1
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '': 1 - closed
[W] AudioPlayer.h : 237 - -> selectStream: 1 ''
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[... cut ...]
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '1.mp3': 0 - open
[W] AudioPlayer.h : 237 - -> selectStream: 0 '1.mp3'
[I] MetaDataID3.h : 551 - virtual void audio_tools::MetaDataID3::end()
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

pschatzmann, 2022-01-24:
Which board is this ?

aszc79, 2022-01-24:
it is ESP32-A1S AI-Thiker v 2.2 2762.

pschatzmann, 2022-01-24:
For the broken playing I suggest to comment out all logic related to buttons. It seems that some actions are executed w/o pressing any buttons.

pschatzmann, 2022-01-24:
You can base your validation if the buttons are working on the following sketch
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-audiokit/streams-pins-audiokit

pschatzmann, 2022-01-24:
ps. if I got this right: if the SD is active you must not use PIN_KEY2 - which uses the same GPIO like the SD CS

aszc79, 2022-01-24:
Yes, I have noticed a conflict on PIN 13 but I have commented it out. And also commented some of keys related logic and I can see some progress - now it plays whole mp3 but with breaks - like hearing streaming on bad Internet connection - audio-silence-audio-silence. 

Output of device monitor:

-- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 115 - processActions: -1 Active 0 - 0
[I] AudioKit.h : 333 - static void audio_tools::AudioKitStream::actionStartStop(bool, int, void*)

Could you advise me what to comment out? It seems that start/stop playback condition is triggered.

pschatzmann, 2022-01-24:
something like the following should work:
kit.audioActions().setEnabled(PIN_KEY2 , false);

aszc79, 2022-01-24:
I inserted the line of code in my main.cpp. Alas, no luck - audio is torn. Please find a fragment of my main.cpp listing below:

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  kit.begin(cfg);
  
  //AQQ
  kit.audioActions().setEnabled(BUTTON_MODE_ID, false);

  /* 
 // setup additional buttons 
  kit.addAction(PIN_KEY4, next);
  kit.addAction(PIN_KEY3, previous);
  */

  // setup player
  player.setVolume(0.7);
  player.begin();
}

void loop() {
  player.copy();
  kit.processActions();
}

pschatzmann, 2022-01-24:
Does it improve if you comment out the kit.processActions(); ?
If not what are the log entries now ?

aszc79, 2022-01-24:
Well, I have followed the path you suggested but with some modifications. I have  modified ai_thinker_ac101/board_def.h - my change is:

#define PIN_KEY1 -1 //36
#define PIN_KEY2 13
#define PIN_KEY3 19
#define PIN_KEY4 23 
#define PIN_KEY5 18
#define PIN_KEY6 5
 
Now audio is ok. Thanks for you assistance. Tomorrow I will try to get some sound from bluetooth streaming.

aszc79, 2022-01-24:
Using pins configured as in my previous post the kit plays mp3 just fine. Sometimes it is true but now and then, although software is the same,  the kit is mute.  Also I do not think that a solution presented in my previous post is particularly elegant. Could you please advise me on a better solution? 

Below are logs - the first was generated when the kit was playing mp3, whereas the other was generated when the kit was silent.

0) proper playback of an mp3:

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 16
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 22
Warning: reset succeed
[I] AudioKit.h : 191 - Channels 2->2
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioKit.h : 523 - void audio_tools::AudioKitStream::setupActions()
[I] AudioActions.h : 38 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 54 - pin 5 -> INPUT_PULLUP
[I] AudioKit.h : 276 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[W] AudioKit.h : 517 - Undefined ActionLogic for pin: -1 
[I] AudioActions.h : 38 - ActionLogic::add pin: -1 / logic: 0
[I] AudioActions.h : 54 - pin -1 -> INPUT_PULLUP
[I] AudioKit.h : 276 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[W] AudioKit.h : 517 - Undefined ActionLogic for pin: 5 
[I] AudioActions.h : 38 - ActionLogic::add pin: 5 / logic: 0
[I] AudioKit.h : 276 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[W] AudioKit.h : 517 - Undefined ActionLogic for pin: 18 
[I] AudioActions.h : 38 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 54 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 276 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[W] AudioKit.h : 517 - Undefined ActionLogic for pin: 23 
[I] AudioActions.h : 38 - ActionLogic::add pin: 23 / logic: 0
[I] AudioActions.h : 54 - pin 23 -> INPUT_PULLUP
[I] AudioKit.h : 276 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[W] AudioKit.h : 517 - Undefined ActionLogic for pin: 19 
[I] AudioActions.h : 38 - ActionLogic::add pin: 19 / logic: 0
[I] AudioActions.h : 54 - pin 19 -> INPUT_PULLUP
[I] AudioPlayer.h : 773 - setVolume(1.000000)
[I] AudioOutput.h : 450 - setVolume: 1.000000
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioPlayer.h : 333 - directory: '/'
[I] AudioPlayer.h : 340 - -> getFile: '1.mp3': 0 - open
[W] AudioPlayer.h : 237 - -> selectStream: 0 '1.mp3'
[I] AudioPlayer.h : 691 - sample_rate: 48000
[I] AudioPlayer.h : 692 - bits_per_sample: 16
[I] AudioPlayer.h : 693 - channels: 2
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... 73 lines cut ...]
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... cut ...]
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

pschatzmann, 2022-01-24:
Strange that kit.audioActions().setEnabled(PIN_KEY2 , false); was not working.
I changed the code in the AudioActions so that negative pins are ignored...

aszc79, 2022-01-24:
Could please you point me to a better solution that does not rely on assigning negative pins? 

I have checked and  SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS); in main.cpp seems to be a way but I do not know if the only one. I have to admit that I also use  -DAUDIOKIT_SETUP_SD=0 build flag and commented USE_SDFAT  in my main.cpp.  However, please take into account that I have come to this setup only before you added support to AC101. If you think that newer version of AAT can resolve some of SD card related problems,  I can set up a new clean sandbox project to test SD support via SdFat. .

pschatzmann, 2022-01-24:
Better solution: Deactivate all default actions and just add the one to the sketch that you want to have. See setupActions() in https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioLibs/AudioKit.h

I think it is worth the effort to have the SD functionality working in the AudioKitHAL.h.

aszc79, 2022-01-25:
I have successfully run basic-a2dp-audiokit At first audio was torn. However, I commented out kit.processActions(); in loop and audio now it is perfect. Thanks.

pschatzmann, 2022-01-27:
I think I found the issue with the SD functionality. In the AudioKitConfig the default value for sd_active was false. I changed it to true

aszc79, 2022-01-27:
I have just tried it - commented spi.begin in main.cpp and changed sd_active to true in  AudioKit.h and then also in AudioKitHal.h and unfortunately I get:

[...]
Do not reformat the SD.
No card, wrong chip select pin, or wiring error?
SdError: 0X1,0XFF

after uncommenting SPI.begin in main.cpp all works as expected.

pschatzmann, 2022-01-27:
I suggest that after you upgrade both libraries to the latest version you try again. In the example maybe it helps if you define the AudioKitStream before the AudioSourceSdFat. For me however both are working with the latest version.

aszc79, 2022-01-27:
I have checked out the latest source and now I am struggling to compile the new project. As for now I am getting:

.pio/libdeps/esp32dev/audio-tools/src/AudioLibs/AudioKit.h:518:72: error: 'INPUT_KEY_DEFAULT_INFO' was not declared in this scope

pschatzmann, 2022-01-27:
I tried to add the missing definitions. 
I hope I got everything right...

aszc79, 2022-01-27:
I can confirm that SD card works with the latest versions - I removed SPI.begin call from my main.cpp and it works. However, every now and then I can see  streamCopy:: copy and can hear no sound - resetting the board helps. I suspect that it might be connected with some errors related to SD card initialization. I will get to it tomorrow and will also check INPUT_KEY_DEFAULT_INFO issue against the latest sources

aszc79, 2022-01-28:
I can confirm that AAT with ac101 now compiles and uploads firmware flawlessly. However, it is still an issue that sometimes I can see streamCopy:: copy and can hear no sound.

aszc79, 2022-01-28:
I am not sure whats the case with the issue I have mentioned but playing mp3 from SD works more reliably with https://github.com/schreibfaul1/ESP32-audioI2S and https://github.com/schreibfaul1/AC101 The tested main.cpp is:

// found this sketch on https://www.mikrocontroller.net/topic/474383
// thanks to Michael U. (amiga)

#include "Arduino.h"
#include "WiFi.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"
#include "Wire.h"
#include "AC101.h" //https://github.com/schreibfaul1/AC101
// #include "ES8388.h"  // https://github.com/maditnerd/es8388
#include "Audio.h" //https://github.com/schreibfaul1/ESP32-audioI2S

// SPI GPIOs
#define SD_CS          13
#define SPI_MOSI       15   // SD Card
#define SPI_MISO       2
#define SPI_SCK        14

//amplifier enable
#define GPIO_PA_EN    21

#define IIS_SCLK                    27
#define IIS_LCLK                    26
#define IIS_DSIN                    25
#define IIS_DOUT                    35

#define IIC_CLK                     32
#define IIC_DATA                    33

#define GPIO_PA_EN                  GPIO_NUM_21
#define GPIO_SEL_PA_EN              GPIO_SEL_21

//Switch S1: 1-OFF, 2-ON, 3-ON, 4-OFF, 5-OFF

String ssid =     "xxxxxxxxx";
String password = "xxxxxxxxx";

static AC101 dac;                                 // AC101
//    ES8388 dac;                                 // ES8388 (new board)
int volume = 100;                                  // 0...100

Audio audio;

//#####################################################################

void setup()
{
  Serial.begin(115200);
  Serial.println("\r\nReset");
  Serial.printf_P(PSTR("Free mem=%d\n"), ESP.getFreeHeap());

  pinMode(SD_CS, OUTPUT);
  digitalWrite(SD_CS, HIGH);
  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPI.setFrequency(1000000);

  SD.begin(SD_CS);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid.c_str(), password.c_str());

  /*while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(100);
  }*/

  Serial.printf_P(PSTR("Connected\r\nRSSI: "));
  Serial.print(WiFi.RSSI());
  Serial.print(" IP: ");
  Serial.println(WiFi.localIP());

    Serial.printf("Connect to DAC codec... ");
    while (not dac.begin(IIC_DATA, IIC_CLK))
    {
        Serial.printf("Failed!\n");
        delay(1000);
    }
    Serial.printf("OK\n");

    dac.SetVolumeSpeaker(volume);
    dac.SetVolumeHeadphone(volume);
//  ac.DumpRegisters();

    // Enable amplifier
    pinMode(GPIO_PA_EN, OUTPUT);
    digitalWrite(GPIO_PA_EN, HIGH);

    //audio.setInternalDAC(true);

    //audio.i2s_mclk_pin_select(IIS_);

    audio.setPinout(IIS_SCLK, IIS_LCLK, IIS_DSIN);
    audio.setVolume(21); // 0...21

    audio.connecttoFS(SD, "/1.mp3");
//  audio.connecttoSD("/Banana Boat Song - Harry Belafonte.mp3");
//  audio.connecttohost("http://mp3channels.webradio.antenne.de:80/oldies-but-goldies");
//  audio.connecttohost("http://dg-rbb-http-dus-dtag-cdn.cast.addradio.de/rbb/antennebrandenburg/live/mp3/128/stream.mp3");
//  audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");

}

//-----------------------------------------------------------------------

void loop()
{
    audio.loop();
}

// optional
void audio_info(const char *info){
    Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info){  //id3 metadata
    Serial.print("id3data     ");Serial.println(info);
}
void audio_eof_mp3(const char *info){  //end of file
    Serial.print("eof_mp3     ");Serial.println(info);
}
void audio_showstation(const char *info){
    Serial.print("station     ");Serial.println(info);
}
void audio_showstreamtitle(const char *info){
    Serial.print("streamtitle ");Serial.println(info);
}
void audio_bitrate(const char *info){
    Serial.print("bitrate     ");Serial.println(info);
}
void audio_commercial(const char *info){  //duration in sec
    Serial.print("commercial  ");Serial.println(info);
}
void audio_icyurl(const char *info){  //homepage
    Serial.print("icyurl      ");Serial.println(info);
}
void audio_lasthost(const char *info){  //stream URL played
    Serial.print("lasthost    ");Serial.println(info);
}
void audio_eof_speech(const char *info){
    Serial.print("eof_speech  ");Serial.println(info);
}

aszc79, 2022-01-28:
I have just randomly added delay(2000); in AudioPlayer.h after line 220 and it seems to be better. However, it still happens that I can get no sound. I will buy a couple of sd cards to see if it is not an issue. 

Logs (with no audio):

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] I2SConfig.h : 66 - rx/tx mode: TX
[I] I2SConfig.h : 67 - port_no: 0
[I] I2SConfig.h : 68 - is_master: Master
[I] I2SConfig.h : 69 - sample rate: 44100
[I] I2SConfig.h : 70 - bits per sample: 16
[I] I2SConfig.h : 71 - number of channels: 2
[I] I2SConfig.h : 72 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 80 - pin_bck: 14
[I] I2SConfig.h : 81 - pin_ws: 15
[I] I2SConfig.h : 82 - pin_data: 22
Warning: reset succeed
[I] AudioKit.h : 192 - Channels 2->2
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioKit.h : 538 - void audio_tools::AudioKitStream::setupActions()
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 58 - pin 5 -> INPUT_PULLUP
[I] AudioKit.h : 291 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioKit.h : 291 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 58 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 291 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 23 / logic: 0
[I] AudioActions.h : 58 - pin 23 -> INPUT_PULLUP
[I] AudioKit.h : 291 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 19 / logic: 0
[I] AudioActions.h : 58 - pin 19 -> INPUT_PULLUP
[I] AudioPlayer.h : 776 - setVolume(1.000000)
[I] AudioOutput.h : 450 - setVolume: 1.000000
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioPlayer.h : 336 - directory: '/'
[I] AudioPlayer.h : 343 - -> getFile: '1.mp3': 0 - open
[W] AudioPlayer.h : 240 - -> selectStream: 0 '1.mp3'
[I] AudioPlayer.h : 694 - sample_rate: 48000
[I] AudioPlayer.h : 695 - bits_per_sample: 16
[I] AudioPlayer.h : 696 - channels: 2
[I] AudioOutput.h : 737 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 750 - FormatConverterStream: no conversion
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

pschatzmann, 2022-01-28:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Performance

aszc79, 2022-01-28:
I have tried changing frequencies and some loops. Still no luck - every now and then (5 out of 10 times) SD player does not produce any sound, whereas Bluetooth streaming works flawlessly.

pschatzmann, 2022-01-29:
Do I understand it correctly: If you restart in 50% you can not hear any audio while you get 
[I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
entries ?

aszc79, 2022-01-29:
I have double checked with player-sd_a2dp-audiokit.ino, you understand it correctly. Switching Button 4 back and forth I randomly get audio from SD or get no audio at all whereas receiving and playing Bluetooth works without any problems. The same regards Resets - sometimes there is some audio from SD and sometimes there is not any.

pschatzmann, 2022-01-29:
Oh...that's strange.
1) If you plug in the earphones - do you get the same issue ?
2) Does it make a difference, if you comment out the   kit.processActions();

pschatzmann, 2022-01-29:
You have a A1S board right ? I just realized that there is also potentially a pin confilict with the headphone detection. But before I change anything I would like that you confirm that all buttons are mapped correctly. 
You can check them e.g. with 
https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-pins-audiokit/

aszc79, 2022-01-29:
It seems not to make any difference. I have tested it with a modified minimal example player-sd-audiokit.ino Below is the listing:

#define USE_SDFAT
#define USE_HELIX  // or USE_MAD
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

using namespace audio_tools;  

const char *startFilePath="/";
const char* ext="mp3";
int speedMz = 10;
AudioSourceSdFat source(startFilePath, ext, PIN_AUDIO_KIT_SD_CARD_CS, speedMz);
AudioKitStream kit;
MP3DecoderHelix decoder;  // or change to MP3DecoderMAD
AudioPlayer player(source, kit, decoder);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  kit.begin(cfg);

  // setup player
  player.setVolume(0.7);
  player.begin();
}

void loop() {
  player.copy();
}

Randomly, after so rests the audio plays fine, after some it is silence.

pschatzmann, 2022-01-29:
I am running out of ideas what could cause this:
1) Does it make a difference if you add a delay(5000) before the kit.begin() ?
2) in AudioKitSettings.h try to change #define AUDIOKIT_USE_WIRE 0

aszc79, 2022-01-29:
I have tested the example you have mentioned. Buttons 3,4,5,6 work as expected. However, buttons 1 and 2 do not trigger any audio output.

pschatzmann, 2022-01-29:
I am running out of ideas what could cause this:

- Does it make a difference if you add a delay(5000) before and after the kit.begin() ?
- Does it make a difference to add cfg.default_actions_active = false ?
- in AudioKitSettings.h try to change the setting to #define AUDIOKIT_USE_WIRE 0

pschatzmann, 2022-02-01:
I have updated recently the logic related to pin conflicts both in the AudioKit and AudioTools libraries.
It might be that this makes a difference.
Unfortunately I can't test it because I do not have any  AC101 board...

pschatzmann, 2022-02-17:
In the meantime I received a AC101 board and did not notice any issues using the latest code

closed 2022-02-17 by pschatzmann