pschatzmann/arduino-audio-tools #87: AudioKit.h : 183 - begin faild: please verify your AUDIOKIT_BOARD  setting: 5

madskjeldgaard, 2022-01-23:
Hello! First of all awesome work with all these wonderful audio libs !

I am trying my hand at the esp32 AI tinker v2.2 example and have modified it to work for my platformio setup. It compiles okay, but I get the following error. As far as I understand from the library, setting `AUDIOKIT_BOARD` to `5` will make it work with the ai tinker v2.2 board, but it seems to disagree:

 ```
[I][esp32-hal-i2c.c:1138] i2cProcQueue(): Bus busy, reinit
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbfefc
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffba908
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffba990 bits=112
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbaafc
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffba96c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
Error:   ->Wire.endTransmission: 4
Error:   codec init failed!
Error:   audio_hal_init
[E] AudioKit.h : 183 - begin faild: please verify your AUDIOKIT_BOARD
 setting: 5
```
Here is the main.cpp file from my project:
```cpp
/**
 * @file basic-a2dp-audiokit.ino
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/basic-a2dp-audiokit/README.md
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#define USE_A2DP
#define AUDIOKIT_BOARD 5 // ai tinker v2.2
#define AUDIOKIT_USE_WIRE 0 // use esp32 wire lib

#include "Arduino.h"
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

BluetoothA2DPSink a2dp_sink;
AudioKitStream kit;

// Write data to AudioKit in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    kit.write(data, length);
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  kit.begin(cfg);

  // register callback
  a2dp_sink.set_stream_reader(read_data_stream, false);
  a2dp_sink.start("AudioKit");
}

void loop() {
  kit.processActions();
}
```

And the platformio.ini:
```
[platformio]
description = A2DP Example
default_envs = esp32dev

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = Wire, https://github.com/pschatzmann/ESP32-A2DP, https://github.com/pschatzmann/arduino-audio-tools, https://github.com/pschatzmann/arduino-audiokit, https://github.com/pschatzmann/arduino-libhelix
build_flags = -DCORE_DEBUG_LEVEL=5
monitor_speed = 115200
```

Any pointers on what to do? THANKS!

pschatzmann, 2022-01-23:
I suggest that you try it as documented in https://github.com/pschatzmann/arduino-audio-tools/wiki/Working-with-PlatformIO

If this is not working you might not have an old AudioKit version  which is not using the es8388 but the old AC101 (which I do not support)

madskjeldgaard, 2022-01-23:
> AC101
Thanks for the reply! 

If I use the examples in that post, it compiles alright and the serial monitor does not display any problems but once I start using AudioKit modifications of it, the trouble starts again. Damn...

pschatzmann, 2022-01-23:
I might be able to provide a DRAFT implementation for the AC101. But you would need to find and resolve any issues to complete the implementation.

madskjeldgaard, 2022-01-23:
> I might be able to provide a DRAFT implementation for the AC101. But you would need to find and resolve any issues to complete the implementation.

I will investigate a bit further whether or not it's actually the AC101 version I have here. (it says "v2.2 2957" on the board)

pschatzmann, 2022-01-23:
See https://github.com/pschatzmann/arduino-audio-tools/issues/81
3478 seems to be AC101

madskjeldgaard, 2022-01-23:
According to this: https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit#developing-with-the-esp-adf

It seems that if it says "2974" under the antenna, it is a ES8388 board, but not sure.

pschatzmann, 2022-01-23:
You can check with https://github.com/Yveaux/AC101/tree/master/examples/ESP32AudioKit
If this works it's a AC101

madskjeldgaard, 2022-01-23:
> You can check with https://github.com/Yveaux/AC101/tree/master/examples/ESP32AudioKit If this works it's a AC101

Thanks. This does work on my board. Sounds is output and the serial monitor looks alright, so I guess it is a ac101 board I have here. Thanks for the help

pschatzmann, 2022-01-23:
I committed a DRAFT implementation for the AC101 support

madskjeldgaard, 2022-01-23:
> I committed a DRAFT implementation for the AC101 support

Thanks that's very sweet of you. I seem to have gotten it to work now - will test the sound properly and get back to you!

closed 2022-02-17 by pschatzmann

madskjeldgaard, 2022-04-30:
Hi Phil

Finally getting back to this. It seems it ALMOST works but not entirely. The code now compiles, and I can tell from the monitor output that it receives audio data and prints along the lines of:

```
[130485][D][BluetoothA2DPSink.cpp:932] audio_data_callback(): audio_data_callback
[130486][D][BluetoothA2DPSink.cpp:939] audio_data_callback(): stream_reader
```
Which I guess means that that part of it works fine. But there's still no audio output. I wonder what the problem might be ? 

Below you can see my code:
```cpp
/**
 * @file basic-a2dp-audiokit.ino
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/basic-a2dp-audiokit/README.md
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#define USE_A2DP
#define AUDIOKIT_BOARD 6 // ai tinker v2.2
#define AUDIOKIT_USE_WIRE 1 // use esp32 wire lib

#include "Arduino.h"
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

BluetoothA2DPSink a2dp_sink;
AudioKitStream kit;

// Write data to AudioKit in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    kit.write(data, length);
}

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  kit.begin(cfg);
  kit.setVolume(50);
  kit.setActive(true);
  kit.setMute(false);

  // register callback
  a2dp_sink.set_stream_reader(read_data_stream, false);
  a2dp_sink.start("heheheheheHE");

}

void loop() {
  kit.processActions();
}
```

And the .ini file:

 ```
 ; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
description = A2DP Example
default_envs = esp32dev

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	Wire
	https://github.com/Yveaux/AC101/
	https://github.com/pschatzmann/ESP32-A2DP
	https://github.com/pschatzmann/arduino-audio-tools
	https://github.com/pschatzmann/arduino-audiokit
	https://github.com/pschatzmann/arduino-libhelix
lib_ldf_mode = deep+
build_flags = -DCORE_DEBUG_LEVEL=5 -DAUDIOKIT_BOARD=5 -DUSE_HELIX
monitor_speed = 115200
```

pschatzmann, 2022-04-30:
Remove the defines from your sketch: This is not supported!
The right place is the build_flags.
ps you seem to have a mess with the DAUDIOKIT_BOARD=5  and in the sketch you use 6

madskjeldgaard, 2022-04-30:
> Remove the defines from your sketch: This is not supported! The right place is the build_flags. ps you seem to have a mess with the DAUDIOKIT_BOARD=5 and in the sketch you use 6

Ahhh you're right!!! It works now! Thanks again Phil <3