pschatzmann/arduino-audio-tools #88: Buttons problem

mattia714, 2022-01-27:
Hi.
I've uploaded the example sketch streams-pins-audiokit.ino and without  touching any button it continues to repeat "button 3, button 4, volume down, volume up".
Have you any clue? Maybe my audio kit is faulty? How can I troubleshoot this?
TIA

pschatzmann, 2022-01-27:
Strange - For which board is this ?
The logic can be found in AudioTools/AudioActions.h. If the board has capacitive touch buttons you might adjust the touchLimit

pschatzmann, 2022-01-27:
Cool, then just use the latest versions of the libraries and it will work again...

mattia714, 2022-01-27:
Now  all buttons work except button 2..
Maybe some dip-switch configuration? Mine are all off.

mattia714, 2022-01-27:
I have no SD in the SD drive, but I read that dip-switches configuration is related to SD.
Have you some link to dip-switches docs?

pschatzmann, 2022-01-27:
you need to deactivate it with config.sd_active = false;

mattia714, 2022-01-27:
I read the wiki, but I don't understand the second option "At runtime: by setting the sd_active to false in the AudioKitConfig that is passed to the begin method."

pschatzmann, 2022-01-27:
e.g.  
auto cfg = kit.defaultConfig();
  cfg.bits_per_sample = 16;
  cfg.channels = 1;
  cfg.sample_rate = 8000;
  cfg.sd_active = false;
  kit.begin(cfg);

mattia714, 2022-01-27:
Solved.
You are really kind.
Thanks a lot for the support and for your remarkable work on this libs.

closed 2022-02-17 by pschatzmann