pschatzmann/arduino-audio-tools #92: Clicking Noise mp3 from SD

souza107, 2022-02-04:
Getting a clicking noise on an MP3 files in headphones ...using player-sd-audiotkit modified for Lyra 4.3. I tried just lowering volume but noise is still there any suggestions?

pschatzmann, 2022-02-04:
What did you modify ?

pschatzmann, 2022-02-04:
Oh - No, just read the readme of the https://github.com/pschatzmann/arduino-audiokit project

souza107, 2022-02-04:
Thats were I started but needed to use the sdfat library and mp3. I guess I could convert to wave but couldn't figure out how to incorporate sdfat because the Lyra used different pins

pschatzmann, 2022-02-04:
For me this was working with all dip switches in the on position.

souza107, 2022-02-04:
ok so I used this exact file "Arduino audio tools/examples/player-sd-audiotkit" with all the dip switches up and it plays but I am still getting a clicking noise when song plays

pschatzmann, 2022-02-04:
Did you switch the log level to Warning ?

souza107, 2022-02-04:
I did ... Do I need to set buffer size or sample rate?

pschatzmann, 2022-02-04:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Performance

souza107, 2022-02-04:
I tested the cable from computer to speaker and it works fine...So I don't think its the hardware. I really appreciate all the help

souza107, 2022-02-04:
No luck...Any other suggestions?

pschatzmann, 2022-02-04:
What is the sample rate of your files ?
Is it with all files ?
Could could try some other decoders...
Maybe you can share one of your files so that I can double check on my board...

souza107, 2022-02-04:
libhelix -  -> invalid frame size: -1 / max: 1600

souza107, 2022-02-04:
Sure I can share a file, rate its 44100 and I've saved it with 16bit 24 bit and 32bit float How would you like me send file

souza107, 2022-02-04:
https://www.icloud.com/iclouddrive/0c8ZzCYC6zwdhKrJbEFbQFjqg#FireWorks2022

souza107, 2022-02-04:
I have tried a few songs with this is the one I need to work

souza107, 2022-02-05:
Did you get a chance to try that file out with your board? I was tinkering around all night with no success. Tried another mp3 decoder also with no luck.

pschatzmann, 2022-02-06:
I am not sure - I am an old Vinyl guy and I tend to overhear light clicking noises as natural. 
In the example that you sent me I  think that I might hear it in the beginning - but then temporarily it goes away: My gut feeling is that might be a problem of the mp3 decoder.

souza107, 2022-02-06:
Is there an example of playing wav or raw file from Sd?

pschatzmann, 2022-02-06:
You can start with https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-sd_mp3-i2s/streams-sd_mp3-i2s.ino
- for Wav just replace the decoder with WAVDecoder
- for raw just use the file as direct source

souza107, 2022-02-07:
One last question hopefully you can help. I need to access 1 external pin for input. Do you know which one I can use while still accessing the sd and headphone out

souza107, 2022-02-08:
Nvm used pin 0 and it works… thanks again for all help

closed 2022-02-08 by souza107