pschatzmann/arduino-audio-tools #93: mp3 playback for a2dp sink events

daitj, 2022-02-06:
First of all thank you for these nice libraries. 

I would like to play short mp3 audio for a2dp sink events, like disconnected and connected.  I tried to use RAW PCM data, it was too large to fit. So I wondering if I can use mp3 files and decode them using `ESP8266Audio` and use your library's `AudioOutputWithCallback`, but I am bit lost on how to do this correctly. I saw from examples that `AudioOutputWithCallback` is actually used with a2dp source. So is my idea even possible with `AudioOutputWithCallback`  or is this totally wrong implementation of it? 

```
AudioFileSourceSPIFFS *file=nullptr;
AudioGeneratorMP3 *mp3=nullptr;
AudioOutputWithCallback *out=nullptr;
SoundData *notificationsound=nullptr;
```
inside the  `setup()`
```
    mp3 = new AudioGeneratorMP3();
    out = new AudioOutputWithCallback(512,5);
    a2dp_sink.set_on_connection_state_changed(connection_state_changed);
```

```
void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  if(state == ESP_A2D_CONNECTION_STATE_CONNECTING){
    play_sound("/connecting.mp3");
  }else if(state == ESP_A2D_CONNECTION_STATE_DISCONNECTED){
    play_sound("/disconnected.mp3");
  }else if(state == ESP_A2D_CONNECTION_STATE_CONNECTED){
    play_sound("/connected.mp3");
  }else if(state == ESP_A2D_CONNECTION_STATE_DISCONNECTING){
    play_sound("/connected.mp3");
  }
}
```

**I am sure that in the following function, I have to change something to make it work, if it is possible at all, after `mp3->begin(file, out)` and before `SoundData`. Any help will be appreciated.**
```
void play_sound(const char *filename){
  Serial.printf("play sound %s \n", filename);
  file = new AudioFileSourceSPIFFS(filename);
  mp3->begin(file, out);

  Frame* data;
  int32_t len;
  out->read(data, len);

  SoundData *notificationsound = new OneChannelSoundData(data);
  write_sound_data_to_i2s();
  Serial.println("play sound done");
}

```

I found this code in [issue #80 of ESP32-A2DP](https://github.com/pschatzmann/ESP32-A2DP/issues/80): 
```
void write_sound_data_to_i2s() {
 int pos=0;
  int len=512;
  size_t written;
  uint8_t buffer[len];
  int available = 0;
  do {
    available = notificationsound->get2ChannelData(pos, len, buffer);
    i2s_write(I2S_NUM_0, buffer, available, &written, portMAX_DELAY);
    pos+=available;
  } while(available>0);
}
```

pschatzmann, 2022-02-06:
Here is how I would solve the problem:
- I would make sure that all output is managed by the same library using the same output class. I would take my arduino-audio-tools: I2SStream.
- In the A2DP Library I would switch off the output to I2S and replace it with an output via the callback. [This sketch](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/basic-a2dp-spdif) shows how this is done. Just replace SPDIF with I2S
- To decode and play the MP3 you can use the logic demonstrated [here](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-memory_mp3-analog/streams-memory_mp3-analog.ino). I would just replace the copy() with copyAll()

daitj, 2022-02-06:
Thank you for the recommendations. 

I stored mp3 files in a memory and followed your instruction, so here it is what it looks like now. I am bit confused where to have `decoded.begin();` and `copier.copyAll();` but anyway when I run this, I am getting error this error. Is this an OOM error?

```
play sound connecting
abort() was called at PC 0x401ce693 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f7e8:0x3fff5680 0x4008fa65:0x3fff56a0 0x401ce693:0x3fff56c0 0x401ce6da:0x3fff56e0 0x401cddb3:0x3fff5700 0x401ce0ea:0x3fff5720 0x401cde51:0x3fff5740 0x400d1359:0x3fff5760 0x400d152b:0x3fff5780 0x400d22d2:0x3fff57a0 0x400d2ddb:0x3fff57c0 0x400d2ebb:0x3fff5870 0x400d8d17:0x3fff5890 0x400d9146:0x3fff58b0 0x400d88e9:0x3fff58e0 0x401fe2dd:0x3fff5900 0x400d8c4e:0x3fff5920 0x400d8835:0x3fff5950 0x40090b1a:0x3fff5970

Rebooting...
```

```

BluetoothA2DPSink a2dp_sink;
I2SStream out; 

void play_sound(const unsigned char *file, unsigned int file_len){
  MemoryStream mp3(file, file_len);
  EncodedAudioStream decoded(&out, new MP3DecoderHelix());
  StreamCopy copier(decoded, mp3);
  decoded.begin();
  copier.copyAll();
  Serial.println("play sound done");
}

void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  if(state == ESP_A2D_CONNECTION_STATE_CONNECTING){
    Serial.println("play sound connecting");
    play_sound(connecting_mp3, connecting_mp3_len);
  }else if(state == ESP_A2D_CONNECTION_STATE_DISCONNECTED){
    Serial.println("play sound disconnected");
    play_sound(disconnected_mp3, disconnected_mp3_len);
  }else if(state == ESP_A2D_CONNECTION_STATE_CONNECTED){
    Serial.println("play sound connected");
    play_sound(connected_mp3, connected_mp3_len);
  }else if(state == ESP_A2D_CONNECTION_STATE_DISCONNECTING){
    Serial.println("play sound disconnecting");
    play_sound(disconnected_mp3, disconnected_mp3_len);
  }
}

// Write data to I2S Stream in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    out.write(data, length);
}
void setup() {
    Serial.begin(115200);
    // register callback
    a2dp_sink.set_stream_reader(read_data_stream, false);

    //play some note on connection state changed
    a2dp_sink.set_on_connection_state_changed(connection_state_changed);
    
    //esp32 bluetooth audio bug fix L and R are swapped
    a2dp_sink.set_swap_lr_channels(true);

    //start bt
    a2dp_sink.start("TrueWiredStereo");

    auto config = out.defaultConfig(TX_MODE);
    config.sample_rate = a2dp_sink.sample_rate();
    config.channels = 2;
    config.bits_per_sample = 16;
    out.begin(config);

    Serial.println("started bluetooth a2dp sink TrueWiredStereo");

}
void loop() {
}
```

pschatzmann, 2022-02-06:
You will need to decode the stacktrace to figure out where the problem is coming from.

However I would suggest to define the following as global variables
-  MemoryStream mp3_con(connecting_mp3, connecting_mp3_len);
-  MemoryStream mp3_disc(disconnected_mp3, disconnected_mp3_len);
- EncodedAudioStream decoded(&out, new MP3DecoderHelix());
- StreamCopy copier;

and in the play_sound(MemoryStream &mp3) you would call copier.begin(decoded, mp3) instead of the allocation of the objects above.

daitj, 2022-02-06:
It finally somewhat works and this is the final code. Your suggestion to start with simple sketch was helpful indeed. Thanks.

MP3 audio has some clips and cracks, which I can live with. But my main issue now is that the audio from bluetooth is not right, like it is slowed down, similar to when you slow some audio and you hear male voice for females. Could this be because `read_data_stream` callback function adds some miliseconds to write the output to i2s?

  
```
#include <Arduino.h>
#define USE_A2DP
#define USE_HELIX 
#include "BluetoothA2DPSink.h"
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include "connecting.h"
#include "connected.h"
#include "disconnected.h"

BluetoothA2DPSink a2dp_sink;
I2SStream out; 
MemoryStream mp3_connecting(connecting_mp3, connecting_mp3_len);
MemoryStream mp3_disconnected(disconnected_mp3, disconnected_mp3_len);
MemoryStream mp3_connected(connected_mp3, connected_mp3_len);
EncodedAudioStream decoded(&out, new MP3DecoderHelix());
StreamCopy copier;
bool deviceIsConnected = false;

void play_sound(MemoryStream &mp3){
  copier.begin(decoded, mp3);
  copier.copyAll();
  Serial.println("play sound done");
}

void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  if(state == ESP_A2D_CONNECTION_STATE_CONNECTING){
    Serial.println("play sound connecting");
    play_sound(mp3_connecting);
  }else if(state == ESP_A2D_CONNECTION_STATE_DISCONNECTED && deviceIsConnected){
    Serial.println("play sound disconnected");
    play_sound(mp3_disconnected);
    deviceIsConnected = false;
  }else if(state == ESP_A2D_CONNECTION_STATE_CONNECTED){
    Serial.println("play sound connected");
    play_sound(mp3_connected);
    deviceIsConnected = true;
  }
}

// Write data to I2S Stream in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    out.write(data, length);
}

void setup() {
    Serial.begin(115200);
    
    decoded.setNotifyAudioChange(out);

    auto config = out.defaultConfig(TX_MODE);
    config.sample_rate = a2dp_sink.sample_rate();
    config.channels = 2;
    config.bits_per_sample = 16;
    config.pin_ws = 25;
    config.pin_bck = 26;
    config.pin_data = 22;
    out.begin(config);
    Serial.println("out mode configured");
    decoded.begin();
    Serial.println("mp3 decoder begin");

    // register callback
    a2dp_sink.set_stream_reader(read_data_stream, false);

    //play some note on connection state changed
    a2dp_sink.set_on_connection_state_changed(connection_state_changed);
    
    //esp32 bluetooth audio bug fix L and R are swapped
    a2dp_sink.set_swap_lr_channels(true);

    //start bt
    a2dp_sink.start("TrueWiredStereo");
    Serial.println("started bluetooth a2dp sink TrueWiredStereo");

}

void loop() {
}

```

pschatzmann, 2022-02-06:
Add AudioLogger::instance().begin(Serial, AudioLogger::Info); and double check if you find any inconsistent changes of the sample_rate.

In the worst case you could call out.end() and out.begin(config) after the mp3 output to set the sampling rate to the A2DP expected value again...

daitj, 2022-02-06:
All the MP3 had sample rate  of 44100, as `a2dp_sink.sample_rate();` had default value of 44100, that was not the issue.

But when I commented out `decoded.setNotifyAudioChange(out)` bluetooth audio started to work as expected. But MP3 was high pitched and  crackling is still there, this was the log output
```
[I] AudioCopy.h : 74 - buffer_size=1024
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 517 -> 517 -> 517 bytes - in 1 hops
```

pschatzmann, 2022-02-07:
So I guess the difference is the number of channels. A2DP is in stereo and your MP3 are mono ?

daitj, 2022-02-07:
Yes you were correct, it was that. Now this is the code which works as it should. MP3 has crackling noise and everytime new mp3 is played, some remnants of last MP3 is also played.  Like when "Connected" is played after "Connecting" I head "Ing" and then "Connected."

I have uploaded the recording here:
https://binaydevkota.com/share/githubDF2EJNiSEakwvH4qyZLt/test.mp3
The remnants are played during this duration: 2.8-3.7s and 6.7-7.4s, you can also hear MP3 crackling.

```
#include <Arduino.h>
#define USE_A2DP
#define USE_HELIX 
#include "BluetoothA2DPSink.h"
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include "connecting.h"
#include "connected.h"
#include "disconnected.h"

BluetoothA2DPSink a2dp_sink;
I2SStream out; 
MemoryStream mp3_connecting(connecting_mp3, connecting_mp3_len);
MemoryStream mp3_disconnected(disconnected_mp3, disconnected_mp3_len);
MemoryStream mp3_connected(connected_mp3, connected_mp3_len);
EncodedAudioStream decoded(&out, new MP3DecoderHelix());
StreamCopy copier;
auto config = out.defaultConfig(TX_MODE);
bool deviceIsConnected = false;
bool connectingPlayed = false;

void play_sound(MemoryStream &mp3){
  out.end();
  config.channels = 1;
  out.begin(config);
  copier.begin(decoded, mp3);
  copier.copyAll();
  out.end();
  config.channels = 2;
  out.begin(config);
}

void connection_state_changed(esp_a2d_connection_state_t state, void *ptr){
  Serial.printf("a2dp sink state received %s\n", a2dp_sink.to_str(state));
  if(state == ESP_A2D_CONNECTION_STATE_CONNECTING && !connectingPlayed){
    Serial.println("play connecting");
    play_sound(mp3_connecting);
    connectingPlayed = true;
  }else if(state == ESP_A2D_CONNECTION_STATE_DISCONNECTED && deviceIsConnected){
    Serial.println("play disconnected");
    play_sound(mp3_disconnected);
    deviceIsConnected = false;
    ESP.restart();
  }else if(state == ESP_A2D_CONNECTION_STATE_CONNECTED){
    Serial.println("play connected");
    play_sound(mp3_connected);
    deviceIsConnected = true;
  }
}

// Write data to I2S Stream in callback
void read_data_stream(const uint8_t *data, uint32_t length) {
    out.write(data, length);
}

void setup() {
    Serial.begin(115200);
    
    config.sample_rate = a2dp_sink.sample_rate();
    config.channels = 2;
    config.bits_per_sample = 16;
    config.pin_ws = 25;
    config.pin_bck = 26;
    config.pin_data = 22;
    out.begin(config);
    Serial.println("i2s mode configured");
    decoded.begin();
    Serial.println("mp3 decoder started");

    // register callback
    a2dp_sink.set_stream_reader(read_data_stream, false);

    //play some note on connection state changed
    a2dp_sink.set_on_connection_state_changed(connection_state_changed);
    
    //esp32 bluetooth audio bug fix L and R are swapped
    a2dp_sink.set_swap_lr_channels(true);

    //start bt
    a2dp_sink.start("TrueWiredStereo");
    Serial.printf("a2dp sink sample rate %d\n", a2dp_sink.sample_rate());
    Serial.println("started bluetooth a2dp sink TrueWiredStereo");

}

void loop() {
}

```

pschatzmann, 2022-02-07:
Maybe it will help to add a delay after the copyAll() to give the system the chance to play all sound.
For the MP3 crackling I suggest to open a separate issue by indicating how you recorded the mp3...

daitj, 2022-02-07:
Same issue after adding `delay(5000);`. After some tests using some simple example,  I found out that crackling issue is because of `copier.copyAll()`. 
If I do not use `copier.copyAll()` and only use `copier.copy()` in the loop it works. But I do not know how to fix this for three different mp3s.

pschatzmann, 2022-02-07:
Oh, I see - the copyAll() has some build in delay, I will change it so that it can be adjusted via a parameter

pschatzmann, 2022-02-07:
copyAll() has been adjusted and accepts a parameter now: copyAll(0); has no delay. The delay is necessary if you process large files which might trigger the watchdog...

daitj, 2022-02-07:
Thank you very much.

By the way I didn't the answer for my second question (probably you missed it because of the edit)

Are there some examples where you have multiple mp3 playback? I could use that as base to test the remnants playback issue and try to figure out what is happening in my free time?

pschatzmann, 2022-02-07:
Not sure what you mean with multiple playback ? Multiple files ?
Maybe this helps:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player

daitj, 2022-02-07:
We can close this because this issue is fixed. Thanks.

closed 2022-02-07 by daitj