pschatzmann/arduino-audio-tools #95: Problem in every start of a file from sd

serper, 2022-02-14:
I'm testing AudioPlayer to play music from SD to I2S and/or A2DP.

My problem is that although working works, every time a song is going to start it takes quite a long time. While it starts you can hear clicks and once it is running it is a perfect sound.

At each start I get many messages of the type:

libhelix - -> invalid frame size: 0 / max: 1600
libhelix - -> invalid frame size: -480 / max: 1600
libhelix - -> invalid frame size: -1 / max: 1600
libhelix - -> invalid frame size: -1 / max: 1600
libhelix - -> invalid frame size: -223 / max: 1600
libhelix - -> invalid frame size: -1 / max: 1600
libhelix - -> invalid frame size: -1 / max: 1600
libhelix - -> invalid frame size: 0 / max: 1600
libhelix - -> invalid frame size: 0 / max: 1600
libhelix - -> invalid frame size: 0 / max: 1600
libhelix - -> invalid frame size: -465 / max: 1600
libhelix - -> invalid frame size: -1 / max: 1600
libhelix - -> invalid frame size: -1 / max: 1600
libhelix - -> invalid frame size: 0 / max: 1600
libhelix - -> invalid frame size: 0 / max: 1600
libhelix - -> invalid frame size: 0 / max: 1600

I have tried configuring sdfat with dedicated SPI in case it is slow to read the card but it does the same thing.

serper, 2022-02-14:
I have tried with the log at the Error level, the messages do not appear but the same thing happens.

I have seen on a page (http://fileformats.archiveteam.org/wiki/ID3) how to skip the id3 data:

How to skip past an ID3v2 segment
To identify an audio file's format, it is best to skip past any ID3v2 segment at the beginning of the file before looking for a magic signature, and not just assume that ID3 implies MP3. Unfortunately, doing so is not trivial. Here is an attempt to summarize the algorithm:

Let OFFSET = 0.
Read and remember the first 10 bytes of the file.
If bytes 0-2 are not ASCII "ID3", stop. An ID3v2 segment is not present.
Let OFFSET = 10 (for the 10-byte header).
Decode bytes 6-9 as a 32-bit "synchsafe int" (refer to any ID3v2 spec). Let OFFSET = OFFSET + this decoded int.
If the 0x10 bit of byte 5 is set, let OFFSET = OFFSET + 10 (for the footer).
OFFSET is now the file offset of the payload audio data.

pschatzmann, 2022-02-14:
Or even better: extend the logic in the MetaData parser to separate between audio data and metadata. Currently it is only used to extract metadata.

closed 2022-02-19 by serper