pschatzmann/arduino-audio-tools #96: ESP32 and I2S passthru reader/writer problem

thaaraak, 2022-02-14:
I have been playing around with the [streams-i2s-i2s.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-i2s-i2s/streams-i2s-i2s.ino) example. I am using an ESP32 and the pMod I2S2 board and share the MCLK, BCLK and WS lines between the input and output (data input and output are separate of course)

I am feeding the input with a simple sine wave from a signal generator I have noticed some instabilities with the output - particularly on a warm boot. Essentially the output audio is garbage - which only returns to normal after I remove power to both ESP32 and the pMod board. Not sure what is causing this but seems as if input/output isn't synchronized perhaps?

Based on what I have seen with the I2S streams in the ESP-ADF toolkit I decided to change the underlying code to have the I2S input and output streams share the same i2s port. This does mean:
- You have to change the begin code to call i2s_driver_install only once
- You have to pass the tx and rx pin numbers into i2s_set_pin 
- You have to set the i2s_config mode to be master | RX | TX

My suspicion is using the same i2s port, because its the same interrupt removes the synchronization problem. Anyway - I have included my hacked code. 

[i2s.zip](https://github.com/pschatzmann/arduino-audio-tools/files/8063123/i2s.zip)

thaaraak, 2022-02-14:
Forgot to add the calling code...

`  // start I2S in
  Serial.println("starting I2S...");
  auto config_in = in.defaultConfig(RX_MODE);
  config_in.sample_rate = sample_rate;
  config_in.bits_per_sample = 16;
  config_in.i2s_format = I2S_STD_FORMAT;
  config_in.is_master = true;
  config_in.port_no = 0;
  config_in.pin_ws = 18;
  config_in.pin_bck = 5;
  config_in.pin_data = 17;
  config_in.pin_mck = 0;

  config_in.use_apll = true;
   
  in.beginit(config_in, 19, 17, true );

  // start I2S out
  auto config_out = out.defaultConfig(TX_MODE);
  config_out.sample_rate = sample_rate;
  config_out.bits_per_sample = 16;
  config_out.i2s_format = I2S_STD_FORMAT;
  config_out.is_master = true;
  config_out.port_no = 0;
  config_out.pin_ws = 18;
  config_out.pin_bck = 5;
  config_out.pin_data = 19;
  config_out.pin_mck = 0;

  config_out.use_apll = true;

  out.beginit(config_out, 19, 17, false );`

pschatzmann, 2022-02-14:
This code is correct for the setup with 2 ports. 
If you want to use on port it will need to be something like this:

```
Serial.println("starting I2S...");
auto config_in = in.defaultConfig(RXTX_MODE);
config_in.sample_rate = sample_rate;
config_in.bits_per_sample = 16;
config_in.i2s_format = I2S_STD_FORMAT;
config_in.is_master = true;
config_in.port_no = 0;
config_in.pin_ws = 18;
config_in.pin_bck = 5;
config_in.pin_data = 19;
config_in.pin_data_in = 17;
config_in.pin_mck = 0;
```
I have this kind of logic working for the LyraT and AudioKit support with the AudioKitStream. Unfortunately it is not available yet for pure I2S, but I can try to commit a correction tomorrow....

thaaraak, 2022-02-14:
Thank you I'll try your code out.

pschatzmann, 2022-02-14:
I have committed the related changes to I2SConfig.h and I2SESP32.h. 
I hope I did not forget anything...

pschatzmann, 2022-02-14:
ps. here is an [audiokit example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-audiokit/streams-audiokit-audiokit.ino) that uses the same stream to do input and output

thaaraak, 2022-02-14:
Ah - so just use the same I2S Stream instance for input and output to the Copier. Thank you - I will give it a try ASAP

thaaraak, 2022-02-14:
I think you missed a "return"

```
/home/xenir/Arduino/libraries/arduino-audio-tools/src/AudioTools/Resample.h:395:52: warning: no return statement in function returning non-void [-Wreturn-type]
         int available() override { up.available(); }
```

pschatzmann, 2022-02-14:
Thanks...corrected

thaaraak, 2022-02-14:
Worked like a charm!! Although I am still getting a compile error in line 395 of Resample.h

Thank you - off to try the FIR filters now!

pschatzmann, 2022-02-14:
Strange. I just doule checked on github: line 395 is

        int available() override { return up.available(); }

thaaraak, 2022-02-14:
never mind - problem my end - all good now