pschatzmann/arduino-audio-tools #100: Trying to get started. Please help?

GitFunkyHub, 2022-02-18:
Hello,

First of all thank you so much for this library. I've been looking for a decent audio library to use.

I'm having an issue trying to test the 'streams-url_mp3-i2s' example which I can't seem to resolve.

When I try to compile it I get the following error:

c:/users/mhype/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: sketch\URLStreamTest.ino.cpp.o:(.rodata._ZTVN11audio_tools9I2SStreamE[vtable for audio_tools::I2SStream]+0x38): undefined reference to `audio_tools::AudioStream::begin()'
M

I'm not sure what I'm doing wrong as I've not come across this before. It's not finding the reference for some reason.

Please could someone tell me what could be causing this? My experience with Arduino is still relatively basic I would say.

Thank you so much.

Sunny

pschatzmann, 2022-02-18:
Try to upgrade the library to the latest version and let me know if the issue persists

closed 2022-02-18 by pschatzmann

pschatzmann, 2022-02-18:
I have committed some changes yesterday which caused this issue. The related corrections were committed yesterday as well - so you still have the bad version. I was checking all sketches after my corrections and I can confirm that the issue has been resolved.

Just update the code!

GitFunkyHub, 2022-02-18:
Ok great, thank you.