pschatzmann/arduino-audio-tools #101: For url_mp3-i2s example the very end get's cut off

GitFunkyHub, 2022-02-18:
Hello,

I have managed to get the example working however there appears to be a slight issue in that the last second or so of the mp3 appears not to play properly, and gets cut off.

Has anyone else experienced this?

This is straight out of the box example (streams-url_mp3-i2s.ino) using an ESP32 with quite a short mp3 of about 6 seconds, although I tried varying lengths which had the same issue.

Any ideas what could be causing this?

Thank you

Sunny

pschatzmann, 2022-02-18:
In most of the examples I am just constantly streaming data - so this is not notiable.
If you have just a short  mp3 - I suggest to end the decoder to flush out the output: Im my new [simple-tts](https://github.com/pschatzmann/arduino-simple-tts) project I am using this approach  in the [TextToSpeech](https://github.com/pschatzmann/arduino-simple-tts/blob/main/src/TextToSpeech.h) class. Have a look at the say method

GitFunkyHub, 2022-02-18:
Hi, I tried the above by amending the code as follows (to gracefully end everything), but as simple as it seems, it doesn't seem to play anything at all:

```
bool ended = false;

void loop(){

  if (!copier.copy() && !ended)
  {
    copier.end();
    url.end();
    dec.end();
    ended = true;
  }
}
```

GitFunkyHub, 2022-02-18:
Or even with the following:

```
bool ended = false;

void loop(){

  if (!copier.copy() && !ended)
  {
    dec.end();
    ended = true;
  }
}
```

pschatzmann, 2022-02-19:
I just realized that it is better to check if the source still has data:
- [example with copy()](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-memory_mp3_short-i2s)
- [example with copyAll()](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-memory_mp3_short-i2s-2). I prefer this one

GitFunkyHub, 2022-02-19:
Thank you so much for this. I am looking at this now.

GitFunkyHub, 2022-02-19:
Thank you but this seems a bit tricky. I tried url.available() as its an http stream but it didn't work.

Listening carefully to the end of the mp3, it doesn't actually get cut off I think -- the sound just becomes warbled for the last second or so but is very noticeable.

I suspect something is not translating correctly at the very end, and perhaps this is only prevalent for http streams as I haven't tried anything else. I will be using this to play short mp3 files from an http server.

Thank you for your help.

Sunny

GitFunkyHub, 2022-02-20:
I think I can live with that for now and could do a work around of having an mp3 with a second of so silence at the end. I'm trying to do something else which I thought would be easy to do but I'm finding difficult.

I'm trying to move the URLStream url(ssid,password) line to the setup() section but this is proving to be very difficult because the StreamCopy copier(out, url) line depends on it, and I can't move StreamCopy copier(out, url) to setup() because then copier.copy() is out of scope in the loop() section. I know this should be easy to do but not sure. I'm not really from a C++ background, so really trying to get up to speed.

The reason for this is the ESP32 is in deep sleep and when it comes out of deep sleep the first thing it does is to run setup(), and I need it to connect to wifi, after which it plays a short mp3 file from an http server -- but for all this to happen I need the urlStream to connect in the setup() section of the code.

If this is easy to do and I'm just missing something then please let me know. So far I haven't got it to work. Once I have got the hang of the first few things I should be able to work things out for myself of how things work, as I do have a development background.

Thank you very much.

Sunny

GitFunkyHub, 2022-02-20:
This is turning out to be an issue indeed. There seems to be no way of knowing when an mp3 has reached the end which is critical for what I am trying to do. I have tried the various methods but they don't seem to work. I'm not sure why. I suppose this may just be an issue when streaming an mp3 from http, as I haven't tried to do this from memory.

pschatzmann, 2022-02-20:
So far I have only checked with Files and MemoryStrems. 
I will double check with URLStream...

GitFunkyHub, 2022-02-20:
Thanks a lot. I will have a look at the options. I'm just trying to keep the file size down, hence using mp3.

pschatzmann, 2022-02-20:
I suggest that you try to comment out 

//#define USE_URLSTREAM_TASK
in the AudioConfig.h

This setting is to be used for long streaming tasks...

GitFunkyHub, 2022-02-20:
Yes, I can confirm that the urlstream->available() command does now appear to work. Thank you for having a look at that. This has been ever so helpful. The end of the mp3 does still get warbled but I can extend the mp3 by a second or so with silence as a workaround, although it would be good to know the root cause.

Please could I ask one thing: the serial monitor appears to have a lot of cryptic logging going on with characters being displayed constantly going across. Is it possible to switch this off so that I can use Serial.println() statements for my development and debugging?

pschatzmann, 2022-02-20:
https://github.com/pschatzmann/arduino-audio-tools
search for logging

GitFunkyHub, 2022-02-20:
Yes, I did try using:
#define USE_AUDIO_LOGGING false
but I still had these cryptic characters going across the serial monitor non-stop so wasn't sure if there was something else I needed to do. The sketches were just the examples I used from the repository.

pschatzmann, 2022-02-20:
Strange: I was never getting this. Can you give an example ?

GitFunkyHub, 2022-02-20:
Yes, of course. This is one of the examples. I've modified it very slightly to adapt it for my needs.

```
/**
 * @file player-url-i2s.ino
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-url-i2s/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

// set this in AudioConfig.h or here after installing https://github.com/pschatzmann/arduino-libhelix.git
#define USE_HELIX 

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

I2SStream i2s;
MP3DecoderHelix decoder;
URLStream *url;
AudioSourceURL *source;
AudioPlayer *player;

void setup() {

  Serial.begin(115200);
//  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  url = new URLStream("home", "Deco-101");

  // setup output
  auto cfg = i2s.defaultConfig(TX_MODE);
  cfg.pin_ws = 26;
  cfg.pin_bck = 27;
  cfg.pin_data = 25;
  i2s.begin(cfg);

  const char *urls[] = {
  "http://192.168.68.103/TMS/Audio/SystemRestartingEN.mp3"
  };

  source = new AudioSourceURL(*url, urls, "audio/mp3");
  player = new AudioPlayer(*source, i2s, decoder);
  player->begin();

}

void updateVolume() {
  float vol = 0.8;
  player->setVolume(vol);
}

void loop() {
  updateVolume(); // remove comments to activate volume control

  if (url->available())
  {
    player->copy();
  } 
}

```

pschatzmann, 2022-02-20:
Sorry - I meant an example how the strange characters look on the screen so that I get an idea what you talk about.

GitFunkyHub, 2022-02-20:
Ah yes, here we are. It's non-stop but here's a snippet:

20:52:43.988 ->  ⸮@⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮xx⸮⸮⸮⸮⸮⸮⸮B⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮xx⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮xx⸮⸮⸮⸮⸮⸮⸮8xx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮x⸮⸮⸮⸮⸮x⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮x⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮x⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮xx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮p⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮88⸮xx⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮x⸮⸮⸮x⸮⸮⸮⸮8⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮x⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮xx⸮x⸮⸮⸮⸮⸮⸮xx⸮⸮x⸮⸮x⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮x⸮⸮x⸮⸮xx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮x⸮⸮x⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮A⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮A⸮x⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮x⸮⸮⸮⸮⸮x⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮x⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮xx⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮a⸮A⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮xpx⸮⸮xx⸮⸮xx⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮A⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮x⸮⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮@⸮⸮⸮⸮⸮⸮⸮x⸮x⸮⸮⸮⸮⸮⸮⸮⸮⸮

GitFunkyHub, 2022-02-20:
Actually could this be to do with my code? I had to move some assignments into setup() (for the wifi connection and the mp3 url) and the only way for the 'player' to remain in scope in the loop() segment was to define it as a pointer at the very beginning before the setup(). I'm not sure to be honest. Although I'm from a development background, it's the first time I'm dabbling with c++.

pschatzmann, 2022-02-20:
It looks as if something is printing the received audio data to Serial1. I am sure that I don't do anything like this - and if you have deactivated the logger it can definitely not come from my code.
What ESP32 Arduino version are you using - maybe it helps if you downgrade ?

GitFunkyHub, 2022-02-20:
Yes, maybe it's because of the amendments I made in the code as seen above. The latest arduino IDE version 1.8.19 and the latest ESP arduino board version 2.0.2. Okay, I will investigate this. Thank you.

pschatzmann, 2022-02-20:
I suggest that you try with ESP arduino board version 1.0.6 first

GitFunkyHub, 2022-02-20:
Oh I see. Is there a reason because I think version 1.0.6 is quite old isn't it?

pschatzmann, 2022-02-20:
Yes but very stable - all the newer releases tend to have some issues

GitFunkyHub, 2022-02-20:
Ok thanks for letting me know. I think I will do that then.

GitFunkyHub, 2022-02-21:
Yes, that certainly resolved the issue. I will be using version 1.0.6 from now on. I have decided to abandon the http route for streaming as I find it is too hit and miss for my needs. I need it such that when a button is pressed that it connects to wifi within a couple of seconds and streams a short mp3, and even though it works most of the time, there are periods where it can take several seconds to establish a connection and start streaming the mp3. It's very inconsistent. I don't know why this happens but it's a deal breaker for me. I need it to be less than a couple of seconds every time.

To that end, I have decided to go the SD card route which will store the mp3 files. I have tried the audioplayer example for streaming from SD and it works quite well, but please could I ask how again you would determine when an mp3 has come to an end? With URLStream we had the available() method, but when using the audioplayer for SD it's not so obvious. Again, thank you so much for your help. I just need to sort this little bit out, and should be enough to adapt it for my needs.

pschatzmann, 2022-02-21:
Oh, if you have the solution working with URLStream just replace it with a File, there is no need to use the audioplayer

GitFunkyHub, 2022-02-21:
Yes, thank you. I know what you are saying (and the link you have given is the one I was originally using) but I started using the audio player because I wasn't sure how to incorporate the volume control feature and the audio player already had that functionality. Also, I know this sounds trivial, but what would be the correct way to re-start an mp3 from the beginning again, as I needed this functionality as well, and wanted to do it the proper way? Many thanks again

pschatzmann, 2022-02-21:
Volume control is easy: just wrap the i2s: https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control
Restart:
- option 1, close file - open file again
- option 2, move to the beginning with theFile.seek(0);

GitFunkyHub, 2022-02-21:
Ah great, that was really easy to do. Thanks!

One thing that has come out of this, which is very interesting. At the beginning of the post my original issue was the end of the mp3 seemed to get cut. Well, in order to test the re-start of the file, I devised the following code in the loop() segment. What's interesting is that after the short delay the file indeed re-starts, but at the very beginning of the re-started mp3 I can hear the tail end of the mp3 that was cut off previously!

I've had a look at the code very briefly and it seems that the available() method is saying nothing when there is still more data to read. Perhaps something to do with the buffers not filling up to the very end? Not sure, but thought I should mention this.

pschatzmann, 2022-02-21:
This is a feature of the mp3 decoder: It return the data only when the pcm buffer is full. I also just noticed that if the word is too short even a decoder.end is not helping...

GitFunkyHub, 2022-02-21:
Ok, so the root cause is the decoder. Hmmm, I wonder how this could be addressed at the root? Unless there is a way to force it to return data even when the buffer is not full

pschatzmann, 2022-02-21:
So far it was working for me to call end on the decoder - but just today I was stumbeling over the case that when it is too short even this will not work...

GitFunkyHub, 2022-02-21:
Yes good idea - I hadn't thought of that until now you mentioned it. Cheers :)

GitFunkyHub, 2022-02-22:
Hi, I've tried to incorporate the 'streams-sd_mp3-i2s' example into my own program and it's quite straightforward but I keep on getting the following error.

`libhelix - CommonHelix not active
`

Would you know what's causing this? I've installed all the necessary libraries and have the #define USE_HELIX command at the top of my code

pschatzmann, 2022-02-22:
There is some missing begin() in your sketch that opens the decoder

GitFunkyHub, 2022-02-22:
Thanks that solved the issue, although I don't understand why it's not required for your example. The code I used was taken directly from your example but in my code I have to explicitly call decoder.begin to make it work

pschatzmann, 2022-02-22:
Hmm...I would have expected the begin on the EncodedAudioStream...

GitFunkyHub, 2022-02-22:
Yes, that's exactly right and was expecting the same, but I had to put decoder.begin() to make it work, so a bit strange

closed 2022-04-20 by pschatzmann