pschatzmann/arduino-audio-tools #102: How to switch the output correctly?

serper, 2022-02-19:
I am trying to implement the switching between the I2S and A2DP output so that I can use the wifi while the output is I2S (to play internet radio), while when the output is A2DP I play local content (in SD).
The problem I'm running into is that I don't know the correct way to disable one output and enable the other.
While I2SStream has the end method and it works perfectly, A2DPStream does not have this method.
I have tried through the .source() call to the disconnect() method, it disconnects but instantly reconnects. I also call the end() method of the source but it keeps reconnecting.
The thing is that switching from I2S to A2DP the first time works even though I can't stop bluetooth to start Wi-Fi, I can then go back to I2S, but when I go back to A2DP for the second time it crashes and bounces:

`ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x400942f0
file: ".pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp" line 196
func: virtual void BluetoothA2DPSource::start_raw(std::vector<const char*>, music_data_cb_t)
expression: esp_bt_controller_mem_release(ESP_BT_MODE_BLE)

ELF file SHA256: 0000000000000000

Backtrace: 0x40093de0:0x3ffca380 0x400942f3:0x3ffca3a0 0x400e94c7:0x3ffca3c0 0x400e935d:0x3ffca3f0 0x400e9649:0x3ffca420 0x400d657a:0x3ffca460 0x400d7273:0x3ffca4a0 0x400d7322:0x3ffca4c0 0x400d76d1:0x3ffca4f0 0x400e9b1e:0x3ffca510 0x400e9a8a:0x3ffca530 0x400e98aa:0x3ffca550 0x400d86c8:0x3ffca570 0x400eece4:0x3ffca590 0x40095102:0x3ffca5b0
  #0  0x40093de0:0x3ffca380 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x400942f3:0x3ffca3a0 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:721
  #2  0x400e94c7:0x3ffca3c0 in BluetoothA2DPSource::start_raw(std::vector<char const*, std::allocator<char const*> >, int (*)(unsigned char*, int)) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:196 (discriminator 1)
  #3  0x400e935d:0x3ffca3f0 in BluetoothA2DPSource::start(std::vector<char const*, std::allocator<char const*> >, int (*)(Frame*, int)) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:158
  #4  0x400e9649:0x3ffca420 in BluetoothA2DPSource::start(char const*, int (*)(Frame*, int)) at .pio/libdeps/esp32dev/ESP32-A2DP/src/BluetoothA2DPSource.cpp:150 (discriminator 1)
  #5  0x400d657a:0x3ffca460 in audio_tools::A2DPStream::begin(audio_tools::A2DPConfig) at .pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioStreams.h:27
  #6  0x400d7273:0x3ffca4a0 in initA2DP(char*) at .pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioStreams.h:27
  #7  0x400d7322:0x3ffca4c0 in switchStream(int) at .pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioStreams.h:27
  #8  0x400d76d1:0x3ffca4f0 in trigger7() at .pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioStreams.h:27
  #9  0x400e9b1e:0x3ffca510 in EasyNex::callTriggerFunction() at lib/EasyNextionLibrary/src/callTriggers.cpp:56
  #10 0x400e9a8a:0x3ffca530 in EasyNex::readCommand() at lib/EasyNextionLibrary/src/readCustomCommands.cpp:61
  #11 0x400e98aa:0x3ffca550 in EasyNex::NextionListen() at lib/EasyNextionLibrary/src/EasyNextionLibrary.cpp:164
  #12 0x400d86c8:0x3ffca570 in loop() at .pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioStreams.h:27
  #13 0x400eece4:0x3ffca590 in loopTask(void*) at /Users/sergio/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:23
  #14 0x40095102:0x3ffca5b0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...`

pschatzmann, 2022-02-19:
https://github.com/pschatzmann/ESP32-A2DP/wiki/Auto-Reconnect
However there is no way to stop bluetooth and release the memory properly. 
You will need to restart the device!

serper, 2022-02-19:
Yes, I had already seen that, I have in the code out_bt.source().set_auto_reconnect(false); but it keeps reconnecting.

It had already crossed my mind to restart when changing the output, but I tried before doing that to see if I was doing something wrong.

Thank you very much for your help, I will do a reset that will be the easiest.

closed 2022-02-19 by serper