pschatzmann/arduino-audio-tools #107: Need help to resolve ESP32 AudioKit v2.2 2444 (AC101) SD Card examples error.

DevPhone, 2022-02-28:
Hi,
Thank you for sharing the arduino-audio-tools library.
I am testing this with the ESP32 AudioKit with v2.2 2444 - AC101 version.

I am able to run the  player-url_icy-audiokit example properly (by changing the broken online-radio urls with new, working ones)
I am also able to run sd card examples when using the esp-adf and idf.py, so I know its not the hardware.

But I cannot get the SD card to read mp3 files in the player-sd-audiokit example.
I have changed #define AUDIOKIT_BOARD 6 in the AudioKitSettings.h file to my particular board.
It gets stuck at  AudioPlayer.h : 217 - SD.begin failed with cs=13!

Any help in getting it to work would be appreciated. Thank you.

Here is the detailed serial monitor log :

04:35:58.631 -> [I] AudioKit.h : 193 - Channels 2->2
04:35:58.631 -> [I] AudioOutput.h : 635 - void audio_tools::FormatConverterStream::setup()
04:35:58.631 -> [I] AudioOutput.h : 648 - FormatConverterStream: no conversion
04:35:58.664 -> [I] AudioKit.h : 539 - void audio_tools::AudioKitStream::setupActions()
04:35:58.664 -> [W] AudioKit.h : 550 - Headphone detection ignored because of conflict: 5 
04:35:58.664 -> [W] AudioKit.h : 557 - Mode Button ignored because of conflict: 13 
04:35:58.664 -> [I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
04:35:58.698 -> [I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
04:35:58.698 -> [I] AudioActions.h : 58 - pin 5 -> INPUT_PULLUP
04:35:58.698 -> [I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
04:35:58.698 -> [I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
04:35:58.698 -> [I] AudioActions.h : 58 - pin 18 -> INPUT_PULLUP
04:35:58.698 -> [I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
04:35:58.731 -> [I] AudioActions.h : 42 - ActionLogic::add pin: 23 / logic: 0
04:35:58.731 -> [I] AudioActions.h : 58 - pin 23 -> INPUT_PULLUP
04:35:58.731 -> [I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
04:35:58.731 -> [I] AudioActions.h : 42 - ActionLogic::add pin: 19 / logic: 0
04:35:58.731 -> [I] AudioActions.h : 58 - pin 19 -> INPUT_PULLUP
04:35:58.764 -> [I] AudioPlayer.h : 772 - setVolume(0.700000)
04:35:58.764 -> [I] AudioStreams.h : 718 - setVolume: 0.700000
04:35:58.764 -> [E] AudioPlayer.h : 217 - SD.begin failed with cs=13!
04:35:58.764 -> begin() failed
04:35:58.764 -> Do not reformat the SD.
04:35:58.764 -> No card, wrong chip select pin, or wiring error?
04:35:58.764 -> SdError: 0X1,0XFF

pschatzmann, 2022-03-01:
Hmm, I could not reproduce your issues with the player-sd-audiokit example.
My jumper positions are: OFF,ON,ON,OFF,OFF

Are you sure that you have the latest versions of all libraries ?

You could also try with a lower speedMz:
AudioSourceSdFat source(startFilePath, ext, PIN_AUDIO_KIT_SD_CARD_CS, speedMz);

DevPhone, 2022-03-01:
Thank you.
It worked. 

Thanks a ton for the recommended jumper positions. 
Lowering the speedMz was not necessary. 

You hit the nail on the head : ) and its solved now.

My jumper positions were ON,OFF,ON,OFF,OFF and that was the reason I got that SdError.
I received it that way and never changed the factory settings for the dip switches - thus causing the error.
But the esp-adf idf.py SD Card example worked flawlessly with the factory settings.

I would like to learn why the dip-switch settings matter so much for this arduino library, but did not seem to matter to the espressif esp-adf sdcard examples.

Any good link I can refer to? So that I understand the meaning of the dip-switch settings on this board, in detail ?
I know this is a dated board, and newer boards out there do not have these issues, but I'm just curious to learn.

Nice library - I look forward to learning and implementing a lot more projects using this.
Thank you again for your time and prompt reply... and for this great library.

pschatzmann, 2022-03-01:
DIP Switch 3 activates the MOSI pin which is needed by the SPI interface for the SD card.
I could not figure out what IO15 would be needed for...

https://docs.ai-thinker.com/_media/esp32-a1s_v2.3_specification.pdf
https://github.com/pschatzmann/arduino-audiokit/blob/main/src/audio_board/ai_thinker_ac101/board_def.h