pschatzmann/arduino-audio-tools #111: sorry, unimplemented: non-trivial designated initializers not supported. Can't compile any examples

etiennemontenegro, 2022-03-02:
I'm unable to compile any example today, and they we're compiling fine yesterday. I keep getting this error: 

> Compiling .pio\build\esp32dev\src\main.cpp.o
In file included from .pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudio.h:3:0,
                 from .pio\libdeps\esp32dev\audio-tools\src/AudioTools.h:19,
                 from src\main.cpp:2:
.pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudioESP32.h: In member function 'bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)':
.pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudioESP32.h:215:9: sorry, unimplemented: non-trivial designated initializers not supported
         };
         ^
.pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudioESP32.h:215:9: sorry, unimplemented: non-trivial designated initializers not supported
.pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudioESP32.h:215:9: sorry, unimplemented: non-trivial designated initializers not supported
.pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudioESP32.h:215:9: sorry, unimplemented: non-trivial designated initializers not supported
.pio\libdeps\esp32dev\audio-tools\src/AudioAnalog/AnalogAudioESP32.h:215:9: sorry, unimplemented: non-trivial designated initializers not supported
In file included from .pio\libdeps\esp32dev\audio-tools\src/AudioLibs/AudioKit.h:3:0,
                 from src\main.cpp:3:
.pio\libdeps\esp32dev\audiokit\src/AudioKitHAL.h: In member function 'i2s_config_t AudioKitConfig::i2sConfig()':
.pio\libdeps\esp32dev\audiokit\src/AudioKitHAL.h:117:34: warning: narrowing conversion of 'AudioKitConfig::sampleRate()' from 'uint32_t {aka unsigned int}' to 'int' inside { } [-Wnarrowing]
         .sample_rate = sampleRate(),
                                  ^
*** [.pio\build\esp32dev\src\main.cpp.o] Error 1

When looking at commit f8a1e0e655000853373d9c8fa5b43e08af71ba6a it looks like the comment left on line 208 is causing this error . When I uncomment it the examples compile fine.

pschatzmann, 2022-03-02:
Sorry about this. 
For some strange reasons I did not get any errors in the Arduino IDE...
I added the commented out line again...

etiennemontenegro, 2022-03-02:
Thanks for you great work!

closed 2022-03-02 by etiennemontenegro