pschatzmann/arduino-audio-tools #114: Streams-generator-a2dp example does no more connect to speaker

SensorsIot, 2022-03-04:
Yesterday I had this example working. Unfortunately, today it only connects and disconnects (no sound).
 
[I] SoundGenerator.h : 200 - setFrequency: 494
[I] SoundGenerator.h : 201 - active: false
[I] SoundGenerator.h : 168 - SineWaveGenerator::begin(channels=2, sample_rate=44100)
[I] SoundGenerator.h : 162 - void audio_tools::SineWaveGenerator<T>::begin() [with T = short int]
[I] AudioTypes.h : 38 - sample_rate: 44100
[I] AudioTypes.h : 39 - channels: 2
[I] AudioTypes.h : 40 - bits_per_sample: 16
[I] AudioA2DP.h : 134 - Connecting to MI Portable Bluetooth Speaker
[I] AudioA2DP.h : 141 - Starting a2dp_source...
[W] AudioA2DP.h : 297 - ==> state: Connecting
[W] AudioA2DP.h : 297 - ==> state: Disconnected
[W] AudioA2DP.h : 297 - ==> state: Connecting
[W] AudioA2DP.h : 297 - ==> state: Disconnected
[W] AudioA2DP.h : 297 - ==> state: Connecting
[W] AudioA2DP.h : 297 - ==> state: Disconnected
[W] AudioA2DP.h : 297 - ==> state: Connecting
[W] AudioA2DP.h : 297 - ==> state: Disconnected

I tried also earphones with the same effect.

pschatzmann, 2022-03-05:
I have updated the auto-tools library and disabled the automatic reconnect which might be the source of trouble. If you have any doubts you can try to set the log level to debug in the Arduino IDE and you will see the real BT address to which the ESP tries to connect to.  If it is still not working it might help to have this detailed log...

pschatzmann, 2022-04-20:
I did not get any feed back, so I assume that the correction was working...

closed 2022-04-20 by pschatzmann