pschatzmann/arduino-audio-tools #118: Feature request: ESPHome integration

timotoots, 2022-03-06:
Hello Phil. First of all thanks for the great initiative for this library. Seems perfect for all kinds of crazy stuff.

Would this library be feasible to use as a ESPHome custom component? That project is in big need for audio player component and it would get a lot of use as home music players etc.

Some discussion in here:
https://github.com/esphome/feature-requests/issues/599

pschatzmann, 2022-03-06:
I think you can use my different libraries together with ESPHome to build some cool stuff. E.g. with my library [arduino-simple-tts](https://github.com/pschatzmann/arduino-simple-tts) you could play some sound recordings based on keywords...

jesserockz, 2022-03-08:
We would love to pull something like this into ESPHome.

The main issue I see though is that it has a reliance on Arduino (and it's provided libraries) which raises flags for us as we are trying to move ESP32 fully to ESP-IDF.

pschatzmann, 2022-03-08:
Well on my side it was a conscious decision to make this library dependent on Arduino because of the Arduino Streams, so you can deal with (audio) data via Serial, UDP, TCP, Files, Bluetooth...with just one interface. To do this in ESP-IDF would be a big nightmare...

timotoots, 2022-03-08:
@jesserockz If the way forward with ESPHome is with ESP-IDF then I guess it could be possible to include ESP-ADF as well? It provides a lot of audio streams functionality as well.

pschatzmann, 2022-03-08:
I agree - that would be the logical choice. It's just not as easy to use ...

pschatzmann, 2022-09-17:
Moved to discussion - ideas