pschatzmann/arduino-audio-tools #125: i2c on high bitrates

podaen, 2022-03-10:
Hi,

You know I had some problems before by using the spi library simontainusly with audiotools. We solved the issue, but I couldn't use the spi when I was playing a song, but I could work arround that. 

Now I having issues by adding a i2c pcf8574 expander board for a keypad that is using the wire.h library. When I play songs with a low bitrate the songs play well, but when I have songs with a bitrate of 320kbps than the music stops for a second and continue. I have done everything I could in my opinion on the side of the wire library.

> 	Wire.setTimeOut(5);
> 	Wire.begin(16, 17, 400000);

I need to read that keypad when the music is playing... Is there something more I could do on the side of the autotools?

pschatzmann, 2022-03-10:
You could try to use a separate FreeRtos Task to make sure that delays in the loop are not impacting the audio output.
E.g. move the copy into a separate task.

podaen, 2022-03-10:
I have done that but it doesn't help. Same issue as before. 

Something I notest before was when I cut the power of the i2c keypad module the music stops. And when I plug it back in the music continues.

podaen, 2022-03-10:
I think the problem is the old version of wire library. The last update is since 2017.

pschatzmann, 2022-03-10:
The latest version of the ESP32 Arduino library is 2.0.2. 
Did you try with that ?

podaen, 2022-03-10:
I still have the v1.0.6. I am checking the https://github.com/espressif/arduino-esp32/commits/master/libraries/Wire/src but I can't find any major changes that could lead to this issue yet.

I am sure it is not a power issue, I double checked and I have 2.5A power supply on the micro usb of the esp. It all started when I have added the I2C module.

podaen, 2022-03-10:
Maybe this is

https://github.com/espressif/arduino-esp32/commit/deaf339bde33e5ed3048d70d5b3f8bb1ff0676d4

What do you think?

podaen, 2022-03-11:
I tried, but it diden't work. And out of what I can see it is not a timing problem. What I think is that there can not be an other transaction be established when the audiotools is buzzy. It confirms whit what I find out with the spi transaction too.

pschatzmann, 2022-03-11:
A am afraid that I did never use my framework together with a I2C device.  

- Have you already tried by omitting the frequency in the begin ?
- You could also try to disable the HAL locks CONFIG_DISABLE_HAL_LOCKS=y in tools/sdk/include/config/sdkconfig.h

podaen, 2022-03-11:
> * Have you already tried by omitting the frequency in the begin ?

Yes, but no result.

> * You could also try to disable the HAL locks CONFIG_DISABLE_HAL_LOCKS=y in tools/sdk/include/config/sdkconfig.h

Not yet. I will do.

podaen, 2022-03-11:
I can not edit the sdkconfig.h in the Arduino environment.

podaen, 2022-03-11:
Diden't know that. I am still stuck on the earlier version of Arduino and not planning to change that in a short amount of time. I will continue with the direct interface to the esp for now.

podaen, 2022-04-19:
I had issues with esp32 and a long wire to connect a keypad. Now I m going to implement the I2C keypad again and do what you told me when I was testing with 1.0.6 and couldn't, now I'm installing now 2.0.2. 

I have updated to 2.0.2 but I still can't edit the sdkconfig.h in de arduino environment...

podaen, 2022-04-19:
There is a comment on top of the document...

 * Automatically generated file. DO NOT EDIT.

The only place I can find the CONFIG_DISABLE_HAL_LOCKS is in the wire.h as a condition. So I will manualy added to my shetch before Wire.h. It works... but do I have to add a task like this now, a side of my task of the audioplayer?

```
void Task4Keypad(void* pvParameters) {
	Wire.begin();
	keypad.begin(makeKeymap(keys));
	keypad.addEventListener(keypadEvent);                                      // Add an event listener.
	keypad.setHoldTime(50);  //500                                                 // Default is 1000mS
	for (;;) {//infinit loop
		CheckKeypadCode();
		delay(50);
	}
}
```
Because I see it is a SemaphoreHandle_t...

podaen, 2022-04-20:
Wait, I found some issues in my shetch of the keypad and wiring. I will comeback when I have cleared it out.

podaen, 2022-04-20:
Is The latest version of the ESP32 Arduino library is 2.0.2 now supported?

pschatzmann, 2022-04-20:
The only thing which is not working is the Analog Output. 
This is fixed in 2.0.3...

podaen, 2022-04-20:
I m testing now with only the url player and it seems to work in the first look. but I get al lot of special characters in my serial monitor when I start

```
	auto cfg = i2s.defaultConfig(TX_MODE);
	cfg.i2s_format = I2S_LEFT_JUSTIFIED_FORMAT;
	i2s.begin(cfg);
```

pschatzmann, 2022-04-20:
Then it's time to upgrade my library. 
This was corrected quite some time ago...

podaen, 2022-04-20:
Ok, I will do.

podaen, 2022-04-20:
Hellix to?

pschatzmann, 2022-04-20:
I am note sure, but I would say yes to be on the save side...

podaen, 2022-04-20:
I did it and it is working now, exept the https stream... I m still investigating. The unknown characters are gone.

After updating I get an error

`CodecMP3LAME.h: 4:10: fatal error: MP3EncoderLAME.h: No such file or directory`

pschatzmann, 2022-04-20:
This should only happen if you have a #define USE_LAME or #define USE_DECODERS somewhere...

podaen, 2022-04-20:
yes I did define use_lame because my mp3 player plays a lot of files with the lame codec. But I m programming only on the url player so I supose that is used only with the sd player...

pschatzmann, 2022-04-20:
So is liblame not installed on your machine?

podaen, 2022-04-20:
Hmmm, that's new. It for encoding I see. On the mp3 player I decode I think. Not?

More https stream fails than before, not sure what changed...

pschatzmann, 2022-04-20:
Right, liblame is for encoding.
No changes related to https from my side...

podaen, 2022-04-20:
Than I don't needed. Thanks

podaen, 2022-04-20:
Did you changed parameter timeout for ssl connection or wifi clientsecure?

void WiFiClientSecure::setHandshakeTimeout(unsigned long handshake_timeout)

Because it runs over it much faster over it than 12sec as default.

In the sslclient.cpp start_ssl_client() I changed this     

 //if ((millis() - handshake_start_time) > ssl_client->handshake_timeout) {
        if ((millis() - handshake_start_time) > 12000) {

and now my https streams are working again. Could you add this parameter to audio tools?

pschatzmann, 2022-04-20:
I am afraid, I don't quite understand. 
Are you proposing to increase the handshake_timeout ?

podaen, 2022-04-20:
yes, you have to call back WiFiClientSecure::setHandshakeTimeout(unsigned long handshake_timeout) and set it to 12000ms that was set as default, because it is somewhere changed to a much shorter time.

pschatzmann, 2022-04-20:
OK, I will add a parameter to the  AudioConfig.h

podaen, 2022-04-20:
I see a change of that parameter in here

URLStream.h line 180

pschatzmann, 2022-04-20:
I planned to add it right after line 180.
ps. https://github.com/espressif/arduino-esp32/issues/6077

podaen, 2022-04-20:
Don't know how to solve this. As you know I have made my own classe to use url-player and sd-player. 

`MultiAudioPlayer(AudioSource& SDsource, AudioSource& URLsource, AudioPrint& output, AudioDecoder& decoder) `

I get "No instand of constructor"  in this below.

`MultiAudioPlayer player(SDsource, Urlsource, volume, decoder);`

I don't get it... SDsource is a AudioSource

`AudioSourceSdFat SDsource(startFilePath, ext, spiconfig);`

I think it has something to do with this change

```
//VolumeOutput volume(i2s);
VolumeStream volume(i2s);
```

podaen, 2022-04-21:
I deleted this->p_final_print = &output; and I changed AudioPrint& output too AudioStream& output. Now it seems to work but I diden't do the audio check yet.

pschatzmann, 2022-04-21:
I just committed the change for the URL_HANDSHAKE_TIMEOUT.
Can you help to test it ?

podaen, 2022-04-21:
I do. 

I just discoverd that the reading of the keys is to slow and that is why it chunks. I see in my log

> hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

and when I do

> 	Serial.println("clock");
> 	Serial.println(Wire.getClock());

after wire.begin I get 

> clock
> 100000

Can I do something about the speed of the esp32 doit devkit v1?

podaen, 2022-04-21:
I tried #define  CONFIG_DISABLE_HAL_LOCKS y, but it still chunks

podaen, 2022-04-21:
Hmmm, there are http links that chunck too, but more offend on https

podaen, 2022-04-21:
No, there is a something not right. The key acting strange when I press them. Maybe I have to run the keys on core 0...

podaen, 2022-04-21:
Flash frequentie 80MHz if that is what you mean

pschatzmann, 2022-04-21:
No, -> Tools -> CPU Frequency

podaen, 2022-04-21:
Tools? Can find it in visual micro. I can try to increase it and feels of it gets hot...

podaen, 2022-04-21:
I did set it on setCpuFrequencyMhz(160); but it seems I am lowing the speed now.

> [��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> VMDPR�^  4037][D][esp32-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz

podaen, 2022-04-21:
It seem beter now. I deleted some delays, I don't check the keypad keys in the loop anymore with the url player, only whit the event call back. 

Not with every https link, then I get this

> W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpLineReader.h : 27 - HttpLineReader readlnInternal->no Data
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> [E] URLStream.h : 259 - Error code recieved ... stop waiting for reply
> https://25433.live.streamtheworld.com:443/TOP_RADIO.mp3

But the most https links works now. But my keypad event is not working. Still under investigation.

pschatzmann, 2022-04-21:
If you check this url you get a forbidden error message. So it is ok that it is not working...

podaen, 2022-04-21:
First, to get the event working of the keypad without delaying the audioplayer this must be in the loop

`char key = keypad.getKey();//This needs to stay in the loop for the event`

> If you check this url you get a forbidden error message. So it is ok that it is not working...

Can be, sometimes the link works and sometimes it don't. But It doesn't crash. It is the only link that have this.

podaen, 2022-04-21:
I called to early, I have a loop of crashes if I change the radiostation manualy

> Opening port
> Port open
> [��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> VMDPR_[  4047][D][Hall_Alarm_Music_V2.ino:589] setup(): Debug
> [  4047][I][Hall_Alarm_Music_V2.ino:590] setup(): Info
> [  4047][W][Hall_Alarm_Music_V2.ino:591] setup(): Warning
> [  4051][E][Hall_Alarm_Music_V2.ino:592] setup(): Error
> clock
> 100000
> load local config
> write eeprom RadioActived: 0
> Read Eprom RadioActived: 0
> write eeprom ProgramSelector: 2
> Read Eprom ProgramSelector: 2
> End Setup / Start Task
> xPortGetFreeHeapSize: 188912
> 
> WiFi try to reconnect
> [  5155][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  5391][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
> ....[  7652][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  7865][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [  7865][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.1.198, MASK: 255.255.255.0, GW: 192.168.1.1
> .WiFi connected
> ESPNow Init Success
> Pair success
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> player stoped
> Read Eprom UrlSelector: 11
> E (15027) ESPNOW: Peer channel is not equal to the home channel, send fail!
> Send Status: Invalid Argument
> player started
> end setup loop 1
> 
> Brownout detector was triggered
> 
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1184
> load:0x40078000,len:12804
> ho 0 tail 12 room 4
> load:0x40080400,len:3032
> entry 0x400805e4
> [��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> VMDPR_[  4046][D][Hall_Alarm_Music_V2.ino:589] setup(): Debug
> [  4046][I][Hall_Alarm_Music_V2.ino:590] setup(): Info
> [  4046][W][Hall_Alarm_Music_V2.ino:591] setup(): Warning
> [  4050][E][Hall_Alarm_Music_V2.ino:592] setup(): Error
> clock
> 100000
> load local config
> write eeprom RadioActived: 0
> Read Eprom RadioActived: 0
> write eeprom ProgramSelector: 2
> Read Eprom ProgramSelector: 2
> End Setup / Start Task
> xPortGetFreeHeapSize: 188912
> 
> WiFi try to reconnect
> [  5154][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  5423][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
> ....[  7711][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  7925][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[... 207 lines cut ...]
> load local config
> write eeprom RadioActived: 0
> Read Eprom RadioActived: 0
> write eeprom ProgramSelector: 2
> Read Eprom ProgramSelector: 2
> End Setup / Start Task
> xPortGetFreeHeapSize: 188912
> 
> WiFi try to reconnect
> [  5154][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  5391][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
> ....[  7615][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  7832][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [  7832][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.1.198, MASK: 255.255.255.0, GW: 192.168.1.1
> .WiFi connected
> ESPNow Init Success
> Pair success
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> player stoped
> Read Eprom UrlSelector: 0
> E (15005) ESPNOW: Peer channel is not equal to the home channel, send fail!
> Send Status: Invalid Argument
> player started
> end setup loop 1

podaen, 2022-04-21:
It reboots... It is a burnout, maybe because of my long cable. and than it restart on the same url, because I override the selected url again. I save now to the eprom the last url, we will see...

podaen, 2022-04-21:
it still reboots two times and than it runs good.

podaen, 2022-04-21:
after a time it stops, or maybe it esp now that keeps pending.

podaen, 2022-04-21:
> Opening port
> Port open
> [��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> VMDPR_[  4046][D][Hall_Alarm_Music_V2.ino:589] setup(): Debug
> [  4046][I][Hall_Alarm_Music_V2.ino:590] setup(): Info
> [  4046][W][Hall_Alarm_Music_V2.ino:591] setup(): Warning
> [  4050][E][Hall_Alarm_Music_V2.ino:592] setup(): Error
> clock
> 100000
> load local config
> write eeprom RadioActived: 0
> Read Eprom RadioActived: 0
> write eeprom ProgramSelector: 2
> Read Eprom ProgramSelector: 2
> End Setup / Start Task
> xPortGetFreeHeapSize: 188912
> 
> WiFi try to reconnect
> [  5154][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  5373][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
> ....[  7662][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  7798][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [  7798][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.1.198, MASK: 255.255.255.0, GW: 192.168.1.1
> .WiFi connected
> ESPNow Init Success
> Pair success
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> player stoped
> Read Eprom UrlSelector: 0
> E (14986) ESPNOW: Peer channel is not equal to the home channel, send fail!
> Send Status: Invalid Argument
> player started
> end setup loop 1
> 
> Brownout detector was triggered
> 
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1184
> load:0x40078000,len:12804
> ho 0 tail 12 room 4
> load:0x40080400,len:3032
> entry 0x400805e4
> [��mum����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
> VMDPR_[  4047][D][Hall_Alarm_Music_V2.ino:589] setup(): Debug
> [  4047][I][Hall_Alarm_Music_V2.ino:590] setup(): Info
> [  4047][W][Hall_Alarm_Music_V2.ino:591] setup(): Warning
> [  4051][E][Hall_Alarm_Music_V2.ino:592] setup(): Error
> clock
> 100000
> load local config
> write eeprom RadioActived: 0
> Read Eprom RadioActived: 0
> write eeprom ProgramSelector: 2
> Read Eprom ProgramSelector: 2
> End Setup / Start Task
> xPortGetFreeHeapSize: 188912
> 
> WiFi try to reconnect
> [  5155][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  5324][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
> ....[  7592][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> .[  8022][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [  8022][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.1.198, MASK: 255.255.255.0, GW: 192.168.1.1
> .WiFi connected
[... 320 lines cut ...]
> load local config
> write eeprom RadioActived: 0
> Read Eprom RadioActived: 0
> write eeprom ProgramSelector: 2
> Read Eprom ProgramSelector: 2
> End Setup / Start Task
> xPortGetFreeHeapSize: 189052
> 
> WiFi try to reconnect
> [  5155][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [  5316][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
> .....[  7868][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
> [  7974][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
> [  7975][D][WiFiGeneric.cpp:880] _eventCallback(): STA IP: 192.168.1.198, MASK: 255.255.255.0, GW: 192.168.1.1
> .WiFi connected
> ESPNow Init Success
> Pair success
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> player stoped
> Read Eprom UrlSelector: 3
> E (15427) ESPNOW: Peer channel is not equal to the home channel, send fail!
> Send Status: Invalid Argument
> player started
> end setup loop 1

podaen, 2022-04-21:
No, I think it is still the I2C keyboard that causing this.

podaen, 2022-04-21:
This is why it reboots all the time when I push a button to change the station

> [417013][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
> [417013][W][WiFiGeneric.cpp:852] _eventCallback(): Reason: 8 - ASSOC_LEAVE
> [417016][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 3 - STA_STOP

podaen, 2022-04-21:
It is sure now, the burnouts comes for sure when I call player.copy() on the first boot. I have that since I have updated the Arduino espressif to 2.0.2.

pschatzmann, 2022-04-21:
I guess you mean Brownout ? It seems that you have problems with your power supply...

podaen, 2022-04-21:
Yeah brownout. Its getting much to follow. I will try to connect directly to the power supply and see what it does.

podaen, 2022-04-21:
No reboot anymore, but when I press a key the music stops every second and start back. But no action is couple to the pressed key.

podaen, 2022-04-21:
Running on core 0 won't help. There is something wrong in player.copy(). This is a log from only the mp3 player without kep

> Opening port
> Port open
> VMDPR_here
> Read Eprom UrlSelector: 4
> write eeprom UrlSelector: 4
> in task
> .....
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> name: JOE_80S.mp3
> icyDescription: 
> icyGenre: Oldies
> end setup
> xPortGetFreeHeapSize: 105828
> 
> xPortGetFreeHeapSize: 105788
> 
> xPortGetFreeHeapSize: 104164
> 
> xPortGetFreeHeapSize: 104164
> 
> xPortGetFreeHeapSize: 106192
> 
> xPortGetFreeHeapSize: 104496
> 
> xPortGetFreeHeapSize: 104496
> 
> xPortGetFreeHeapSize: 106120
> 
> xPortGetFreeHeapSize: 104432
> 
> xPortGetFreeHeapSize: 104432
> 
> xPortGetFreeHeapSize: 104432
> 
> xPortGetFreeHeapSize: 104432
> 
> xPortGetFreeHeapSize: 104440
> 
> xPortGetFreeHeapSize: 104440
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 106124
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> xPortGetFreeHeapSize: 104428
> 
> 
> Brownout detector was triggered
> 
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0030,len:1184
> load:0x40078000,len:12804
> ho 0 tail 12 room 4
> load:0x40080400,len:3032
> entry 0x400805e4
> VMDPR_here
> Read Eprom UrlSelector: 4
> write eeprom UrlSelector: 4
> in task
> .....
> [W] HttpHeader.h : 231 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> name: JOE_80S.mp3
> icyDescription: 
> icyGenre: Oldies
> end setup
> xPortGetFreeHeapSize: 106144

The first time it runs and brownout, the second time it runs. I measured the voltage drop, and there is as much on the second time as on the first.

[18 more replies]

closed 2022-04-22 by podaen