pschatzmann/arduino-audio-tools #126: No sound on Audio Kit 2.2 (AC101 Codec)

DatanoiseTV, 2022-03-10:
I have a Audio Kit 2.2 Board and the sound on the headphone jack stays slient.
The Jumper block is set to 1,0,1,0,0.

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
[W] AudioKit.h : 43 - sd_active = false
Warning: reset succeed
[W] AudioKit.h : 542 - Deactivating SPI because SD is not active
.....
[W] HttpHeader.h : 230 - Waiting for data...
[W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
[W] MetaDataICY.h : 265 - callback not defined
[W] AudioKit.h : 43 - sd_active = false
E (9689) I2S: register I2S object to platform failed
Error:   i2s_driver_install
[I] I2SConfig.h : 85 - rx/tx mode: TX
[I] I2SConfig.h : 86 - port_no: 0
[I] I2SConfig.h : 87 - is_master: Master
[I] I2SConfig.h : 88 - sample rate: 44100
[I] I2SConfig.h : 89 - bits per sample: 16
[I] I2SConfig.h : 90 - number of channels: 2
[I] I2SConfig.h : 91 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 93 - I2S_AUTO_CLEAR=0
[I] I2SConfig.h : 103 - pin_bck: 14
[I] I2SConfig.h : 104 - pin_ws: 15
[I] I2SConfig.h : 105 - pin_data: 0
[W] AudioKit.h : 43 - sd_active = false
Warning: reset succeed
[I] AudioKit.h : 193 - Channels 2->2
[I] AudioOutput.h : 635 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 648 - FormatConverterStream: no conversion
[I] AudioKit.h : 539 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 542 - Deactivating SPI because SD is not active
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 13 / logic: 0
[I] AudioActions.h : 58 - pin 13 -> INPUT_PULLUP
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
[I] AudioActions.h : 61 - pin 39 -> INPUT
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 58 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 58 - pin 5 -> INPUT_PULLUP
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 23 / logic: 0
[I] AudioActions.h : 58 - pin 23 -> INPUT_PULLUP
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 19 / logic: 0
[I] AudioActions.h : 58 - pin 19 -> INPUT_PULLUP
[I] AudioPlayer.h : 772 - setVolume(0.700000)
[I] AudioStreams.h : 720 - setVolume: 0.700000
[I] MetaDataID3.h : 545 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioPlayer.h : 429 - selectStream: 0/4 -> https://ice2.somafm.com/live-128-mp3
[I] URLStream.h : 69 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://ice2.somafm.com/live-128-mp3
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://ice2.somafm.com/live-128-mp3
[I] Url.h : 110 - host->ice2.somafm.com
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/live-128-mp3
[I] Url.h : 113 - port->443
.....
[... 154 lines cut ...]
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioKit.h : 342 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 324 - incrementVolume: -2 -> 40
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] MetaDataICY.h : 103 - Data ended
[I] MetaDataICY.h : 114 - metaDataLen: 0
[I] MetaDataICY.h : 125 - Data found
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioKit.h : 342 - static void audio_tools::AudioKitStream::actionVolumeDown(bool, int, void*)
[I] AudioKit.h : 324 - incrementVolume: -2 -> 38
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
```

pschatzmann, 2022-03-11:
Strange, I could not reproduce your problem. I have sound both on the speaker and the earphones.
I don’t think that this is related to the jumper settings, but mine are off,on,on,off,off

DatanoiseTV, 2022-03-11:
@pschatzmann Indeed a bit strange. I am Version 2.0.1 of the ESP32 Arduino BSP and my board is new.

pschatzmann, 2022-03-11:
I currently working with 2.0.2 and have tested 1.0.6.
What is the revision number after the V2.2 on the board ?

olup, 2022-04-08:
I also have an ac101 2.2 board (bought way back) and don't have sound even with a working sketch. @DatanoiseTV did you ever find a solution to your problem ? @pschatzmann has been very kind in trying to help me in the discussion page (https://github.com/pschatzmann/arduino-audio-tools/discussions/160) but right now nothing could do. We might also just have faulty boards ?

DatanoiseTV, 2022-04-08:
> I also have an ac101 2.2 board (bought way back) and don't have sound even with a working sketch. @DatanoiseTV did you ever find a solution to your problem ? @pschatzmann has been very kind in [...]

I have used my board before with the esp-idf and the codec was working.

pschatzmann, 2022-04-08:
You must define your board and the default settings in the libraries/arduino-audiokit/src/AudioKitSettings.h file. Please note that the functionlity will not work properly if this is done in the Sketch:
See https://github.com/pschatzmann/arduino-audio-tools/discussions/160

DatanoiseTV, 2022-04-09:
@pschatzmann Maybe this could be solved in a nice way by providing a build flag to a custom BSP with a boards.ini -DBOARD=xyz and then having the corresponding macros in the header files? Then the users could select the board via the menu in the Arduino IDE.

pschatzmann, 2022-04-09:
There are also no existing boards to choose from. To my knowledge the boards.ini is not project specific and changing the ESP32-Arduino project is outside of my scope. 

By the way - if you use PlatformIO you can set this definition properly in the platformio.ini

closed 2022-04-13 by pschatzmann