pschatzmann/arduino-audio-tools #128: ICY-Stream with redirection

ulegan, 2022-03-13:
I'm trying to connect to german radiostation SWR3.
They publish the url: https://liveradio.swr.de/sw282p3/swr3/play.mp3 (http is also fine).
But this is not the stream, it redirects twice before getting to the stream.
Calling the url on a linux machine with wget gives this result:
`uwe@iot-server:~$ wget https://liveradio.swr.de/sw282p3/swr3/play.mp3`
`--2022-03-13 17:38:14--  https://liveradio.swr.de/sw282p3/swr3/play.mp3`
`Resolving liveradio.swr.de (liveradio.swr.de)... 2600:1901:0:b6ff::, 34.149.96.185`
`Connecting to liveradio.swr.de (liveradio.swr.de)|2600:1901:0:b6ff::|:443... connected.`
`HTTP request sent, awaiting response... 302 Found`
`Location: https://dispatcher.rndfnk.com/swr/swr3/live/mp3/128/stream.mp3?aggregator=web [following]`
`--2022-03-13 17:38:14--  https://dispatcher.rndfnk.com/swr/swr3/live/mp3/128/stream.mp3?aggregator=web`
`Resolving dispatcher.rndfnk.com (dispatcher.rndfnk.com)... 2a05:d014:f28:3800:1f7f:8e00:268a:8d2, 2a05:d014:f28:3802:f473:8500:7e25:6576, 3.67.45.155, ...`
`Connecting to dispatcher.rndfnk.com (dispatcher.rndfnk.com)|2a05:d014:f28:3800:1f7f:8e00:268a:8d2|:443... connected.`
`HTTP request sent, awaiting response... 302 Found`
`Location: https://d141.rndfnk.com/ard/swr/swr3/live/mp3/128/stream.mp3?aggregator=web&sid=26L4Nta1A5BCvDVRNDb7IFOOMir&token=UDNR7e-P17fDMy0UZtEj2n-0uSKfX_9AZha9dttDsnA&tvf=4dpmfsMR3BZkMTQxLnJuZGZuay5jb20 [following]`
`--2022-03-13 17:38:15--  https://d141.rndfnk.com/ard/swr/swr3/live/mp3/128/stream.mp3?aggregator=web&sid=26L4Nta1A5BCvDVRNDb7IFOOMir&token=UDNR7e-P17fDMy0UZtEj2n-0uSKfX_9AZha9dttDsnA&tvf=4dpmfsMR3BZkMTQxLnJuZGZuay5jb20`
`Resolving d141.rndfnk.com (d141.rndfnk.com)... 2a11:7400:d1:ca::141, 195.243.98.141`
`Connecting to d141.rndfnk.com (d141.rndfnk.com)|2a11:7400:d1:ca::141|:443... connected.`
`HTTP request sent, awaiting response... 200 OK`
`Length: unspecified [audio/mpeg]`
`Saving to: ‘play.mp3.1’`

Calling http://liveradio.swr.de/sw282p3/swr3/play.mp3
in the player-url-i2s example crashes my ESP32, your swiss url's are fine.
Is it possible to support such redirections?

pschatzmann, 2022-03-18:
I have implemented the http protocol on top of the Arduino Client.h. I tried to add some logic to handle redirects, but it seems that it does not covert these complicated cases.

I will double check if I can replace my current implementation with a recursive one...

pschatzmann, 2022-03-18:
I just committed a correction that handles multiple redirects

closed 2022-03-18 by pschatzmann

ulegan, 2022-03-19:
> Android [4].
> Links:

works like a charm...
thank you very much, I eally appreciate your work