pschatzmann/arduino-audio-tools #129: streams-generator-i2s.ino example runtime errors on Espressif ESP32-C3-DevKitM-1 board

jimcondit, 2022-03-16:
Using the example:
/**
 * @file streams-generator-i2s.ino
 * @author Phil Schatzmann
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-i2s/README.md 
 * @copyright GPLv3
 */

Pins used: (code snippet from example source)
// start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.pin_bck = 4;
  config.pin_ws = 5;
  config.pin_data = 18;
  config.sample_rate = sample_rate; 
  config.channels = channels;
  config.bits_per_sample = 16;
  out.begin(config);
I'm using the Espressif ESP32-C3-DevKitM-1 board.  

Code compiles, but get runtime error:

E (133) gpio: gpio_set_level(226): GPIO output gpio_num error
starting I2S...
E (164) gpio: gpio_set_level(226): GPIO output gpio_num error
started...

Sound is something like a beating drum.

I was able to use Espressif's i2s_basic example which seems to work correctly, so I assume the pins I selected should work.  (Does roughly the same as your generator example.)

pschatzmann, 2022-03-17:
Since I don't have any ESP32-C3, I cant really help. 
My suggestion would be try other pins. Maybe 18 is already used for USB ?
If this does not help I suggest to open an issue with the Arduino ESP32 project. They should be  better able to help...

jimcondit, 2022-03-17:
Thanks, Phil.  I tried three different sets of pins.  The pins shown above are the ones that worked with Espressif's example.  I'm very interested to use your library.  Can you point me to the line that throwing the error message?
Thanks for your reply!

pschatzmann, 2022-04-15:
I finally received my ESP32-C3 from China. I noticed that the log level Info is much worse then on an ESP32. Just switching to Warning resolved the issue.

closed 2022-04-20 by pschatzmann