pschatzmann/arduino-audio-tools #130: No sound - URLStream MP3 to AudioKit speakers

JonSilver, 2022-03-17:
I'm planning to stream an MP3 audio blob from an Azure blob store to wired speakers upon receipt of a trigger message containing the MP3 url to play. I'm using an ESP32 LyraT board with your AudioKit HAL. Alas, all I get is a brief speaker pop that sounds like the sound is being played far too fast or something. 

@pschatzmann please can you tell me if there's anything obviously wrong? Or if my approach is just completely & fundamentally flawed?

In the log output there's this error... no idea if it's relevant:

```
[W] AudioKit.h : 43 - sd_active = false
E (5839) I2S: register I2S object to platform failed
```

There's also a never-ending series of touch pin events at the end, even though there's nothing anywhere near the touch buttons.

```
[I] AudioActions.h : 167 - touch pin: 13 value 50 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 27 value 68 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 13 value 50 (limit: 20) -> false
```

For the sample code below, I've created an example URL for testing purposes, valid for a couple of months from today, with read-only access to a simple MP3 example, so don't worry about the apparent security leak.

Thank you in advance,
Jon

---

Here's the code:

```
#define USE_HELIX 

#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include "AudioLibs/AudioKit.h"

const char *wifi = "ssid";
const char *password = "pw";
const char *uri = "https://bobbstorage.blob.core.windows.net/public/hello.mp3?sv=2020-10-02&st=2022-03-17T12%3A26%3A11Z&se=2022-05-18T11%3A26%3A00Z&sr=b&sp=r&sig=1LC5FCJCuEaEnFdvloEKbqPUPwfwIHakhnjoOGvdk0g%3D";

URLStream url(wifi, password);
AudioKitStream kit;
EncodedAudioStream dec(&kit, new MP3DecoderHelix());
StreamCopy copier(dec, url);

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);  

  auto config = kit.defaultConfig(TX_MODE);
  config.sd_active = false;

  kit.setVolume(100);
  kit.begin(config);
  
  dec.setNotifyAudioChange(kit);
  dec.begin();

  url.begin(uri, "audio/mp3");
}

void loop(){
  copier.copy();
  kit.processActions();
}
```

This is the output:

```
Error:   audio_hal handle is null
[I] I2SConfig.h : 89 - rx/tx mode: TX
[I] I2SConfig.h : 90 - port_no: 0
[I] I2SConfig.h : 91 - is_master: Master
[I] I2SConfig.h : 92 - sample rate: 44100
[I] I2SConfig.h : 93 - bits per sample: 16
[I] I2SConfig.h : 94 - number of channels: 2
[I] I2SConfig.h : 95 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 97 - auto_clear:0
[I] I2SConfig.h : 104 - buffer_count:10
[I] I2SConfig.h : 105 - buffer_size:512
[I] I2SConfig.h : 109 - pin_bck: 14
[I] I2SConfig.h : 110 - pin_ws: 15
[I] I2SConfig.h : 111 - pin_data: 0
[W] AudioKit.h : 43 - sd_active = false
[I] AudioKit.h : 193 - Channels 2->2
[I] AudioOutput.h : 635 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 648 - FormatConverterStream: no conversion
[I] AudioKit.h : 539 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 542 - Deactivating SPI because SD is not active
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 0
[I] AudioActions.h : 58 - pin 39 -> INPUT_PULLUP
[I] AudioActions.h : 42 - ActionLogic::add pin: 19 / logic: 2
[I] AudioActions.h : 61 - pin 19 -> INPUT
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 13 / logic: 3
[I] AudioActions.h : 61 - pin 13 -> INPUT
[I] AudioKit.h : 292 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 27 / logic: 3
[I] AudioActions.h : 61 - pin 27 -> INPUT
[I] URLStream.h : 69 - virtual bool audio_tools::URLStreamDefault::begin(const char*, const char*, MethodID, const char*, const char*): https://bobbstorage.blob.core.windows.net/publ
[I] Url.h : 73 - Url::parse
[I] Url.h : 109 - url->https://bobbstorage.blob.core.windows.net/public/hello.mp3?sv=2020-10-02&st=2022-03-17T12%3A26%3A11Z&se=2022-05-18T11%3A26%3A00Z&sr=b&sp=r&sig=1LC5FCJCuEa
[I] Url.h : 110 - host->bobbstorage.blob.core.windows.net
[I] Url.h : 111 - protocol->https
[I] Url.h : 112 - path->/public/hello.mp3?sv=2020-10-02&st=2022-03-17T12%3A26%3A11Z&se=2022-05-18T11%3A26%3A00Z&sr=b&sp=r&sig=1LC5FCJCuEaEnFdvloEKbqPUPwfwIHakhnjoOGvdk0g%3D
[I] Url.h : 113 - port->443
........
[I] URLStream.h : 202 - WiFiClientSecure
[I] URLStream.h : 202 - WiFiClientSecure
[I] HttpRequest.h : 182 - process connecting to host bobbstorage.blob.core.windows.net port 443
[I] HttpRequest.h : 170 - connect bobbstorage.blob.core.windows.net
[I] HttpRequest.h : 193 - Free heap: 138812
[I] HttpHeader.h : 353 - HttpRequestHeader::setValues - path: /public/hello.mp3?sv=2020-10-02&st=2022-03-17T12%3A26%3A11Z&se=2022-05-18T11%3A26%3A00Z&sr=b&sp=r&sig=1LC5FCJCuEaEnFdvloEK
[I] HttpHeader.h : 253 - HttpHeader::write
[I] HttpHeader.h : 362 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 377 - -> GET /public/hello.mp3?sv=2020-10-02&st=2022-03-17T12%3A26%3A11Z&se=2022-05-18T11%3A26%3A00Z&sr=b&sp=r&sig=1LC5FCJCuEaEnFdvloEKbqPUPwfwIHakhnjoOGvdk0g%3D HTT
[I] HttpHeader.h : 190 -  -> Host: bobbstorage.blob.core.windows.net 
[I] HttpHeader.h : 190 -  -> Connection: close 
[I] HttpHeader.h : 190 -  -> Accept: audio/mp3 
[I] HttpHeader.h : 293 -  -> <CR LF> 
[I] HttpRequest.h : 217 - Request written ... waiting for reply
[I] HttpHeader.h : 223 - HttpHeader::read
[W] HttpHeader.h : 230 - Waiting for data...
[I] HttpHeader.h : 160 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Content-Length: 2866
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Content-Type: audio/mpeg
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Content-MD5: MnP7h4tQNesbKm1cck6bjg==
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Last-Modified: Thu, 17 Mar 2022 12:41:04 GMT
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Accept-Ranges: bytes
[I] HttpHeader.h : 160 - HttpHeader::readLine -> ETag: "0x8DA081366831B90"
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Vary: Origin
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Server: Windows-Azure-Blob/1.0 Microsoft-HTTPAPI/2.0
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-request-id: e7d3130a-101e-001c-21fc-394203000000
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-version: 2020-10-02
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-creation-time: Thu, 17 Mar 2022 12:41:04 GMT
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-lease-status: unlocked
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-lease-state: available
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-blob-type: BlockBlob
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Content-Disposition: 
[I] HttpHeader.h : 160 - HttpHeader::readLine -> x-ms-server-encrypted: true
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Date: Thu, 17 Mar 2022 12:43:20 GMT
[I] HttpHeader.h : 160 - HttpHeader::readLine -> Connection: close
[I] HttpHeader.h : 160 - HttpHeader::readLine -> 
[I] URLStream.h : 84 - size: 2866
[W] AudioKit.h : 43 - sd_active = false
E (7284) I2S: register I2S object to platform failed
Error:   i2s_driver_install
[I] AudioOutput.h : 635 - void audio_tools::FormatConverterStream::setup()
[I] AudioOutput.h : 658 - FormatConverterStream: multiply channels by 2
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioActions.h : 167 - touch pin: 13 value 50 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 27 value 68 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 13 value 50 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 13 value 50 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 27 value 68 (limit: 20) -> false
[I] AudioActions.h : 167 - touch pin: 13 value 50 (limit: 20) -> false
```

pschatzmann, 2022-03-17:
The audio information should be updated automatically from the mp3 info. If this is not working you can try to set it to the right values to start with:

config.channels = 1
config.sample_rate = 22050

Please also note that if the mp3 is too short you might not hear the full sound because the remainder is still sitting in the buffer. You can avoid this issue by switching to wav or adding more silence at the end of your file

pschatzmann, 2022-04-13:
If you did not press any keys, it seems that the touch limits are too low. You can set your own limit by calling  e.g. kit.audioActions().setTouchLimit(70)

closed 2022-04-20 by pschatzmann