pschatzmann/arduino-audio-tools #133: SPIFFS support on ESP32?

Quas7, 2022-03-18:
Hi,

first of all: great lib!
Coming from the [earlephilhower/ESP8266Audio](https://github.com/earlephilhower/ESP8266Audio) I wanted use the SPIFFS as before but it looks like there is no direct support yet for that, right? 
Not a big deal, as I can move the data to the programming flash but is anything planned for SPIFFS? :)

Thanks.

pschatzmann, 2022-03-18:
I have never tried, but I would expect that if SPIFFS has proper Arduino support, the file is implementing the Stream interface and thus you should be able to use it a source...

Quas7, 2022-03-18:
ah, ok. I will play around with it and report back.

Quas7, 2022-03-21:
I got it working with SPDIFFS on an ESP32 using the internal DAC.

Only the volume adjustments I still do not get via VolumeStream. 
Do I understand the issue at least somewhat correctly, that VolumeStream should scale the decoder stream but here the decoder is directly going for &out?

Here the essential lines of code for anybody looking for a quick starting point.

```

#include <Arduino.h>
#include <SPIFFS.h>
#include <SD.h>
#include "AudioTools.h"

File audioFile;
AnalogAudioStream out;                  // output via the internal DAC of the ESP32
EncodedAudioStream decoder(&out, new WAVDecoder());      //decoding wave files 16bit signed PCM
VolumeStream volume(decoder);     // the volume control does not work yet
StreamCopy copier;                          // copies sound via i2s to the DAC

void setup()
{
    // Open Serial 
  Serial.begin(115200);
  
//AudioLogger::instance().begin(Serial, AudioLogger::Debug);    // using debug will result in stutter (as to be expected)
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  if (!SPIFFS.begin()) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }

  auto config = out.defaultConfig(TX_MODE);   

  config.sample_rate = 44100;   //44.1kHz  --- always make sure, that the same config is loaded for decoder and out ---
  config.channels = 1;      // mono sound only here
  config.bits_per_sample = 16;   //16 bit signed PCM

  out.begin(config);
  audioFile = SPIFFS.open("/read.wav");  // load ready wave file from SPIFFS
  decoder.setNotifyAudioChange(out);
  decoder.begin(config);
  volume.setVolume(1);      // does not work yet
  copier.begin(decoder, audioFile);
}

void loop()
{
  copier.copy();   // play
}

```

And debug output as reference:
```

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I] AnalogAudioESP32.h : 70 - I2S_MODE_DAC_BUILT_IN
[I] AnalogAudioESP32.h : 195 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 38 - sample_rate: 44100
[I] AudioTypes.h : 39 - channels: 1
[I] AudioTypes.h : 40 - bits_per_sample: 16
[I] AnalogAudioESP32.h : 90 - analog left output pin: 25
[I] AnalogAudioESP32.h : 91 - analog right output pin: 26
[I] AnalogAudioESP32.h : 263 - TX_MODE
[I] AudioStreams.h : 720 - setVolume: 1.000000
[I] AudioCopy.h : 66 - buffer_size=1024
[I] CodecWAV.h : 49 - WAVHeader len: 1024
[I] CodecWAV.h : 215 - WAVHeader sound_pos: 44
[I] CodecWAV.h : 216 - WAVHeader channels: 1
[I] CodecWAV.h : 217 - WAVHeader bits_per_sample: 16
[I] CodecWAV.h : 218 - WAVHeader sample_rate: 44100
[I] CodecWAV.h : 219 - WAVHeader format: 1
[I] CodecWAV.h : 314 - WAV sample_rate: 44100
[I] CodecWAV.h : 315 - WAV data_length: 53056
[I] CodecWAV.h : 316 - WAV is_streamed: 0
[I] CodecWAV.h : 317 - WAV is_valid: true
[I] CodecWAV.h : 335 - isValid: true
[I] AnalogAudioESP32.h : 175 - virtual void audio_tools::AnalogAudioStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] CodecWAV.h : 338 - WAVDecoder writing first sound data
[I] AudioCopy.h : 257 - try write - 2
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 2 hops
...
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 121 - StreamCopy::copy 876 -> 876 -> 876 bytes - in 1 hops
```

pschatzmann, 2022-03-21:
The following chain should work: EncodedAudioStream -> VolumeStream -> AnalogAudioStream.

AnalogAudioStream out;                  // output via the internal DAC of the ESP32
VolumeStream volume(out);     // the volume control does not work yet
EncodedAudioStream decoder(&volume, new WAVDecoder());      //decoding wave files 16bit signed PCM

Quas7, 2022-03-22:
ah, nice - volume is working now. Thanks for the help.

closed 2022-03-22 by Quas7