pschatzmann/arduino-audio-tools #135: Multiple connections to WAVE/AAC Stream

IsiTapier, 2022-03-21:
I noticed that I can't connect to the encoder server, after one device allready connected to it. But I planed to broadcast audio from one esp into Wifi and receive it from multiple others. Is there an way to fix this? Or does someone know where this problem comes from?

pschatzmann, 2022-03-22:
The integrated server is just a hack that I have implemented so that I don't need to wire anything up to test some audio.
So, it just supports one connection. 

What exactly are you trying to do ?

IsiTapier, 2022-03-22:
I'm trying to use one ESP32 Audio boards to send audio (from aux in) over wifi to multiple other boards (which give it out via aux out
So I need a way to transmit the audio stream via Wifi

pschatzmann, 2022-03-22:
Right now I do not have any ready and tested solution. 
However I have started on a couple of approaches which could be promising.  The first one is based around esp-now. Please let me know if you are interested to spend some effort to help me to complete it to have a working solution...

IsiTapier, 2022-03-22:
With your assistance, I might could try it, but I'm not that experienced with this kind of stuff.
Do you have these approches somewhere on github, so I might could take a look?

pschatzmann, 2022-03-22:
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/sandbox/communication
- espnow-receive
- espnow-send

IsiTapier, 2022-03-22:
I rewrote the AudioServer lib so that it can handle multiple clients, but with the AudioEncoderServer, I ran in the issue that the Encoder only takes an output Stream on whiches it writes itself, but I need it as a stream object, which I can pass to the audio server. Is there a way to reroute the stream like that?

pschatzmann, 2022-03-22:
If you want to go down that road you need to be aware that the logic depends on the audio format: If you want to distribute PWM data you need to make sure that for each new connection you send a WAV header first. For other audio formats the logic is different and you might get away with just writing the same encoded data to all connected devices.

IsiTapier, 2022-03-22:
Thanks a lot, the hint with the Multi Output was great.

I'm probably going to use both Wifi and esp now, since Wifi is better, because I can use existing networks to have a huge area cover. But esp now is best when I don't have a wifi network available..

I have one little problem:
The encodedAudioStream took a Print object, which I was able to convert my WifiClient object into (original code)
But since I'm now using the MultiAudio instead, I can't put my WifiClients into it, because it only takes a AudioPrint object, not a Print one.

pschatzmann, 2022-04-13:
I extended the MultiOutput, so that you can add objects which are subclasses of Print

closed 2022-04-20 by pschatzmann