pschatzmann/arduino-audio-tools #136: Maximilian Quad Mixing

mageeagle, 2022-03-22:
I'm trying to use ESP32 to drive 2 External Stereo DACs for Quad Panning. However, I can't seem to find how to put two i2s streams into Maximilian to use it. If there are any leads on this?

pschatzmann, 2022-03-22:
The only way I see would be to use 2 Maximilian objects linked to 2 I2SStreams:

I2SStream out1;
Maximilian maximilian1(out1);
I2SStream out2;
Maximilian maximilian1(out2);

In the i2s config you need to assign different ports (0 and 1) and pins...

pschatzmann, 2022-03-22:
I recommend to set the log level to Warning.
It might also help to decrease the sampling rate...

mageeagle, 2022-03-22:
Decreasing sample rate helped!
I'm not sure if I understand the buffer size well:
if the buffer size is the second argument?
`Maximilian maximilian(out, 32);
Maximilian maximilian2(out2, 32);`

Right now, the two audio boards are producing identical signals: i.e.
output[0] and output[1] goes to both boards

while it should be:
output[0] and output[1] to first board and output[2] and output[3] to the second board

Am I missing something?

pschatzmann, 2022-03-22:
I see what's happening: There is only one play function but we actually would need a second different one for the second output...

pschatzmann, 2022-03-22:
I committed some change which should allow the following:
Maximilian maximilian(out, 32); Maximilian maximilian2(out2, 32, play2);

void play2(float *output) {
    output[1]=myQuadOutput[2] * 0.125;
    output[2]=myQuadOutput[3] * 0.125;
}

mageeagle, 2022-03-22:
Thanks, The panning works now!
Though weirdly, the output from the second board sounds bit crushed / bit reduced, while the first board sounds just fine
Any leads?

pschatzmann, 2022-03-23:
I am afraid, this time I can't help.
I have never tried to use 2 I2S output at the same time...

closed 2022-03-23 by mageeagle