pschatzmann/arduino-audio-tools #138: SineWaveGenerator saturates sinf()

mikemccauley, 2022-03-22:
Current implementation of SineWaveGenerator::readSample() saturates the sinf() function after a few minutes.
Symptoms are initially wierd overtones int the sine waveoutput, and eventually after 15 minutes flatlines. Main cause is float lack or precision. Also affects SquareWaveGenerator.

Humbly suggest that readSample should be like:
        /// Provides a single sample
        virtual T readSample() override {
            float angle = PI * 2.0 * m_cycles + m_phase;
            T result = m_amplitude * sinf(angle);
	    m_cycles += m_frequency * m_deltaTime;
	    if (m_cycles > 1.0)
		m_cycles -= 1.0;
            return result;
        }

and the members of SineWaveGenerator:

    protected:
        volatile float m_frequency = 0;
        float m_cycles = 0.0; // Varies between 0.0 and 1.0
        float m_amplitude = 1.0;  
        float m_deltaTime = 0.0;
        float m_phase = 0.0; // 0 to 2 * PI

this keeps the value passed to sinf within the range 0 to (2 * PI + m_phase)

pschatzmann, 2022-03-23:
The correction has been committed - Thanks for your input

closed 2022-03-23 by pschatzmann

mikemccauley, 2022-03-24:
Thanks for your quick response and a very interesting library.

Cheers.