pschatzmann/arduino-audio-tools #139: Feature Request: Volume Control for separate channels

mageeagle, 2022-03-23:
I wanted to make a panner out of my ESP32 board. 
Right now there is the volume control feature which controls the volume of audio streams.
https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control

Though it controls the amplitude of both channels of a stereo output stream together with one value.
Wonder if it's possible to control the amplitude separately for each channel? So that it works like a panner for an audio stream?

pschatzmann, 2022-03-24:
I extended the [VolumeControl](https://pschatzmann.github.io/arduino-audio-tools/html/classaudio__tools_1_1_volume_stream.html). Can you help me to test it ?

mageeagle, 2022-03-24:
It works perfectly! Thanks!

closed 2022-03-24 by mageeagle