pschatzmann/arduino-audio-tools #143: Chipmunks (fast/higher pitch) sound after stopping and trying to resume a stream

nfauchelle, 2022-03-25:
Hello,

Thanks very much for this library. I've used it to get some microphones working and now trying to stream some audio.

The goal is to used MQTT to start/stop an audio stream. The URL is included in the start method. I can get it to start and stop, but when I go to start a stream again, the audio comes out "chipmunk" style.

Using Arduino sketch.
```
    URLStream urlStream;
    I2SStream i2sSpeaker;
    VolumeStream speakerVolume(i2sSpeaker);
    MP3DecoderHelix helix;
    EncodedAudioStream speakerStreamDecoder(&speakerVolume, &helix);
    StreamCopy speakerStream(speakerStreamDecoder, urlStream);
    auto speakeri2sConfig = i2sSpeaker.defaultConfig(TX_MODE);
    static const i2s_port_t is2SpeakerPort = I2S_NUM_1;
```

void setup method
```
    speakeri2sConfig.port_no = is2SpeakerPort;
    speakeri2sConfig.pin_bck = GPIO_NUM_32;
    speakeri2sConfig.pin_ws = GPIO_NUM_23;
    speakeri2sConfig.pin_data = GPIO_NUM_14;
```

void start method
```
    i2sSpeaker.begin(speakeri2sConfig);
    speakerStreamDecoder.setNotifyAudioChange(i2sSpeaker);    
    speakerStreamDecoder.begin();
    speakerVolume.begin(speakeri2sConfig);
    speakerVolume.setVolume(0.25);
    urlStream.begin(mqttMessage.c_str(), "audio/mp3");
```

void stop method
```
    speakerStream.copy();
    urlStream.end();
    helix.end();
    speakerStreamDecoder.end();
    i2sSpeaker.flush();
    i2sSpeaker.end();
    speakerVolume.end();
```

As mentioned, it starts and stops fine the first time. Starting again = chipmunk but stops again fine.
I assume when I stop/start something is getting doubled up.

I considered using the player, but that takes an array of URLS and I only want a single stream. It wasn't clear how to swap out the list of urls.

Can you see where I am going wrong here / any points of starting/stopping and changing urls when the URL isn't known at compile time.

Thanks again

pschatzmann, 2022-03-26:
You would get such an effect when the sample rate is not right. I suggest that you set the log level to info and check out the log messages related to the sample rate. So far I had never issues with setNotifyAudioChange which is supposed to update the sample rate in i2s based on the information from the decoder.  

If you know the rate in advance you can set it with speakeri2sConfig.sample_rate = value....

nfauchelle, 2022-03-26:
@pschatzmann 
Pulling in the change for the libhelix resolved the issue!

I will mark this off as closed, but wanted to ask, is the above code correct for stopping a urlStream, changing the URL and then starting again, or is there a better way? It could be several minutes between stopping and starting another stream, so fully stopping is preferred. Thanks again!

pschatzmann, 2022-03-27:
The logic looks correct to me. 
In the Audio Player, I just stop the copy and reopen the URLStream - but this is because I do not expect any long breaks...

nfauchelle, 2022-03-27:
Having a bit more of a play, when I play a new stream there is half a second / a second of the previous audio playing first. I wonder if I order I am ending / flushing is not correct and some data is being left in some buffer in one of the objects.

Looking at my stop method in the issue, it seems pretty good to me, so I need to call copy on the StreamCopy after I've ended most things? When I tried that I get a panic.

pschatzmann, 2022-03-27:
This is a feature of the Decoder. If you close the decoder the buffer gets cleared.
In the AudioPlayer I manage the copy via a boolean variable. You might do the same thing as well...

nfauchelle, 2022-03-28:
Thanks for the response, maybe I didn't explain myself properly.

When I use my stop code above, then wait a few minutes and then hit play, I still have half a second to a second of the previous mp3 play first. When I use that stop code, I am aiming to clear / empty the buffers. You mentioned if I close the decoder the buffer is cleared. Is calling end the same as closing?

pschatzmann, 2022-03-28:
Correct: end is closing the decoder

nfauchelle, 2022-03-28:
I am doing that when I stop, yet it's not fully clearing then. Do I need to end/flush something else, or let another copy event fire?

pschatzmann, 2022-03-28:
In this case I would not know what else could cause this...

nfauchelle, 2022-03-30:
Thanks for the suggestions.
I swapped out to the WAVDecoder and gave that a go. Doesn't seem to suffer from the same issue.

closed 2022-04-20 by pschatzmann