pschatzmann/arduino-audio-tools #146: URL player and private MQTT client

podaen, 2022-03-28:
Hi,

The last weekend I tried to implement an https MQTT client added to the URL player (http or https all the same outcome). I have updated the sercureclient library to the latest version and tested everything without the URL player and it works. When I added audiotools to my project and start with player.begin I am not able to connect to the my MQTT client anymore. This is the log I got when it get wrong.

X509 - Certificate verification failed, e.g. CRL, CA or signature check failed  
ssl_client.h ->  while ((ret = mbedtls_ssl_handshake(&ssl_client->ssl_ctx)) != 0) 

I re added the CA certificate and the I tied with player.end(), hoping it would continu, but it doesn't.

podaen, 2022-03-28:
Some news, when I start with a http stream and after some time playing the player gets a timeout, then the mqtt is publishing my messages to the server. But than this happens [D][WiFiClient.cpp:515] connected(): Disconnected: RES: 0, ERR: 128

pschatzmann, 2022-03-28:
Well, the error message is that you do not have enough memory:
[E][ssl_client.cpp:516] _handle_error(): [start_ssl_client():748]: (-32512) SSL - Memory allocation failed

podaen, 2022-03-28:
I'm against the limit with a stack of 5000, when I lower it it get stuck. How much memory do I need for a insecure https connection? I seems more than 11576... That is much!

> xPortGetFreeHeapSize: 11576
> player stoped
> https://25493.live.streamtheworld.com:443/NOSTALGIEWAFNEWWAVE.mp3
> try to send a msg to mqtt
> WiFi allready connected
> Attempting MQTT connection...
> clientId: ESP32Client-1393
> MQTT connected
> Try publish Message [RadioStation]: https://25493.live.streamtheworld.com:443/NOSTALGIEWAFNEWWAVE.mp3
> 25493.live.streamtheworld.com
> 443
> [I][ssl_client.cpp:654] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> Setting hostname for TLS session...
> [E][ssl_client.cpp:516] _handle_error(): [start_ssl_client():748]: (-32512) SSL - Memory allocation failed
> [E][WiFiClientSecure.cpp:483] connect(): start_ssl_client: -32512
> [E] HttpRequest.h : 187 - Connect failed

podaen, 2022-03-28:
Are these every time added when I start playing the music? If yes, than remains around 2000kb for the security. If the default setting is 2048kb than I am close...

```
#define DEFAULT_BUFFER_SIZE 1024
#define I2S_BUFFER_SIZE 512
#define I2S_BUFFER_COUNT 10
```

pschatzmann, 2022-03-28:
Yes - tls uses a lot of resources. You could try to replace mqtt with simple udp calls. 
This should be very lean...

podaen, 2022-03-28:
Strange that a mqtt client uses only 3000b. What is in fact encrypted, the data or the header?

podaen, 2022-03-28:
You are using an insecure connection, soo I think the data is encryped...

podaen, 2022-03-29:
I did some research and you are right. The fact that https is using more memory is because it uses more resources.
The wifi is also using the memory to keep up with the data income. But In my case I don't have the possibility to do that, because I simply don't have an udp cable on that location.

I did some further investigation on the hopsize and it seems that there is no memory saves on that...

> [I] AudioCopy.h : 118 - StreamCopy::copy 418 -> 418 -> 418 bytes - in 1 hops
> xPortGetFreeHeapSize: 51664
> 
> [I] AudioCopy.h : 118 - StreamCopy::copy 627 -> 627 -> 627 bytes - in 1 hops
> xPortGetFreeHeapSize: 58748
> 
> [I] AudioCopy.h : 118 - StreamCopy::copy 626 -> 626 -> 626 bytes - in 1 hops
> xPortGetFreeHeapSize: 56352
> 
> [I] AudioCopy.h : 118 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
> xPortGetFreeHeapSize: 58296

Note: In my tests with the https streams, some fails. This is because the websites don't allow the insecure connection setting in the ssl_client. The certificate is verified but no data is coming in. The all end like this. 

> [W] HttpHeader.h : 227 - Waiting for data...

A solution to that would be, download the certificate and store it and use it to set a asynchronous encryption. 

I don't think anymore it will save me allocated memory.

ps: I still have some links that works with insecure setting so I am not angry about using it.

The most convenient thing to do to save memory is clean up my code...

pschatzmann, 2022-03-29:
For udp you don't need any cable. This is a stateless network protocol that you can use with the wificlient. 
The solution with the certificate only works if you always connect to the same url or you will need to provide certificates for all relevant destinations.
If you have some local webserver (e.g. nginx or apache) you could create proxies to your https locations and use http from the esp32 to your webserver

podaen, 2022-03-29:
> For udp you don't need any cable. This is a stateless network protocol that you can use with the wificlient.

How can I do that... Replace this with what?
URLStream urlStream(wifi, password);

> The solution with the certificate only works if you always connect to the same music url or you will need to provide certificates for all relevant destinations.

Your sure? https://maakbaas.com/esp8266-iot-framework/logs/https-requests/

pschatzmann, 2022-03-29:
It would replace your MQTT functionality. By the way if you use the URLStream together with MQTT you would not pass the wifi, password parameter but the Client in the constructor. This is the same object that you already need to setup up for MQTT.

podaen, 2022-03-29:
And a http connection to my local webserver would work? By sending packages like this.

    udp.beginPacket(udpAddress,udpPort);
    udp.printf("Seconds since boot: %lu", millis()/1000);
    udp.endPacket();

What about esp now? This is an easy way to send packages between esp's I have allready ben using in several projects and is very populair with esp users.

[lhttps://demo-dijiudu.readthedocs.io/en/latest/api-reference/wifi/esp_now.html](https://randomnerdtutorials.com/esp-now-esp32-arduino-ide/)

pschatzmann, 2022-03-29:
If I remember right esp now uses Bluetooth - so this might not work if you use it together with a URLStream which uses tcp/ip

podaen, 2022-03-29:
No, it is not. It uses wifi.h

Here the documentation for espressif
https://demo-dijiudu.readthedocs.io/en/latest/api-reference/wifi/esp_now.html

pschatzmann, 2022-03-29:
Cool - did you measure the memory usage? 
The thing which would worry me is, that it also uses encryption.

podaen, 2022-03-29:
The espnow I am using doesn't use any encryption. Because I diden't update the arduino yet.

> By the way if you use the URLStream together with MQTT you would not pass the wifi, password parameter but the Client in the constructor.

Would it be a problem that I do this below together with URLStream urlStream(wifi, password); ?

```
bool ConnectWifi() {
	if ((WiFi.status() != WL_CONNECTED))
	{
		Serial.println("WiFi try to reconnect");
		if (WiFi.reconnect()) {
			Serial.println("WiFi reconnected");
			return true;
		}
		else
		{
			WiFi.mode(WIFI_STA);
			WiFi.begin(ssid, pass);
			TryConnectPrevious = millis();
			while (WiFi.status() != WL_CONNECTED) {
				delay(500);
				Serial.print(".");
				currentMillis = millis();
				if (currentMillis - TryConnectPrevious > TryConnectDelay) {
					Serial.println("Failed to connect");
					return false;
				}
			}
			if (WiFi.status() == WL_CONNECTED) {
				Serial.println("WiFi connected");
				return true;
			}
			else
			{
				Serial.println("Failed to connect");
				return false;
			}
		}
	}
	else
	{
		Serial.println("WiFi allready connected");
		return true;
	}
}
```

podaen, 2022-03-29:
If I understand it right, for the udp I need to switch my URL player from STA to AP to send a messages another client and than reconnect it to STA to use the player again. This happens when I switch between two stations so I don't an issue here, I hope it doesn't take to long to do the switching.

I did a memory check on only a mtqq shetch.

> Opening port
> Port open
> VMDPV_1|1_xPortGetFreeHeapSize: 262292
> 
> WiFi connected
> IP address: 
> 192.168.1.242
> xPortGetFreeHeapSize: 210508
> 
> Attempting MQTT connection...[V][ssl_client.cpp:540] start_ssl_client(): Free internal heap before TLS 279372
> [V][ssl_client.cpp:546] start_ssl_client(): Starting socket
> [V][ssl_client.cpp:624] start_ssl_client(): Seeding the random number generator
> [V][ssl_client.cpp:633] start_ssl_client(): Setting up the SSL/TLS structure...
> [V][ssl_client.cpp:657] start_ssl_client(): Loading CA cert
> Loading CA cert
> Setting hostname for TLS session...
> [V][ssl_client.cpp:738] start_ssl_client(): Setting hostname for TLS session...
> xPortGetFreeHeapSize: 171504
> Performing the SSL/TLS handshake...
> [V][ssl_client.cpp:757] start_ssl_client(): Performing the SSL/TLS handshake...
> Verifying peer X.509 certificate...
> [V][ssl_client.cpp:779] start_ssl_client(): Verifying peer X.509 certificate...
> Certificate verified.
> [V][ssl_client.cpp:789] start_ssl_client(): Certificate verified.
> [V][ssl_client.cpp:804] start_ssl_client(): Free internal heap after TLS 235592
> [V][ssl_client.cpp:857] send_ssl_data(): Writing HTTP request with 63 bytes...
> connected
> [V][ssl_client.cpp:857] send_ssl_data(): Writing HTTP request with 14 bytes...
> [V][ssl_client.cpp:857] send_ssl_data(): Writing HTTP request with 61 bytes...
> Message publised [sensor1]: 29
> [V][ssl_client.cpp:857] send_ssl_data(): Writing HTTP request with 61 bytes...
> Message publised [sensor2]: 73.00
> xPortGetFreeHeapSize: 164148
> Message arrived [sensor1]11
> message recieved: 11

I did a memory check with only the espnow (old version without encryption)

> Opening port
> Port open
> xPortGetFreeHeapSize: 265212
> 
> WiFi try to reconnect
> .....WiFi connected
> xPortGetFreeHeapSize: 213608
> end setup
> Sent with success
> xPortGetFreeHeapSize: 213276

Soo, 265212 - 213276 uses for the espnow and 262292 - 164148 for the mqtt. That's 50k and I can send and receive messages from the STA mode. I will test if it works with the stream.

pschatzmann, 2022-03-29:
No need to switch: The only thing which is important is that the sending and receiving devices are on the same network

podaen, 2022-03-29:
esp_now.h (espressif - arduino sdk 1.0.6 non encrypted) seems to work... I didn't check if the message actually is received on the other side. That's the next thing to do.

podaen, 2022-03-30:
Could somebody please try if this works with the url-player with framework +2.0.2?

```
/*
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp-now-esp32-arduino-ide/
  
  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files.
  
  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*/

#include <esp_now.h>
#include <WiFi.h>

// REPLACE WITH YOUR RECEIVER MAC Address
uint8_t broadcastAddress[] = {0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF};

// Structure example to send data
// Must match the receiver structure
typedef struct struct_message {
  char a[32];
  int b;
  float c;
  bool d;
} struct_message;

// Create a struct_message called myData
struct_message myData;

esp_now_peer_info_t peerInfo;

// callback when data is sent
void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("\r\nLast Packet Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
}
 
void setup() {
  // Init Serial Monitor
  Serial.begin(115200);
 
  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);

  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }

  // Once ESPNow is successfully Init, we will register for Send CB to
  // get the status of Trasnmitted packet
  esp_now_register_send_cb(OnDataSent);
  
  // Register peer
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
  
  // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
}
 
void loop() {
  // Set values to send
  strcpy(myData.a, "THIS IS A CHAR");
  myData.b = random(1,20);
  myData.c = 1.2;
  myData.d = false;
  
  // Send message via ESP-NOW
  esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &myData, sizeof(myData));
   
  if (result == ESP_OK) {
    Serial.println("Sent with success");
  }
  else {
    Serial.println("Error sending the data");
  }
  delay(2000);
}
```
Source: https://randomnerdtutorials.com/esp-now-esp32-arduino-ide/

If it works without encryption it would be better. But if it doesn't send messages you have to added an pmk key (AES-128 algorithm) that contains the ssid and pasword and has a length of 16bytes
https://esp32.com/viewtopic.php?t=12615
https://github.com/RomelSan/WPA-PMK-Generator

podaen, 2022-03-30:
The disadvantage for udp is that the ip need to be static. With internet connection lost and with fast dhcp devices in house it could give some mal function.

podaen, 2022-03-30:
Here is the log on the receiving side, it is not very clear why messages are not delivered sometimes.

> Opening port
> Port open
> [D][Hall_Alarm_Music_V2.ino:509] setup(): Debug
> [I][Hall_Alarm_Music_V2.ino:510] setup(): Info
> [W][Hall_Alarm_Music_V2.ino:511] setup(): Warning
> [E][Hall_Alarm_Music_V2.ino:512] setup(): Error
> load local config
> Read Eprom RadioActived: 0
> Read Eprom ProgramSelector: 2
> Read Eprom Generate playlist: 0
> xPortGetFreeHeapSize: 166652
> 
> WiFi try to reconnect
> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
> ....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
> [D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.242, MASK: 255.255.255.0, GW: 192.168.1.1
> .WiFi connected
> 23603.live.streamtheworld.com
> 80
> [W] HttpHeader.h : 225 - connected
> [W] HttpHeader.h : 227 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> player stoped
> Read Eprom UrlSelector: 12
> player started
> end setup loop 1
> Select Stream
> WiFi allready connected
> 
> Last Packet Send Status:	Delivery Success
> 22183.live.streamtheworld.com
> 80
> [W] HttpHeader.h : 225 - connected
> [W] HttpHeader.h : 227 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> http://22183.live.streamtheworld.com/JOE.mp3
> write eeprom UrlSelector: 2
> xPortGetFreeHeapSize: 59312
> 
> xPortGetFreeHeapSize: 57616
> 
> Select Stream
> WiFi allready connected
> 
> Last Packet Send Status:	Delivery Fail
> 23623.live.streamtheworld.com
> 443
> [I][ssl_client.cpp:654] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> Setting hostname for TLS session...
> xPortGetFreeHeapSize: 27716
> 
> Performing the SSL/TLS handshake...
> Verifying peer X.509 certificate...
> Certificate verified.
> [W] HttpHeader.h : 225 - connected
> [W] HttpHeader.h : 227 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> https://23623.live.streamtheworld.com:443/TOPRETRO.mp3
> write eeprom UrlSelector: 3
> xPortGetFreeHeapSize: 14468
> 
> xPortGetFreeHeapSize: 13008
> 
> Select Stream
> WiFi allready connected
> 
[... 372 lines cut ...]
> Certificate verified.
> [W] HttpHeader.h : 225 - connected
> [W] HttpHeader.h : 227 - Waiting for data...
> [W] HttpRequest.h : 145 - no CONTENT_LENGTH found in reply
> https://25493.live.streamtheworld.com:443/NOSTALGIEWAFEXTRA.mp3
> write eeprom UrlSelector: 24
> xPortGetFreeHeapSize: 9116
> 
> libhelix -  -> invalid frame size: -1 / max: 3000
> xPortGetFreeHeapSize: 9116
> 
> xPortGetFreeHeapSize: 8704
> 
> Select Stream
> WiFi allready connected
> 
> Last Packet Send Status:	Delivery Fail
> 25493.live.streamtheworld.com
> 443
> [I][ssl_client.cpp:654] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
> Setting hostname for TLS session...
> xPortGetFreeHeapSize: 22972
> 
> Performing the SSL/TLS handshake...
>

podaen, 2022-03-31:
I think I have to add a tasks and use a simple shetch on the receiving side for testing to optimize the results.

podaen, 2022-04-08:
I tested the [ESPnow example](https://github.com/Robocraze/ESP-Now) with 100% succes. What I notest is that wifi need to be disconnected before initialize it. That I can do in the in my sketch and I will try to implement it.

pschatzmann, 2022-04-08:
Good to know!

podaen, 2022-04-09:
I have it working now with the wifi on the sending esp. The problem was that the esp_now was sending messages on channel 1 by default and the wifi from the modem has two channels... In my case channel 6 and above. So what I did was setting the channel of the wifi to 6 and the esp_now also on channel 6.

Esp_now
`slaves[slaveCount].channel = 6;`

Wifi
```
WiFi.mode(WIFI_STA);
WiFi.channel(6);
WiFi.begin(ssid, pass);
```

And now it works together.

podaen, 2022-04-11:
And this setting is also needed on the receiving side.

`WiFi.mode(WIFI_AP_STA);`

closed 2022-04-17 by podaen